tzapu/WiFiManager #1311: AutoConnectWithFSParameters with OnDemandConfigPortal?

cds333, 2021-11-11:
I want to be able to reopen the full config portal from AutoConnectWithFSParameters, at any time so the user may change the stored additional parameters JSON data. 

However following the OnDemandConfigPortal example (using _wm.startConfigPortal()_) only gives us the basic config portal for SSID and password.

For AutoConnectWithFSParameters, if we try to re-run setup() of course it will not give us another portal unless the wifi settings are reset, which we don't want to do.

What is the correct way to do this?

Thank You

tablatronix, 2021-11-11:
Sounds like you are starting a new instance of wifigmanager in scope, use a global  wm object

cds333, 2021-11-13:
When I try to use a global instance it still does not work properly. 

When bringing up the config portal on demand, it works great as long as I enter the SSID and password again, but if I attempt to change the additional parameters and click "save", without re-entering the SSID/password, it will say "**_*wm:[2] No ssid, skipping wifi save_** ", despite the correct SSID and password populating the top two fields (they are greyed).

It then times out and tries to reset.

Here is the serial dump after clicking save:

*wm:[2] processing save 
*wm:[2] No ssid, skipping wifi save 
*wm:[2] Disabling STA 
*wm:[2] Portal Timeout In 106 seconds
*wm:[2] Portal Timeout In 76 seconds
*wm:[2] Portal Timeout In 46 seconds
*wm:[2] Portal Timeout In 16 seconds
*wm:[1] config portal has timed out 
*wm:[2] shutdownConfigPortal 
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed 
*wm:[1] config portal exiting 
failed to connect and hit timeout

It does autoconnect again after it resets, so clearly nothing is wrong with the saved credentials.

How do I make it reconnect, using the previously saved credentials, after changing only the additional parameters?

Thanks!

`
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

char mqtt_server[40];
char mqtt_port[6] = "8080";
char api_token[34] = "YOUR_APITOKEN";

char static_ip[16] = "10.0.1.56";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

#define TRIGGER_PIN 0

unsigned int  timeout   = 120; // seconds to run for

WiFiManager wifiManager;

WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 5);
WiFiManagerParameter custom_api_token("apikey", "API token", api_token, 34);

bool shouldSaveConfig = false;

void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {
  Serial.begin(115200);
  
  wifiManager.setSaveConfigCallback(saveConfigCallback);
 
  if (SPIFFS.begin()) {
    if (SPIFFS.exists("/config.json")) {
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        
        size_t size = configFile.size();
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
        DynamicJsonDocument json(1024);
        auto deserializeError = deserializeJson(json, buf.get());
        serializeJson(json, Serial);
        if ( ! deserializeError ) {
#else
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
#endif
          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
          strcpy(api_token, json["api_token"]);
          if (json["ip"]) {
            strcpy(static_ip, json["ip"]);
            strcpy(static_gw, json["gateway"]);
            strcpy(static_sn, json["subnet"]);
          } 
        } 
      }
    }
  } 

  Serial.println(static_ip);
  Serial.println(api_token);
  Serial.println(mqtt_server);

  IPAddress _ip, _gw, _sn;
  _ip.fromString(static_ip);
  _gw.fromString(static_gw);
  _sn.fromString(static_sn);

  wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn);

  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_api_token);

  wifiManager.setMinimumSignalQuality();

  wifiManager.autoConnect();
  
  Serial.println("connected...yeey :)");

  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(api_token, custom_api_token.getValue());

  if (shouldSaveConfig) {
    Serial.println("saving config");
#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
    DynamicJsonDocument json(1024);
#else
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
#endif
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"] = mqtt_port;
    json["api_token"] = api_token;

    json["ip"] = WiFi.localIP().toString();
    json["gateway"] = WiFi.gatewayIP().toString();
    json["subnet"] = WiFi.subnetMask().toString();

    File configFile = SPIFFS.open("/config.json", "w");

#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
    serializeJson(json, Serial);
    serializeJson(json, configFile);
#else
    json.printTo(Serial);
    json.printTo(configFile);
#endif
    configFile.close();
  }

  Serial.println(WiFi.localIP());
  Serial.println(WiFi.gatewayIP());
  Serial.println(WiFi.subnetMask());
}

void loop() {

  if (digitalRead(TRIGGER_PIN) == LOW) {
   
  wifiManager.setConfigPortalTimeout(timeout);

  if (!wifiManager.startConfigPortal("OnDemandAP")) {
    Serial.println("failed to connect and hit timeout");
  }else{
    Serial.println("connected...yeey :)");
  }
 }
`

tablatronix, 2021-11-13:
you can use shouldbeakafterconfig to always return on save.

or use the new callback for params to flag that params saved
```
/**
 * setSaveParamsCallback, set a save params callback on params save in wifi or params pages
 * @access public
 * @param {[type]} void (*func)(void)
 */
void WiFiManager::setSaveParamsCallback( std::function<void()> func ) {
  _saveparamscallback = func;
}
```

tablatronix, 2021-11-13:
also try

    // add params to its own menu page and remove from wifi, NOT TO BE COMBINED WITH setMenu!
    void          setParamsPage(bool enable);

and see if you like that better to avoid wifi cred confusion

cds333, 2021-11-14:
setParamsPage works perfectly! Thank You!

One last issue- the static ip/gw/sn fields from _http://192.168.4.1/wifi_ populate automatically every time the page is loaded, with the custom values I entered. This is working as intended. 

The params fields however, do not- they populate with the default values from the .ino file, despite the correct new values being saved and showing up in the serial output.

How would I make the fields from _http://192.168.4.1/param_ populate with the new saved values?

tablatronix, 2021-11-14:
You will have to load them from where you stored them and setvalue or use the argument when you declare your params. Are you using littlefs?

cds333, 2021-11-14:
You're right- I didn't notice that the serial output was showing the values I saved previously, before I added the setParamsPage(true) line.

So when the params were together on the wifi page, and I entered ssid+password+params, it saves all of them to the json. However when I have the params on the separate page, clicking save does not save them to json.

I am using SPIFFS btw; I am just using the example you wrote, after modifying the wm to be global.

Either way I should be able to workaround- I am just not quite sure how to programmatically populate the HTML fields. Could you point me towards the line of code that populates the ip field, for example, with the saved value?
[image]
[AutoConnectWithFSParametersAndCustomIP_Test3.ino.txt](https://github.com/tzapu/WiFiManager/files/7533136/AutoConnectWithFSParametersAndCustomIP_Test3.ino.txt)

cds333, 2021-11-14:
Ok I think I get it now... by using setValue on one of the custom parameter objects, it automatically populates the field. Great; got that part working now! Thank you.

Regarding your question about the the file system, I am still using SPIFFS from the example code.

Whoops- I was incorrect; I didn't notice that the serial output was showing the values I saved previously, before I added the setParamsPage(true) line.

So when the params were together on the wifi page, and I entered ssid+password+params, it saves all of them to the json. However when I have the params on the separate page, clicking save does not save them to json.

If however, I go back after clicking save on param page, and then go back to wifi page, and enter ssid and password, and save again, it saves both the wifi creds and the custom params.

So the save of the custom params to the json is still dependent on the ssid+password being entered and saved :(

cds333, 2021-11-14:
Whoops- I was incorrect; I didn't notice that the serial output was showing the values I saved previously, before I added the setParamsPage(true) line.

So when the params were together on the wifi page, and I entered ssid+password+params, it saves all of them to the json. However when I have the params on the separate page, clicking save does not save them to json.

If however, I go back after clicking save on param page, and then go back to wifi page, and enter ssid and password, and save again, it saves both the wifi creds and the custom params.

So the save of the custom params to the json is still dependent on the ssid+password being entered and saved :(

cds333, 2021-11-14:
OK! Now it is starting to make sense. 

The main problem I think is that I was trying to use the example code which was written for a single instance of the configportal, so trying to make it on-demand as well, caused some serious problems. 

I think I managed to get it working pretty well. 

-Added a second callback function for _wifiManager.setSaveParamsCallback()_ that also closes the portal (because setBreakAfterConfig still waits for the timeout for some reason)

-Added a .setValue call for each _custom__ WiFiManagerParameter, to set the saved value I got from the json

-Moved _if(shouldSaveConfig){}_ block down to loop()

cds333, 2021-11-14:
OK! Now it is starting to make sense. 

The main problem I think is that I was trying to use the example code which was written for a single instance of the configportal, so trying to make it on-demand as well, caused some serious problems. 

I think I managed to get it working pretty well. It now allows re-opening the portal as many times as you want; it populates the fields with the data from the json, and it saves the parameters independently, immediately upon clicking save.

Changes:

-Added a second callback function for wifiManager.setSaveParamsCallback() that also calls stopConfigPortal() to close the portal (because _setBreakAfterConfig(true)_ still waits for the portal to timeout for some reason)

-Added a .setValue call for each _custom__ WiFiManagerParameter, to set the saved value I got from the json

-Moved _if(shouldSaveConfig){}_ block down to loop()

[AutoConnectOnDemandWithFSParametersAndCustomIP_v1.ino.txt](https://github.com/tzapu/WiFiManager/files/7533268/AutoConnectOnDemandWithFSParametersAndCustomIP_v1.ino.txt)

closed 2022-01-22 by tablatronix