tzapu/WiFiManager #1313: How to use on PlatformIO and ESP32

marcolino7, 2021-11-16:
Just one quick question, to use on PlatformIO the esp32, I have to use branch esp32s2

Thanks

tablatronix, 2021-12-02:
it should all be in master now

since pio wont pull in beta releases you will probably have to add lib from github

Sulomus, 2022-01-28:
I am facing now the same issue with error code:
`File "ESP8266WiFi.h" can't be open . (dependency of "WiFiManager.h")`

Did you somehow resolve the issue?

Kakomis, 2022-03-02:
I get this message .pio\libdeps\nodemcu-32s\WiFiManager\WiFiManager.h:92:20: fatal error: Update.h: No such file or directory

tablatronix, 2022-03-02:
hmm, That file is in the esp32 framework.. not sure maybe a lib_ldf mode issue?

Kakomis, 2022-03-02:
Yes, you're right, lib_ldf_mode = deep+ fixed it.

tablatronix, 2022-03-02:
I have had to do that alot recently and I am not sure exactly why..

closed 2022-03-26 by tablatronix