tzapu/WiFiManager #1314: First 30 seconds in AP mode

GabrielLAPU, 2021-11-23:
Good morning everyone,

First of all, congratulations for the library, it is one of the most complete and useful I have seen.

I am doing a project using an ESP32 Devkit V4 microprocessor, with which I intend to control some lamps. For the connectivity management I want to use the library with the following working mode:

When the ESP32 is started during the first 30 seconds we enter a configuration mode leaving the micro in access point mode, in case nobody enters the captive portal, the module should try to connect with the last credentials with which it was connected to internet the ESP. 

I have thought about spending this code but I do not know if it is possible to do it like that, or should I save the credentials for example in SPIFF and if no one connects, do a reading of the SSID and the password and connect with them. 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

void setup() {
Serial.begin(9600);
WiFiManager wifiManager;                                //Local intialization. Once its business is done, there is no need to keep it around
wifiManager.setSaveConfigCallback(saveConfigCallback);  //set config save notify callback
wifiManager.setTimeout(30);
wifiManager.autoConnect("AP_ESP32", "12345678")
}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

GabrielLAPU, 2021-11-29:
Hi tablatronix, 
Thanks for your reply. A question about the setConfigPortalTimeout (40) command. This command starts the captive pool for 40 seconds, if no one enters it, does it connect to the Wi-Fi with the last credentials it had? or is it necessary to include some other command to force this connection with those credentials? 
Yhank you in advance!

tablatronix, 2021-11-29:
you mean if autoconnect fails? There is no automatic retry after timeout, other than autoreconnects that the esp library might do

GabrielLAPU, 2021-11-30:
Hi tablatronix, Thank you for your replay. The operation mode that I would like implement in the device is the next, when I switch on the device, during a little bit time (maybe 30 seconds is really poor) I need start with a configuration mode,

I need to start with a configuration mode, allowing to change the Wi-Fi to the I connect, if there is no Wi-Fi change I should try to connect to the Wi-Fi with the last credentials that I did last time, if I cannot connect to that Wi-Fi I will detect it entering an Off-line mode of operation.

Do you still recommend using the setConfigPortalTimeout function?

GabrielLAPU, 2021-12-01:
Hi, I finally seem to have something that works.
Indeed tablatronix use the startConfigPortal function configuring the calls setAPClientCheck (true) and setConfigPortalTimeout, after this I use an autoconnect in case the wifi has not been modified to reconnect with the old credentials. 

Thanks for your support.

closed 2022-01-22 by tablatronix