tzapu/WiFiManager #1320: Using WifiManager with NTPClient

Abhesheksh, 2021-12-15:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32:  ESP8266 / NodeMCU

Hardware:			ESP-12E

Core Version:      	2.4.0, staging

### Description

Using NTPClient to get RTC doesnt work with wifimanager. Same code works when connecting with ssid and password after removing WM.

Problem description

When using NTPClient with WM, time starts from 0:0 and then refreshes but doesn't show actual time. 

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp
#BEGIN

#include <Ticker.h>
#include <ESP8266WiFi.h>
#include "ESP8266Ping.h"
#include <WiFiManager.h>
//#include <EEPROM.h>
#include <EasyButton.h>
#include <NTPClient.h>
#include <WiFiUdp.h>

Ticker ticker;

#define MOSFET 4
#define SWITCH 0
#define DETECT A0
#define LED 2
#define GLED 5
#define DEBUG True

// Update these with values suitable for your network.
WiFiManager wifiManager;

EasyButton button(SWITCH);

// Define NTP Client to get time
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "pool.ntp.org");

//WiFiClient espClient;
bool isButtonPressed = false;
void onPressedForDuration();
void led_blink();
void led_blink2();

void tick()
{
  //toggle state
  int state = digitalRead(LED);  // get the current state of GPIO1 pin
  digitalWrite(LED, !state);     // set pin to the opposite state
}

void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
  //entered config mode, make led toggle faster
  ticker.attach(0.5, tick);
}

void setup() {

  pinMode(SWITCH, INPUT);     // Initialize the GPIO2 pin as an output
  pinMode(MOSFET, OUTPUT);     // Initialize the GPIO0 pin as an output
  pinMode(DETECT, INPUT);
  pinMode(LED, OUTPUT);
  pinMode(GLED, OUTPUT);

  ticker.attach(0.1, tick);

  digitalWrite(MOSFET, LOW);   // Turn MOSFET ON (Note that LOW or HIGH is the voltage level
  //digitalWrite(LED, HIGH);
  digitalWrite(GLED, HIGH);
  button.begin();
  button.onPressedFor(5000, onPressedForDuration);

  wifiManager.setAPCallback(configModeCallback);

#ifdef DEBUG
  Serial.begin(115200);
#endif

  //wifiManager.resetSettings();
  WiFi.mode(WIFI_STA);

  wifiManager.setAPStaticIPConfig(IPAddress(192, 168, 1, 19), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0));

  // fetches ssid and pass from eeprom and tries to connect
  // if it does not connect it starts an access point with the specified name
  // here  "AutoConnectAP"
  // and goes into a blocking loop awaiting configuration
  wifiManager.setConfigPortalTimeout(900);
  wifiManager.setConnectTimeout(90000);
  wifiManager.setWiFiAPHidden(false);
  wifiManager.setSTAStaticIPConfig(IPAddress(192, 168, 1, 99), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0));
  wifiManager.autoConnect("TORLAJE_AP");

  if (wifiManager.autoConnect() == true)
  {
    Serial.println("OK CONNECT");
    ticker.detach();
    digitalWrite(LED, LOW);

  }
  else
  {
    Serial.println("FALSE - NO CONNECT");
   
  }

  // if you get here you have connected to the WiFi
  Serial.println("Connected.");
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());
  
  delay(200);
  timeClient.begin();
}

void loop() {

  button.read();
  timeClient.update();

  
   Serial.print(timeClient.getHours());
   Serial.print(":");
   Serial.println(timeClient.getMinutes());
 

}

void onPressedForDuration()
{
  isButtonPressed = true;
  wifiManager.resetSettings();
  ESP.eraseConfig();
  Serial.println("BUTTON PRESSED.");
  ESP.reset();
  led_blink();
  Serial.println("Button pressed");
}

void led_blink()
{
  int i = 0;
  while (i < 3)
  {
    digitalWrite(LED, HIGH);
    delay(500);
    digitalWrite(LED, LOW);
    delay(500);
    digitalWrite(LED, HIGH);
    i = i + 1;
  }
}
void led_blink2()
{
  int i = 0;
  while (i < 2)
  {
    digitalWrite(LED, LOW);
    delay(50);
    digitalWrite(LED, HIGH);
    delay(50);
    digitalWrite(LED, LOW);
    i = i + 1;
  }
}

#END
```

### Debug Messages
⸮⸮⸮bp⸮⸮c$`r$p⸮n⸮⸮l⸮⸮"n⸮|⸮$⸮⸮⸮c⸮|~⸮N⸮l⸮⸮l ⸮ond`nr⸮⸮⸮obl s⸮⸮Nbl ⸮#⸮⸮$⸮⸮r⸮`⸮⸮o⸮*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP: 192.168.1.99
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom STA IP/GW/Subnet 
*WM: [1] STA IP set: 192.168.1.99
*WM: [1] Connecting to SAVED AP: JHKAHKJAH
*WM: [3] Using Password: JAKHAKAH
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [2] 90000000 ms connectTimeout set
*WM: [2] 19072 ms timeout, waiting for connect...
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.99
*WM: [1] AutoConnect 
*WM: [1] AutoConnect: ESP Already Connected 
*WM: [3] STA static IP: 192.168.1.99
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom STA IP/GW/Subnet 
*WM: [1] STA IP set: 192.168.1.99
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.99
OK CONNECT
Connected.
IP address:	192.168.1.99
0:0
0:0
0:0
0:0
0:0
0:0
0:0
0:0
0:0
0:0
0:0

```
messages here
```

Abhesheksh, 2021-12-16:
https://github.com/tzapu/WiFiManager/issues/1097

Same issue but he solved it by removing static IP. Is there anyway I can keep the static IP?

tablatronix, 2022-02-03:
Not sure why this would make a difference, it should still work

Georgie9117, 2022-11-01:
Just leave `wifiManager.setSTAStaticIPConfig` and use this after `wifiManager.autoConnect`

```
WiFi.begin();
WiFi.config(IPAddress(192, 168, 1, 148), IPAddress(192, 168, 1, 1), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0));
```

It working for me on ESP8266.

samuraiyy, 2023-07-17:
This is my method，it works：

     //wm.setSTAStaticIPConfig(IPAddress(192,168,2,123), IPAddress(192,168,2,1), IPAddress(255,255,255,0) ); // set static ip,gw,sn  
     wm.setShowStaticFields(true);
     wm.setShowDnsFields(true);  

or

     wm.setSTAStaticIPConfig(IPAddress(192,168,2,123), IPAddress(192,168,2,1), IPAddress(255,255,255,0), IPAddress(192,168,2,1) ); // set static ip,gw,sn  
     wm.setShowStaticFields(true);
     wm.setShowDnsFields(true);

ChumKiu, 2025-02-07:
> This is my method，it works：
>
> ```
>  //wm.setSTAStaticIPConfig(IPAddress(192,168,2,123), IPAddress(192,168,2,1), IPAddress(255,255,255,0) ); // set static ip,gw,sn
>  wm.setShowStaticFields(true); [...]

How set static IP into the first way you write?
In the second one it is specified but not into the first one.
Could you please clarify?
Thanks.
BR

ChumKiu, 2025-02-07:
> Just leave `wifiManager.setSTAStaticIPConfig` and use this after `wifiManager.autoConnect`
> 
> ```
> WiFi.begin();
> WiFi.config(IPAddress(192, 168, 1, 148), IPAddress(192, 168, 1, 1), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0));
> ```
> 
> It working for me on ESP8266.

This working form me on a ESP32, here below my code snippet into the setup() function:

`WiFiManager wm;

  // Set static IP (also visible into the AP page)
  //wm.setSTAStaticIPConfig(IPAddress(192,168,2,199), IPAddress(192,168,2,1), IPAddress(255,255,255,0)); // set static ip,gw,sn
  //wm.setShowStaticFields(true); // force show static ip fields
  //wm.setShowDnsFields(true);    // force show dns field always

  bool res;
  res = wm.autoConnect("AutoConnectAP","MyPassword");

  WiFi.begin();
  WiFi.config(IPAddress(192, 168, 2, 199), IPAddress(192, 168, 2, 1), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));
  
  if(!res) {
      wm.resetSettings();
      delay(1000);
      // AP connection failed
      Serial.println("(Start) Failed to connect to the AutoConnectAP AP!");
      Serial.println("(Start) System will be reset...");
      delay(3000);
      ESP.restart();
      delay(5000);
  } 
  else {
      // WiFi connection with saved credential was succesfully
      Serial.print("(Start) Connected to: ");
      Serial.println(WiFi.localIP());

  }

  // --------------------------------------------------
  // NTP settings
  // --------------------------------------------------

  sntp_servermode_dhcp(1);    // (optional)
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" CONNECTED");
  sntp_set_time_sync_notification_cb(timeavailable);  // set notification call-back function
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer1, ntpServer2);`

tablatronix, 2025-02-08:
This is literally in one of the examples and works

closed 2025-02-08 by tablatronix

ChumKiu, 2025-02-08:
> This is literally in one of the examples and works
Hi, I've see the **AutoConnectWithStaticIP.ino** example, but the sequence of the instruction is different, the **setSTAStaticIPConfig()** method is before the **autoConnect()** call and this is not working for me, the only way is to put the instructions sequence:

`
WiFi.begin();`
`
WiFi.config(IPAddress(192, 168, 2, 199), IPAddress(192, 168, 2, 1), IPAddress(192, 168, 2, 1), IPAddress(255, 255, 255, 0));
`

after the **autoConnect()** method call. In this way looking at the log start i can see for a moment the Dinamic IP set through the DHCP immediately followed by the static IP assignement, but this is not a problem for me, here a shot of the output log:

`Starting`
`*wm:AutoConnect`
`*wm:Connecting to SAVED AP: ITALSENSOR-BASE`
`*wm:connectTimeout not set, ESP waitForConnectResult... `
`*wm:AutoConnect: SUCCESS `
`*wm:STA IP Address: 192.168.2.198`
`(Start) Connected to: 192.168.2.199`

the (Start) means my output, the other line is from the library. You can see that after the AutoConnect I've the dynamic IP ...198 followed by the static one ...199.

There is other specific example about to set the static IP?

Thanks and best regards.
ChumKiu

tablatronix, 2025-02-08:
Are you sure this isnt esp autoconnect?
Have you tested without calling wm at all ?

ChumKiu, 2025-02-10:
> Are you sure this isnt esp autoconnect? Have you tested without calling wm at all ?
> 
> or add wifi mode off

Hi, the first IP assign is about DHCP because without static IP the one was with ...198, my static IP is ...199 and this one was also assigned through the WiFi Manager AP. The drawback of this solution is that when I'm configuring the device through the AP is not possible to set also the static IP through the text box on the AP portal (this was possible when the static IP assigned was made before the autoConnect() method, but in this case the NTP is not working. 

So actually the only solution to get WiFiManager work with static IP and NTP is to place the static IP settings after the autoConnect() method, but in this way I'm unable to setup the IP address from the AP web form and the only solution is to get the one that is hardcoded in the way that I've show above.

Here below the log.

```
 Starting
*wm:AutoConnect 
*wm:No wifi saved, skipping 
*wm:AutoConnect: FAILED for  1 ms
*wm:StartAP with SSID:  AutoConnectAP
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal 
*wm:18 networks found
*wm:Connecting to NEW AP: ITALSENSOR-BASE
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:Connect to new AP [SUCCESS] 
*wm:Got IP Address: 
*wm:192.168.2.198 
*wm:config portal exiting 
(Start) Connected to: 192.168.2.199
```

BR
ChumKiu