tzapu/WiFiManager #1321: Show confirmation page with success or error

bwjohns4, 2021-12-18:
How can I show a confirmation page that connection was successful OR what the error was, then allow the user to close WM or the captive portal close automatically after a brief timeout? It looks like handleWifiSave is called when the credentials are submitted and that’s the last page. Then WM tries to connect, so only after then could it know the status but the configPortal is closed immediately and the client has no way to be triggered to reach out to the server after connect for status on the connection.

bwjohns4, 2021-12-19:
I ended up modifying processConfigPortal() so that it does connect upon save but does not shutdown the portal immediately but instead flags itself as connected and waits 30s to close while still allowing the server to respond to requests. 

Then I added some JavaScript in my response sent to client from handleWifiSave that shows "Loading" until the JavaScript request returns all the .innerHTML describing the success or failure.

dronecz, 2022-03-22:
@bwjohns4 I would like to implement similar functionality, could you pls share your changes or your changed library/files? 

Many thanks