tzapu/WiFiManager #1327: Adding the option to start/keep running the portals in the background after connecting

axlan, 2021-12-26:
This is a really simple feature request. I'd like to use this project both for it's Wifi management, but also as a general configuration page. To do this I run the class in non-blocking mode, and I want the config portal to keep running even after the WiFi connection completes successfully. I've made a simple PR to support this for my own use, but you may be interested in merging it upstream.

axlan, 2021-12-28:
I actually built this out a bit with a full example of how I'd be using it:
https://github.com/axlan/extended-wifimanager

tablatronix, 2021-12-28:
This feature is not needed due to non blocking mode already existing

axlan, 2021-12-28:
> This feature is not needed due to non blocking mode already existing, ondemand webportal existing, and this would only work on enableconfigportal and autoconnect failure. So I do not see the utility here, as opposed to manually starting configportal or webportal manually.
> 
> I have other requests for leaving cp open after save, which makes a bit more sense to me.
> 
> Maybe I am misunderstanding this entirely?

I initially hit this before I realized that the settings could be shown on their own page. In that usage mode saving the settings closes the web portal. It was also hard to enter settings in that mode since they would clear the WiFi credentials.

In non-blocking mode with the settings on their own page, I don't think this issue is relevant. I guess you can consider changing the behavior for the case with non-blocking mode, but the settings share the WiFi config page.

tablatronix, 2021-12-28:
Both of these should not happen.
- [ ] Empty creds should not wipe anything
- [ ] nonblocking should not stopconfigportal on saves

tablatronix, 2022-01-20:
do you happen to have logs of this ?

tablatronix, 2022-01-22:
I could not reproduce saves breaking saved creds

axlan, 2022-01-22:
I'm using the master commit: `810f144`

It looks like the main issue (clearing the credentials on an empty save) is fixed.

The configuration portal still closes on connection, which is what I originally proposed a setting to disable.

Here are the logs I captured in testing.

Here's an example main.cc I'm using for reference:

```cpp
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
WiFiManager wm;
WiFiManagerParameter custom_field; // global param ( for non blocking w params )

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    // test custom html(radio)
    const char* custom_radio_str = "<br/><label for='customfieldid'>Custom Field Label</label><input type='radio' name='customfieldid' value='1' checked> One<br><input type='radio' name='customfieldid' value='2'> Two<br><input type='radio' name='customfieldid' value='3'> Three";
    new (&custom_field) WiFiManagerParameter(custom_radio_str); // custom html input
    
    wm.addParameter(&custom_field);

    wm.setConfigPortalBlocking(false);
    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    if(wm.autoConnect("AutoConnectAP")){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }

    wm.startConfigPortal();
}

void loop() {
    wm.process();
}
```

# Issue 1: The web configuration closes on initial connection:

```
*wm:[1] AutoConnect 
*wm:[2] Connecting as wifi client...
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to SAVED AP: XXXX
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED
*wm:[2] Starting Config Portal
*wm:[2] Disabling STA
*wm:[2] Enabling AP
*wm:[1] StartAP with SSID:  AutoConnectAP
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal
*wm:[2] HTTP server started
*wm:[2] Config Portal Running, non blocking/processing
Configportal running
*wm:[2] Starting Config Portal
*wm:[2] Disabling STA
*wm:[2] Enabling AP
*wm:[1] StartAP with SSID:  ESP_DE715E
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal
*wm:[2] HTTP server started
*wm:[2] Config Portal Running, non blocking/processing 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started
*wm:[2] WiFi Scan completed in 2186 ms
*wm:[1] 6 networks found
*wm:[2] DUP AP: XXXX
*wm:[2] AP:  -64 XXXX
*wm:[2] AP:  -69 ATTj8CKQpi
*wm:[2] AP:  -70 ESP_D3B0F4
*wm:[2] AP:  -81 DU-EERO-LR
*wm:[2] AP:  -84 2 Live
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] NUM CLIENTS:  1
*wm:[2] <- HTTP WiFi save  
*wm:[2] Parameters
*wm:[2] --------------------
*wm:[2] :
*wm:[2] --------------------
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to NEW AP: XXXX
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS]
*wm:[1] Got IP Address:
*wm:[1] 192.168.1.141
*wm:[2] shutdownConfigPortal
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed
```

# Issue 2: The web configuration is cleared when only setting the user parameters:

This appears to be fixed in this commit!

```
*wm:[1] AutoConnect 
*wm:[2] Connecting as wifi client...
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to SAVED AP: XXXX
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS
*wm:[1] STA IP Address: 192.168.1.141
connected...yeey :)
*wm:[2] Starting Config Portal
*wm:[2] Enabling AP
*wm:[1] StartAP with SSID:  ESP_DE715E
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal
*wm:[2] HTTP server started
*wm:[2] Config Portal Running, non blocking/processing
*wm:[2] <- HTTP Root 
*wm:[2] NUM CLIENTS:  0
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started
*wm:[2] WiFi Scan completed in 1586 ms
*wm:[1] 7 networks found
*wm:[2] DUP AP: XXXX
*wm:[2] AP:  -63 XXXX
*wm:[2] AP:  -67 ESP_D3B0F4
*wm:[2] AP:  -67 ATTj8CKQpi
*wm:[2] AP:  -70 XXXX
*wm:[2] AP:  -75 DU-EERO-LR Guest
*wm:[2] AP:  -84 2 Live
*wm:[2] <- HTTP WiFi save  
*wm:[2] Parameters
*wm:[2] --------------------
*wm:[2] :
*wm:[2] -------------------- 
*wm:[2] processing save 
*wm:[2] No ssid, skipping wifi save 
*wm:[2] Portal is non blocking - remaining open
```

tablatronix, 2022-01-22:
Yeah but the first condition only occurs if conx fails so you have to manually do it anyway right. Also how can you keep it open if you are saving new wifi, it has to connect to the ap so it has to close softap

tablatronix, 2022-01-22:
I still cannot understand your pr to test as it has code style changes diffs

axlan, 2022-01-22:
Sorry, the PR pulled in further changes I made to that branch for my own application. I created a new issue for the other bug I found, and I'll close it since you already addressed the main issue.

The remaining issue can be summarized as: "When running in the background, I'd like the configuration portal (the web server) to be available regardless of changes to connectivity". Since I'm using the portal for general configuration it avoids needing to restart the board after the initial Wifi configuration. I first access the configuration page through the captive portal, and after the initial configuration it can be accessed at the IP address reservation.

For the PR I had made, it seemed sufficient to just modify the logic not to close the configuration portal. I admit I'm not super familiar with how the web servers are managed, so it's possible to get this to work across different platforms the server might need to be stopped and restarted.

This is admittedly not the main use case of the Wifi manager, so it's understandable if you don't want this behavior.

tablatronix, 2022-01-22:
Well you probably want to be using startWebPortal()
and have proper control over it and not rely on autoconnect other than actual wifi failure recovery

axlan, 2022-01-22:
Sure, though I don't think autoconnect is the issue here. It's still an issue with this example:

```cpp
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);

    wm.setConfigPortalBlocking(false);

    wm.startWebPortal();
    wm.startConfigPortal();
}

void loop() {
    wm.process();
}
```

I suppose the alternative to the feature I'm requesting would be something like:

```cpp
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);

    wm.setConfigPortalBlocking(false);

    wm.startWebPortal();
    wm.startConfigPortal();
}

void loop() {
    wm.process();
    // For a real example would probably want to add a delay between calls to this check
    if (!wm.getConfigPortalActive()) {
      wm.startConfigPortal();
    }
}
```

tablatronix, 2022-01-22:
wm.startConfigPortal(); includes webportal

axlan, 2022-01-22:
No, I want the webserver running all the time regardless of the Wifi status or changes to the Wifi. That way I can always access the settings and update the settings either though the IP or the AP.

axlan, 2022-01-22:
My application has a screen that indicates how you should connect to it, so it's obvious to the user which to use.

tablatronix, 2022-01-22:
Yeah thats how i use it and why non blocking was created.

If you are connected to wifi just use startWebPortal(), you don't want the ap running

tablatronix, 2022-01-22:
check this example

https://github.com/tzapu/WiFiManager/blob/master/examples/NonBlocking/OnDemandNonBlocking/onDemandNonBlocking.ino

See how the startap option works

axlan, 2022-01-22:
I do want the AP to run for the initial setup. The transition from AP to STA is the case I'm trying to handle here. I think I had `startWebPortal` and `startConfigPortal` confused in my previous example, but it still seems like:

```cpp
void loop() {
    wm.process();
    // For a real example would probably want to add a delay between calls to this check
    if (!wm.getWebPortalActive()) {
      wm.startWebPortal();
    }
}
```

would be needed to accomplish this. I could also probably handle similar logic in the callbacks, but that's the basic behavior.  I was able to avoid this with an additional setting.

It seems like this is what you're suggesting with your example, except it's triggered by a specific event, and not just running all the time.

tablatronix, 2022-01-22:
oh yeah its an ondemand example, but you would just change that to if connected or something else

tablatronix, 2022-01-22:
crap, I didn't click comment and lost previous post, feel free to hit up discord ill check it later, I am also working on something atm that will be doing this same behavior with smarter ap starting to deal with power failures and sleep.

I have no problem with a single toggle , but I need to make sure it handles all failure modes and when saving we can keep the http and dns without having to restart it, i highly doubt this will work ok all the time as its a bit finicky.

tablatronix, 2022-02-03:
Sorry I was working on some other stuff, arduino 2.0 testing etc

tavdog, 2022-08-24:
@tablatronix I know this is off topic but the discord link in the main readme has expired.

tablatronix, 2023-09-08:
- [ ] follow up on this

scudrunner, 2023-11-20:
I am joining late, but have a similar use case I am looking at.  I want to be able to easily access saved settings on power up, without overwriting them with the default that is hardcoded in.  I really like the captive portal aspect where the user does not need a website, it just opens.  

I found that: 
The setconfigportal loads the default values.  
the autoconnect goes straight to the saved network, and I can not update values  

I tried the code below from the above post, and it is not saving my parameters when I catch it in the loop.  I also would prefer not to have to know the IP address of the device when connected. 
  
void loop() {
    wm.process();
    // For a real example would probably want to add a delay between calls to this check
    if (!wm.getWebPortalActive()) {
      wm.startWebPortal();
    }
}

scudrunner, 2023-11-20:
I updated the code to read the JSON, and change the defaults to previously saved values.  I am still lost on how to force a connection later without a button.  I would like a period of time on initial boot that I can access the parameters, and after that period of time, the main wifi will be joined.  since there is no real trigger, having it in loop seemed very clumsy - but kind of worked.  I was re-writing the JSON each loop to make sure it was saved, and this just seemed to sloppy and a waste

tablatronix, 2023-11-20:
Do you want to access via the website or the ap?
You can add a configportal timeout and startconfigportal, and when it closes you can autoconnect. Not sure exactly what you are looking for

tablatronix, 2023-11-21:
Yeah thats literally how this works, not sure why its clearing your credentials unless you are doing it in your code..

scudrunner, 2023-11-21:
I will try a simpler version of my code tomorrow.   To be clear - what
example should I follow?   I did have lots of code snippets running around
as got it all working, but I was being careful on this one.

I did find out how to read the json, and overwrite the default values on
boot up, and that may be a nice add as an example in case someone else
wants it.  It ended up being easy enough once I went through the code, but
an example may help others.

 Thank you for the response, it is definitely appreciated

tablatronix, 2023-11-21:
Yeah there is a spiffs/little fs example for storing params.

https://github.com/tzapu/WiFiManager/blob/master/examples/Super/OnDemandConfigPortal/OnDemandConfigPortal.ino

```
// TEST OPTION FLAGS
bool TEST_CP         = false; // always start the configportal, even if ap found
int  TESP_CP_TIMEOUT = 90; // test cp timeout
```

scudrunner, 2023-11-21:
I tried with the CPTEST set to true and false, and did not get the access
point to come up either time.

I tried running the example you pointed to and get the following errors.
Running on a 8266 wemos D1 clone

C:\Users...sketch_nov21a\sketch_nov21a.ino:92:21: error: 'void
WiFiManager::handleNotFound()' is private within this context
   92 |   wm.handleNotFound();
      |                     ^
In file included from C:\Users....sketch_nov21a\sketch_nov21a.ino:6:
c:\Users\....Arduino\libraries\WiFiManager/WiFiManager.h:655:19: note:
declared private here
  655 |     void          handleNotFound();
      |                   ^~~~~~~~~~~~~~
C:\Users\..sketch_nov21a\sketch_nov21a.ino: In function 'void setup()':
C:\Users..Temp\.arduinoIDE-unsaved20231021-19320-7kuv5w.zvso9\sketch_nov21a\sketch_nov21a.ino:125:27:
error: 'WM_DEBUG_DEV' was not declared in this scope; did you mean
'WM_DEBUG_LEVEL'?
  125 |   wm.setDebugOutput(true, WM_DEBUG_DEV);
      |                           ^~~~~~~~~~~~
      |                           WM_DEBUG_LEVEL

exit status 1

tablatronix, 2023-11-21:
Ah you do not have the latest git version, are you using arduino or platformio?
This was updated

scudrunner, 2023-11-21:
Arduino, and I just loaded it last week.  I gave up on platform io, it had
many repositories i was trying to use that would not load properly.

scudrunner, 2023-11-21:
the library manager in arduino is showing 2.0.16-rc.2

scudrunner, 2023-11-21:
in my library folder, the readme has the link to 2.0.16-rc.2, and the JSON
file also has the same version.

tablatronix, 2023-11-21:
Yeah you would need the git dev version

scudrunner, 2023-11-21:
I see you posted in 2018 with a link for the dev branch at that time, but I
dont know how to get the current Dev Branch....

tablatronix, 2023-11-21:
I think there is a guide here somewhere to load dev version git version of arduino libraries manually