tzapu/WiFiManager #1329: wifi: addba response cb: ap bss deleted

Bonv1Arnaud, 2021-12-27:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32:       ESP32-WROOM-32UE

Core Version:           I don't know (I think the newest version)

### Description

Problem description : 
I upload the example sketch at home. And I was able to connect the ESP on my WiFi. 
And now I am in my office and I try with the same board and the same code. I have this error (just bellow)

### Settings in IDE

Module:  			????

Additional libraries: no

### Sketch

```cpp
#BEGIN
  #include <WiFi.h>                                  //Librairie pour la connexion à un wifi
  #include <WiFiManager.h>                   //Ajout de la librairie pour le WiFi

  WiFiManager wifiManager;                    //Déclaratoin de la class pour le wifi manager

void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  delay(1000);
  wifiManager.resetSettings();

  Serial.println("Creation du AP oblo...");
  wifiManager.autoConnect("AutoConnectAP");

}

void loop() {
  // put your main code here, to run repeatedly:

  Serial.print("Adresse IP: ");
  Serial.println(WiFi.macAddress());
  delay(1000);

} 
#END
```

### Debug Messages
15:32:00.933 -> *wm:[2] Starting Config Portal 
15:32:00.933 -> *wm:[2] AccessPoint set password is VALID 
15:32:00.933 -> *wm:[2] Disabling STA 
15:32:00.933 -> *wm:[2] Enabling AP 
15:32:00.933 -> *wm:[1] StartAP with SSID:  Oblo_casa
15:32:01.490 -> *wm:[1] AP IP address: 192.168.4.1
15:32:01.490 -> *wm:[1] Starting Web Portal 
15:32:01.490 -> *wm:[2] HTTP server started 
15:32:01.490 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
15:32:30.629 -> *wm:[2] NUM CLIENTS:  0
15:32:42.884 -> E (42437) wifi:addba response cb: ap bss deleted
15:32:45.974 -> E (45537) wifi:addba response cb: ap bss deleted
15:32:46.488 -> E (46061) wifi:addba response cb: ap bss deleted
15:32:47.003 -> E (46584) wifi:addba response cb: ap bss deleted
15:32:48.079 -> E (47630) wifi:addba response cb: ap bss deleted
15:32:48.592 -> E (48154) wifi:addba response cb: ap bss deleted
15:32:50.367 -> E (49956) wifi:addba response cb: ap bss deleted
15:32:51.954 -> E (51526) wifi:addba response cb: ap bss deleted
15:32:53.500 -> E (53096) wifi:addba response cb: ap bss deleted
15:33:00.657 -> *wm:[2] NUM CLIENTS:  0 

```
messages here
```

tablatronix, 2021-12-27:
No idea, sounds like an esp issue, are there alot of APs around ?

Bonv1Arnaud, 2021-12-27:
My office is in a town so I have 5-6 APs around. 

To be sure that the problem was not with the ESP, I tried with three boards ( 2 x ESP32-WROOM32UE & 1 x ESP-DEVKIT.

I have also check the WiFi channel and my esp is in channel 3 and all APs around are in the channel 6 and more

tablatronix, 2021-12-27:
The ESP lib

Bonv1Arnaud, 2021-12-27:
[image]
the APs around

Bonv1Arnaud, 2021-12-27:
> The ESP lib.
> 
> Power supply issue?

The ESP is powered with a battery. And during the test the board was powered also with my computer (trough an USB cable)

WiFi manager library : https://github.com/tzapu/WiFiManager
I didn't see the same trouble on the "issues" section of this github

tablatronix, 2021-12-27:
https://github.com/espressif/arduino-esp32/issues/4052

https://www.esp32.com/viewtopic.php?t=8191

Looks like there was never any solution, I would do a full erase of flash, make sure your esp lib is up to date.

closed 2022-02-03 by tablatronix

kevin4dhd, 2024-02-04:
> This problem is very strange. I repeatedly burn other programs, and then turn on AP mode. This problem disappears....

Can you give more details about your solution please, I am also having the same problem