tzapu/WiFiManager #1333: Not connected with .autoConnect

SynonZelra, 2022-01-07:
**Sorry I'm not completely sure if this is the right place to ask.
Apologize in advance.**

I bought a brand new esp32 board, and the code is simple
`WiFi_Connection.autoConnect(/*some SSID*/);`
Everything works fine after I entering password.
But every time the board resets, it just won't connect.
Config page shows the board remembers SSID and password, but just not connected.

Tricky part is, this issue never happened on my other old rusty board.

What could possibly go wrong?

izersoft, 2022-01-19:
I have a sampe problem

izersoft, 2022-01-19:
I have a same problem. When I am connecting to original esp9266, showing connection problem.But this error only for ios 15 another devices no any problem

tablatronix, 2022-01-19:
you will have to connect to computer and get serial logs

tablatronix, 2022-01-22:
does it by anychance connect every other reboot?

SynonZelra, 2022-01-27:
Hey.

I just notice from Documentation

> You can and should password protect the configuration access point.

So I tried the change
`WiFi_Connection.autoConnect(/*some SSID, some password*/);`
and suddenly everything works again.

Not quite knowing why, but I'm just happy with that.

Thanks!

closed 2022-01-27 by SynonZelra