tzapu/WiFiManager #1335: Captive Portal Not launching in Android

varunwyest, 2022-01-12:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

The portal is not launching on Android, I have to click on IP address to open the portal. Is the Captive portal supposed to launch automatically.

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2022-02-03:
There is much discussion on this and google dns issues, see esp issues

closed 2022-02-03 by tablatronix

bwjohns4, 2022-02-03:
@tablatronix , do you know of any links to these esp issues? I'd like to read more on this.

tablatronix, 2022-02-03:
https://www.google.com/search?q=github+esp8266+esp32+android+captive+portal+google+dns&ei=R0_8YZS6BtunqtsP0K--0Ag&ved=0ahUKEwjU-bqFw-T1AhXbk2oFHdCXD4oQ4dUDCA4&uact=5&oq=github+esp8266+esp32+android+captive+portal+google+dns&gs_lcp=Cgdnd3Mtd2l6EAM6BwgAEEcQsANKBAhBGABKBAhGGABQsQdYqA9gqBBoAnACeACAAUiIAeYDkgEBN5gBAKABAcgBCMABAQ&sclient=gws-wiz