tzapu/WiFiManager #1348: WiFiManager should make defensive copy of hostname

lptr, 2022-01-26:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  2.0.5-beta
Hardware:			ESP32

### Description

`WiFiManager::setHostname(char*)` should make a defensive copy of parameter value instead of storing the raw parameter value. Currently it is possible to e.g. pass a `hostname` value that gets freed up before `WiFi.setHostname()` etc., resulting in a corrupted hostname being set.

Note: `setCustomHeadElement(char*)` probably suffers from the same problem.

tablatronix, 2022-02-02:
YES, I forgot all about this, but saw that a couple weeks ago

tablatronix, 2022-02-05:
No idea what works best here, ideas? I just changed it to string , it should copy and it seems to work now

closed 2022-04-07 by tablatronix