tzapu/WiFiManager #1350: Can't get the ESP32 working with WifiManager

Sulomus, 2022-01-28:
### Basic Infos
Hello everyone,
I am trying to get `WifiManager `working on ESP-WROOM-32 board with `PlatformIO`
I have looked into several discussions on the internet:
1) https://github.com/tzapu/WiFiManager/issues/1313
2) https://community.platformio.org/t/esp8266wifi-library-not-found-fatal-error/9897
3) https://github.com/tzapu/WiFiManager/issues/1231

I have tried pulling the lib via `github `and adding `lib_ldf_mode = deep+` but with no results.

I have been stuck on this for days now and deadline is coming up ... Any help would be really appreciated! ❤️ 

### Settings in IDE
- Using PlatformIO
- This is my platformio.ini file:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
build_flags = -D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED

lib_ldf_mode = deep+
lib_deps = 
	bblanchon/ArduinoJson@^6.19.1
	mobizt/Firebase Arduino Client Library for ESP8266 and ESP32@^2.8.3
	me-no-dev/AsyncTCP@^1.1.1
	me-no-dev/ESP Async WebServer@^1.2.3
	tzapu/WiFiManager@^0.16.0
        //I have tried importing using just this line also:
        https://github.com/tzapu/WiFiManager
```

### Sketch
```
#include <WiFi.h>
#include <WebServer.h>
#include <WiFiManager.h>

void setup() {
 
    Serial.begin(115200);
     
    WiFiManager manager;    
     
    bool success = manager.autoConnect("ESP32_AP","password");
 
    if(!success) {
        Serial.println("Failed to connect");
    } 
    else {
        Serial.println("Connected");
    }
}
 
void loop(){}
```

### Debug Messages

```
In file included from src/main.cpp:621:0:
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************
```

closed 2022-01-28 by Sulomus

MythicalForce, 2022-03-26:
i have the same problem :S

tablatronix, 2022-03-26:
you have to use the beta versions for esp32

MoJo2600, 2022-03-28:
I got the same error and reduced the imports to only one import and the sample worked for me:

```
#include <WiFiManager.h>
```

But as soon I added the `lib_ldf_mode = deep+` setting to my project, it would not compile anymore. But this time with an error unrelated to WifiManager (`.pio\libdeps\lolin_d32_pro\ezTime\src\ezTime.cpp:10:22: fatal error: EEPROM.h: No such file or directory`).