tzapu/WiFiManager #1353: EXIT button cannot work in non-blocking mode

stacklover, 2022-02-02:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

The EXIT button in the Portal does not work in non blocking mode, because:

void WiFiManager::handleExit() faithfully sets "abort = true",

but "abort" is only checked in the while(1) loop in blocking mode

if one uses non blocking mode it is not checked and there is no method to read this private variable.

I believe it should be handled in the "boolean WiFiManager::process()" function in a similar way it is handled while in blocking mode?

I will try to do that :)

tablatronix, 2022-02-02:
I guess the question should be, should it work ? Do we want it to ?

stacklover, 2022-02-02:
An EXIT button should work the same way in blocking and non-blocking mode, right?
If one could have a callback function to handle it, it should also work the same.
My solution right now is and works how I would expect it to (added stuff between the "++++++":

/**
 * [process description]
 * @access public
 * @return {[type]} [description]
 */
boolean WiFiManager::process(){
    // process mdns, esp32 not required
    #if defined(WM_MDNS) && defined(ESP8266)
    MDNS.update();
    #endif
//+++++++++++++++++++++++
    // check for abort set
    if(abort){
      #ifdef WM_DEBUG_LEVEL
      DEBUG_WM(DEBUG_DEV,F("configportal abort"));
      #endif
      shutdownConfigPortal();
      abort = false; // prevent repetitive calling
    }
//+++++++++++++++++++++++

    if(webPortalActive || (configPortalActive && !_configPortalIsBlocking)){
        uint8_t state = processConfigPortal();
        return state == WL_CONNECTED;
    }
    return false;
}

tablatronix, 2022-02-03:
I mean do you always want to allow an end user to stop portal? 
I might have to add access control mechanism

tablatronix, 2022-02-03:
#982

stacklover, 2022-02-03:
I think, you miss the point: there is an "EXIT" button automatically presented in the portal. And therefore it should work exactly like an end user expects it to.
It is discussable, whether that button should be displayed optionally, aka "setShowExitButton(bool)", but as long as it is shown it should work.
Currently I am happy having that button and with my small patch it is working as I expect it.

Needless to say, after 2-3 days learning curve, the WiFiManager does exactly what I need for my application:
* optional(!) WiFi setup (because WiFi connection is optional)
* optional parameter changing (from factory defaults) in own submenu (setParamsPage(true))
* the portal is only started when a button is pressed (need physical access to start the portal, it never starts on its own)
* WiFi stays off unless its configured
* the application itself continues to run while the portal is open (quite important)

Thanks for this very good project!

tablatronix, 2022-02-03:
The menu is customizable

stacklover, 2022-02-03:
[image]
This is what I get without doing anything special to have that EXIT button!
If you think, that should not be this way, we have to figure out why it is that way ;)

stacklover, 2022-02-03:
ALL I do configure is this:

void wifiman_setup() {
  wifiManager.setDebugOutput(true);
  wifiManager.setSaveParamsCallback(SaveParamsCallback);
  wifiManager.setBreakAfterConfig(true);
  wifiManager.setConfigPortalBlocking(false);
  wifiManager.setShowPassword(true);
  wifiManager.setParamsPage(true);
  wifiManager.setDarkMode(true);
  // Enter all Custom Parameters
  for (PARAMETER *p = parameters; p->name != NULL; p++) {
    p->wpar = new WiFiManagerParameter(p->name, p->name);
    wifiManager.addParameter(p->wpar);
  }

tablatronix, 2022-02-03:
#1354

tablatronix, 2022-02-04:
lol in the code

` * HTTPD CALLBACK exit, closes configportal if blocking, if non blocking undefined`

stacklover, 2022-02-04:
didn't see that, though I had added a printf there to see whether its called at all ;)

commit 4d999b14 (2022-02-04): fix #1353 #1354
closed 2022-02-04 by tablatronix