tzapu/WiFiManager #1356: Question about WiFiManager

renanmbs, 2022-02-12:
Hello, can anyone answer my question?

I want to run the WiFi Manager for 5 minutes, if the person doesn't connect by then, cancel the WiFi Manager.

I am having issues with how to make the WiFi Manager stop and bring the server down.

disconnect() doesn't seem to be doing what I need, so how can I do that?

renanmbs, 2022-02-12:
Also, is this to only stop the portal? Or the server too?

tablatronix, 2022-02-12:
That will stop the entire thing, if you want to run the webportal when on wifi lan, you can use startWebportal() separately..

If you want to layout your exact processm I need to make some new utility examples and this is one I use alot.

eg.
* If no wifi SET run cp
* if no wifi CONX, retry for x seconds, then start cp for x seconds
* Start CP on button press( or toggle )
* Start CP always for timeout on all boot, or use a boot reason filter?
* If button x seconds after boot or held on boot, start x
* If x many reboots start cp
* If x many reboots clearsettings ( tasmota style )
* Button held, clear settings, reboot
*

closed 2022-04-07 by tablatronix