tzapu/WiFiManager #1362: ESP32 wifi manager fix IP + winsock client not working

microicRI, 2022-02-15:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master
1.12.1 ESPAsyncWiFiManager.h

Esp32: Wrover module

Hardware:			ESP32 board 

Core Version:      	2.0.2

### Description

when  wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn); //is active in program
if (client.connect(server_na_koji_se_spaja, 27015)){       // give error
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net

### Settings in IDE
[image]

Module:  			WROOM

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include "SPIFFS.h"
#include <SPI.h>
#include <Wire.h>
#include <ESPmDNS.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

#include <AsyncElegantOTA.h>
//#include <WebSocketClient.h>

// Replace with your network credentials

const char* ssid = "kuci";
const char* password = "kucikuci01";

String remDev = "";
String serPsswd = "";
//String remDev = "LjudevitaMatesica-1";
//String serPsswd = "*Oknardaj1!*";
//const char *ssid = "kuci";     // replace with your wifi ssid and wpa2 key
//const char *pass = "kucikuci01";
//const char *server1 = "jadrand.ddns.net";
//char host[] = "jadrand.ddns.net";
#define LED_BUILTIN 16
#define SENSOR  2

long currentMillis = 0;
long previousMillis = 0;
int interval = 1000;
boolean ledState = LOW;
float calibrationFactor = 4.5;
volatile byte pulseCount;
byte pulse1Sec = 0;
float flowRate;
unsigned long flowMilliLitres;
unsigned int totalMilliLitres;
float flowLitres;
float totalLitres;
[... 461 lines cut ...]
WiFi Connected to: kuci
WiFi Signal strength: -67
WiFi connected with IP: 192.168.1.133
SETUP: uzimanje vrijednosti iz upisanog podatka sa wifi manager stranice 
SETUP: kopirani podaci sa web portala u varijable: 
lokacija_uredjaja = LjudevitaMatesica
broj_uredjaja = 1
server_na_koji_se_spaja = jadrand.ddns.net
password_za_server = *Oknardaj1!*
local ip
192.168.1.133
192.168.1.1
255.255.255.0
DNS= VODA-1
mDNS responder started
Flow rate: 0.00L / min	Output Liquid Quantity: 0mL / 0.00L
server_na_koji_se_spaja= jadrand.ddns.net
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
Not connected to server !
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net
Not connected to server !
[E][WiFiGeneric.cpp:739] hostByName(): DNS Failed for jadrand.ddns.net

```

tablatronix, 2022-02-15:
ill give it a test, if you can make the example any smaller that would be great

microicRI, 2022-02-15:
Hi, unfortunately can not be smaller .
there is no much in program, most of them are functions for wifi manager portal. program itself in loop have one function only,collect data and send data to server.
Thanks,
 Damir