tzapu/WiFiManager #1368: WiFi Manager Reset Settings

renanmbs, 2022-03-05:
Hello, I have a question:

If I don't specify  'wifiManager.resetSettings(); ' the previous wifi information will not be deleted, correct? It will run once, save the information and use it.

How would a person change their wifi settings without changing the code, if they want to? Do I have to create a button that if it is pressed, resets the settings, then reboots the esp?

ChristophCaina, 2022-03-11:
that's how I would do it.
In my case, I have also included a "debug" slide switch - the switch will be checked during setup() - if the switch is set to off, the debug would be set to false; if it is on, it will set debug to true - and also do the wifimanager.resetSettings();...

For user-Reset, I would also include a push-button which will call the resetSettings() - maybe, this can be combined with the reset-pin / reset switch on the chip / dev.board?

tablatronix, 2022-03-11:
There are a few ways, you can have a on demand resetsettings to wipe credentials, or on demand start the portal via a button etc ( examples for this are included )

If you only have a reset button there are some libraries to count resets and do it that way.

Wiping settings is by far the easiest to code.as its just reset and reboot especially if its something that rearely has to be done.

You can also run the webserver all the time in sta mode and have it always available..

closed 2022-04-07 by tablatronix

renanmbs, 2022-04-07:
It worked! Thank you