tzapu/WiFiManager #1371: Example is starting configuration Portal and AP Mode with specific SSID -> Then starting again with auto-generated SSID ?!

ChristophCaina, 2022-03-10:
Hi, I am trying some of your examples provided here and it seems that I run into an issue...
Currently, the following Example Sketch was used: AutoConnectNonBlocking.ino

I am trying to create a configuration portal and a Wifi Network with a specific SSID...
When Running the sketch I can see, that this is working as expected in the Serial Monitor - but then, the code starts to repeate (before the Portal timeout has been reached) and then, it is using a different SSID than given in my code...

Here's the Output from the serial monitor

```
12:23:07.388 -> *wm:[1] SETTINGS ERASED 
12:23:07.388 -> *wm:[1] AutoConnect 
12:23:07.388 -> *wm:[1] No Credentials are Saved, skipping connect 
12:23:07.388 -> *wm:[2] Starting Config Portal 
12:23:07.388 -> *wm:[2] Disabling STA 
12:23:07.388 -> *wm:[2] Enabling AP 
12:23:07.388 -> *wm:[1] StartAP with SSID:  NCC1701D
12:23:07.899 -> *wm:[2] AP has anonymous access! 
12:23:08.385 -> *wm:[1] AP IP address: 192.168.4.1
12:23:08.385 -> *wm:[1] Starting Web Portal 
12:23:08.385 -> *wm:[2] HTTP server started 
12:23:08.385 -> *wm:[2] Config Portal Running, non blocking/processing 
12:23:08.421 -> *wm:[2] Portal Timeout In 60 seconds
12:23:08.421 -> Configportal running
12:23:08.421 -> *wm:[2] Starting Config Portal 
12:23:08.421 -> *wm:[2] Disabling STA 
12:23:08.421 -> *wm:[2] Enabling AP 
12:23:08.421 -> *wm:[1] StartAP with SSID:  ESP_FBAB38
12:23:08.926 -> *wm:[2] AP has anonymous access! 
12:23:09.412 -> *wm:[1] AP IP address: 192.168.4.1
12:23:09.412 -> *wm:[1] Starting Web Portal 
12:23:09.412 -> *wm:[2] HTTP server started 
12:23:09.412 -> *wm:[2] Config Portal Running, non blocking/processing 
12:23:09.412 -> *wm:[2] Portal Timeout In 60 seconds
12:23:37.118 -> *wm:[2] Portal Timeout In 31 seconds
12:24:07.126 -> *wm:[2] Portal Timeout In 1 seconds
12:24:08.396 -> *wm:[1] config portal has timed out 
12:24:08.396 -> *wm:[2] shutdownConfigPortal 
12:24:08.396 -> *wm:[2] restoring usermode STA
12:24:09.398 -> *wm:[2] WiFi Reconnect, was idle 
12:24:09.444 -> *wm:[2] wifi status: WL_IDLE_STATUS
12:24:09.444 -> *wm:[2] wifi mode: STA
12:24:09.444 -> *wm:[2] configportal closed
```

This is my current code (100% from the examples)
```
#include <WiFiManager.h>

const char* wifiAutoSSID = "NCC1701D";

WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //reset settings - wipe credentials for testing
    wm.resetSettings();

    wm.setConfigPortalBlocking(false);
    wm.setConfigPortalTimeout(60);
    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    if(wm.autoConnect(wifiAutoSSID)){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }

    wm.startConfigPortal();
    // wm.startWebPortal();
}

void loop() {
    //wm.process();
    // put your main code here, to run repeatedly:
}
```

I am using an ESP 8266 based WeMos D1 R2 ... if that may help?!

tablatronix, 2022-03-10:
wm.startConfigPortal(wifiAutoSSID);

`

ChristophCaina, 2022-03-11:
Hi, just another question so far:

Can you please explain, why this seems to run in a loop?

```
12:23:07.388 -> *wm:[1] No Credentials are Saved, skipping connect 
12:23:07.388 -> *wm:[2] Starting Config Portal 
12:23:07.388 -> *wm:[2] Disabling STA 
12:23:07.388 -> *wm:[2] Enabling AP 
12:23:07.388 -> *wm:[1] StartAP with SSID:  NCC1701D
12:23:07.899 -> *wm:[2] AP has anonymous access! 
12:23:08.385 -> *wm:[1] AP IP address: 192.168.4.1
12:23:08.385 -> *wm:[1] Starting Web Portal 
12:23:08.385 -> *wm:[2] HTTP server started 
12:23:08.385 -> *wm:[2] Config Portal Running, non blocking/processing 
12:23:08.421 -> *wm:[2] Portal Timeout In 60 seconds
12:23:08.421 -> Configportal running
```
and after that, it is starting again with:
```
12:23:08.421 -> *wm:[2] Starting Config Portal 
12:23:08.421 -> *wm:[2] Disabling STA 
12:23:08.421 -> *wm:[2] Enabling AP 
12:23:08.421 -> *wm:[1] StartAP with SSID:  NCC1701D
12:23:08.926 -> *wm:[2] AP has anonymous access! 
12:23:09.412 -> *wm:[1] AP IP address: 192.168.4.1
12:23:09.412 -> *wm:[1] Starting Web Portal 
12:23:09.412 -> *wm:[2] HTTP server started 
12:23:09.412 -> *wm:[2] Config Portal Running, non blocking/processing 
12:23:09.412 -> *wm:[2] Portal Timeout In 60 seconds
```

The code is still the same as given in the Example sketch I've mentioned above... (with the fix provided in your previous comment)

tablatronix, 2022-03-11:
Yes, Its the logging level

you are calling wm.resetSettings(); on every boot.

ChristophCaina, 2022-03-11:
ok, thanks.
the "reset" was due to current development - and I want to reset the settings on each "reboot" of the board... but the current behave happens without any reboot (at least, I don't see the board rebooting)... 

it's just immediately visible after uploading the code to the board... then it is running throught 1) and immediately the second block happens.

So, is the chip rebooting when it could not connect to an existing network / no network information can be found?

tablatronix, 2022-03-11:
ohh let me check if this is a bug, you have non blocking false but commented out process ? thats wrong

ChristophCaina, 2022-03-11:
process was just commented out for a test when I copied the code :)

tablatronix, 2022-03-11:
There is no other serial output between those? I mean it totally looks like its restarting
add a println starting.... to the setup

ChristophCaina, 2022-03-12:
that's from this code...
I have just used the code given in the example without any further changes...

```
    if(wm.autoConnect(wifiAutoSSID)){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }
 ```

ChristophCaina, 2022-03-12:
Maybe, a bit more information:

As you can see in my first post, it was first starting with the SSID I've set in the wm.autoConnect() - but then printing the serial.println from the "else" statement.
Immediately after that, it "restarted" and used the wm.startConfigPortal(); where I haven't used my SSID - and due to that, it took the default ESP..... whatever SSID.

tablatronix, 2022-03-12:
Oh yeah i ran this and it works fine. So no clue, sounds like your esp has to be rebooting

ChristophCaina, 2022-03-12:
hm... thanks for your test.
Would you be so kind and share your code & serial output, that I can compare it and run another test?
I would really like to understand what my ESP is doing here - and therefore trying to figure out, how I could extend / debug this on my side...

tablatronix, 2022-03-13:
Add a println in setup that outputs millis() it should be obvious is it restarts, could be a brownout all these should show startup code in serial though… so its odd i ran your example exactly

ChristophCaina, 2022-03-13:
hm... sorry... no reboot during the startup:

```
19:44:22.114 -> 69                                               // first millis() println
19:44:22.214 -> *wm:[1] AutoConnect 
19:44:22.214 -> *wm:[1] No Credentials are Saved, skipping connect 
19:44:22.214 -> *wm:[2] Starting Config Portal 
19:44:22.214 -> *wm:[2] AccessPoint set password is VALID 
19:44:22.214 -> *wm:[2] Disabling STA 
19:44:22.214 -> *wm:[2] Enabling AP 
19:44:22.214 -> *wm:[1] StartAP with SSID:  ****                              // This time, I am using an existing SSID with Passwort in the AutoConnect(); part
19:44:24.093 -> *wm:[1] AP IP address: 192.168.4.1
19:44:24.093 -> *wm:[1] Starting Web Portal 
19:44:24.093 -> *wm:[2] HTTP server started 
19:44:24.093 -> *wm:[2] Config Portal Running, non blocking/processing 
19:44:24.093 -> *wm:[2] Portal Timeout In 60 seconds
19:44:24.093 -> Configportal running                                   // Portal is running
19:44:24.093 -> *wm:[2] Starting Config Portal                     // Starting configPortal aain
19:44:24.093 -> *wm:[2] Disabling STA 
19:44:24.093 -> *wm:[2] Enabling AP 
19:44:24.093 -> *wm:[1] StartAP with SSID:  NCC1701D      // using the SSID I've set up in the wm.StartConfigPortal();
19:44:24.595 -> *wm:[2] AP has anonymous access! 
19:44:25.114 -> *wm:[1] AP IP address: 192.168.4.1
19:44:25.114 -> *wm:[1] Starting Web Portal 
19:44:25.114 -> *wm:[2] HTTP server started 
19:44:25.114 -> *wm:[2] Config Portal Running, non blocking/processing 
19:44:25.114 -> *wm:[2] Portal Timeout In 60 seconds
19:44:25.114 -> 
19:44:25.114 -> Setup END
19:44:52.041 -> *wm:[2] Portal Timeout In 32 seconds
19:45:22.072 -> *wm:[2] Portal Timeout In 2 seconds
19:45:24.075 -> *wm:[1] config portal has timed out 
19:45:24.111 -> *wm:[2] shutdownConfigPortal 
19:45:24.111 -> *wm:[2] restoring usermode STA
19:45:25.108 -> *wm:[2] WiFi Reconnect, was idle 
19:45:25.108 -> *wm:[2] wifi status: WL_IDLE_STATUS
19:45:25.108 -> *wm:[2] wifi mode: STA
19:45:25.108 -> *wm:[2] configportal closed 
```

tablatronix, 2022-03-14:
Whats wrong with that ? Looks normal to me

tablatronix, 2022-03-14:
Is this with process?

tablatronix, 2022-03-14:
nm, I have it reproduced here!

tablatronix, 2022-03-14:
I mean this is technically not a bug, you are just starting the CP twice, which should be ok say you want to change the credentials, but I am going to add a check and fail to avoid confusion.

ChristophCaina, 2022-03-14:
thanks :) - and sorry for not replying directly... (timezones^^)

it is an WeMos D1 R2 (ESP8266), it was with wm.process() - but the same applies without wm.process();

The main idea (and I think that's what this cp is for was):
To provide the cp to allow a customer to change the "default" WiFi Settings to his own WiFi - therefore, there's no reason to start the CP twice, when it can't find any saved credentials.

So, yes, while it shouldn't be a big issue - it was confusing a bit :)
Thanks for having a look into it again - really appreciated (y)
Now, I need to see how I can setup the CP websites itself... ^^

ChristophCaina, 2022-03-14:
one more question:
How can I print the SSID of the "configPortal" when it is connected / started?
I read something about WiFi.SSID() ... but not sure, if this is working within the ConfigPortal() too, or only when it could connect to a stored WiFi...

closed 2022-04-07 by tablatronix