tzapu/WiFiManager #1372: Dash in SSID is converted to e28093

JohnvanderPol2, 2022-03-15:
While testing  a simple example of the Wifimanager i could connect to one wifi network, but when connecting to another SSID it kept failing.

I traced the problem to the SSID not being correct, even though it was selected in the WifiManager Web Interface
It kept complaining that 
no WPG–OFFICE found, reconnect after 1s
While i use the SSID hardcoded (WPG-OFFICE), the WIFI does connect
When comparing the the two strings that should be equal, to my surprise they were not.

WPG–OFFICE  -> 575047e280934f46464943450a
WPG-OFFICE  -> 575047   2d      4f46464943450a
As can be seen it shows the dash "2d" with an unicode character "e28093"
This happens somewhere in the webinterface, and it takes this malformed SSID.

I have seen several posts more of people having problems with SSID, and I saw most of them also use a dash.
It took me several days of confusion to understand the problem.

I think its a simple error to fix.
I am taking a look now if i can "fix" the unicode character in the saveConfigCallback

Sample code
  if(wifimanager.getWiFiIsSaved())
    {
    String storedssid=wifimanager.getWiFiSSID(true);
    DBG_OUTPUT_PORT.printf("stored ssid to %s\n", storedssid.c_str());
    String storedpw=wifimanager.getWiFiPass(true);
    DBG_OUTPUT_PORT.printf("stored password %s\n", storedpw.c_str());//for debugging only
    WiFi.begin(storedssid.c_str(), storedpw.c_str());//somehow it is not able to autoconnect sometimes, so connect when we can
    int counter=0;
    while(WiFi.status() != WL_CONNECTED)
      {
      delay(500);   
      if(counter==0)
        {
        DBG_OUTPUT_PORT.println(F("Connecting"));
        }
      else
        {
        if(strcmp(storedssid.c_str(),"WPG-OFFICE")!=0)
          {
          DBG_OUTPUT_PORT.print(storedssid.c_str());
          DBG_OUTPUT_PORT.println("WPG-OFFICE");
          }
        if(strcmp(storedpw.c_str(),"xxxxxxx")!=0)
          {
          DBG_OUTPUT_PORT.print(storedpw.c_str());
          DBG_OUTPUT_PORT.println("xxxxxxxx");
          }
        DBG_OUTPUT_PORT.print(F("."));
        }  
      if(counter > 30)
        {
        DBG_OUTPUT_PORT.println(F("Switching to hardcoded credentials"));
        break;  
        }
      counter++;  
      }
    counter=0;
    if(WiFi.status() != WL_CONNECTED)
      {
      WiFi.begin("WPG-OFFICE","xxxxxxxxx");//somehow it is not able to autoconnect sometimes, so connect when we can
      while(WiFi.status() != WL_CONNECTED)
        {
        delay(500);   
        if(counter==0)
          {
          DBG_OUTPUT_PORT.println(F("Connecting"));
          }
        else
          {
          DBG_OUTPUT_PORT.print(F("."));
          }
        if(counter > 30)
          {
          DBG_OUTPUT_PORT.println(F("Switching to wifimanager to connect"));
          break;  
          }
        counter++;  
        }
      } 
    }
  else
    {
    DBG_OUTPUT_PORT.println(F("No saved SSID data"));
    }
  DBG_OUTPUT_PORT.println(F("Starting autoconnect"));
  if(WiFi.status() != WL_CONNECTED) 
    {//use the wifimanager to configure it correctly
    if(!wifimanager.autoConnect("Nursery"))//no Acces Point name, nor a password for now
      {
      DBG_OUTPUT_PORT.println(F("failed to connect and hit timeout"));
      ESP.restart();  
      }
    if(shouldSaveConfig)  
      {//
      saveConfiguration();  
      }
    }   
  DBG_OUTPUT_PORT.printf("Connecting to %s\n", wifimanager.getWiFiSSID().c_str());
  DBG_OUTPUT_PORT.printf("Password %s\n", wifimanager.getWiFiPass().c_str());//for debugging only
  DBG_OUTPUT_PORT.printf("local ip %s\n", WiFi.localIP());

here logging generated with the included sample 

stored ssid to WPG–OFFICE
stored password xxxxxxxx
Connecting
WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.scandone
no WPG–OFFICE found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.scandone
no WPG–OFFICE found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.scandone
no WPG–OFFICE found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.scandone
no WPG–OFFICE found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.scandone
no WPG–OFFICE found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.WPG–OFFICEWPG-OFFICE
.Switching to hardcoded credentials
scandone
wifi evt: 2
Connecting
....scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with WPG-OFFICE, channel 1

token9, 2022-03-20:
I ran into the same issue today. 
When connecting with dlink–xxxxxxxx
*wm:[1] Connecting to NEW AP: dlink–xxxxxxxx
*wm:[2] [EVENT] WIFI_REASON:  201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND

when connecting with SSID dlinkxxxxxxxx
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS]

token9, 2022-03-25:
Yes confirmed now works OK, cheers

*wm:[2] processing save
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to NEW AP: dlink-xxxxxx
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS]
*wm:[1] Got IP Address:
*wm:[1] 192.168.1.100
*wm:[2] [CB] _savewificallback calling
0000312.701: saveWifiConfigCallback called.
*wm:[2] shutdownConfigPortal
*wm:[0] [ERROR] disconnect configportal - softAPdisconnect FAILED
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed
*wm:[1] config portal exiting

tablatronix, 2022-03-25:
Ill cut a new release