tzapu/WiFiManager #1373: Loop when starts esp32 before WIFI router

AFiabani, 2022-03-16:
**I have a problem when a power outage occurs. The ESP32 turned on before my WIFI router, then ESP32 go in a loop.** 

21:23:17.941 -> *wm:[1] AutoConnect 
21:23:17.941 -> *wm:[2] ESP32 event handler enabled 
21:23:17.988 -> *wm:[2] Connecting as wifi client... 
21:23:18.035 -> *wm:[2] setSTAConfig static ip not set, skipping 
21:23:18.083 -> *wm:[1] Connecting to SAVED AP: E UM PAIS DA EUROPA?!
21:23:18.509 -> *wm:[1] connectTimeout not set, ESP waitForConnectResult... 
21:23:20.588 -> *wm:[2] [EVENT] WIFI_REASON:  201
21:23:20.588 -> *wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND 
21:23:20.636 -> *wm:[2] Connection result: WL_NO_SSID_AVAIL
21:23:20.683 -> *wm:[1] AutoConnect: FAILED 
21:23:20.731 -> *wm:[2] Starting Config Portal 
21:23:20.778 -> *wm:[2] AccessPoint set password is VALID 
21:23:20.826 -> *wm:[2] Disabling STA 
21:23:20.826 -> *wm:[2] Enabling AP 
21:23:20.874 -> *wm:[1] StartAP with SSID:  AutoConnectAP
21:23:21.251 -> *wm:[1] AP IP address: 192.168.4.1
21:23:21.298 -> *wm:[1] Starting Web Portal 
21:23:21.345 -> *wm:[2] HTTP server started 
21:23:21.390 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
21:23:44.296 -> *wm:[2] NUM CLIENTS:  0
21:24:14.298 -> *wm:[2] NUM CLIENTS:  0
21:24:44.297 -> *wm:[2] NUM CLIENTS:  0
21:25:14.315 -> *wm:[2] NUM CLIENTS:  0
21:25:44.282 -> *wm:[2] NUM CLIENTS:  0
.
.
**Esp 32 keep in the loop even my WIFI router is on**
.
.

**Then I restart ESP32, and it runs normally**

21:47:54.652 -> *wm:[1] AutoConnect 
21:47:54.699 -> *wm:[2] ESP32 event handler enabled 
21:47:54.746 -> *wm:[2] Connecting as wifi client... 
21:47:54.793 -> *wm:[2] setSTAConfig static ip not set, skipping 
21:47:54.840 -> *wm:[1] Connecting to SAVED AP: E UM PAIS DA EUROPA?!
21:47:55.266 -> *wm:[1] connectTimeout not set, ESP waitForConnectResult... 
21:47:57.530 -> *wm:[2] Connection result: WL_CONNECTED
21:47:57.577 -> *wm:[1] AutoConnect: SUCCESS 
21:47:57.624 -> *wm:[1] STA IP Address: 192.168.100.78
21:47:57.672 -> connected...yeey :)

**How can a solve this problem?**
**I was tink if it has an "ESP.restart();" if it tries too many times it will solve my problem** 
**any suggestion?**

tablatronix, 2022-03-16:
add a configportaltimeout() so it doesnt get stuck in that loop or use the non blocking mode

closed 2022-03-16 by tablatronix

AFiabani, 2022-03-19:
Tnks, That works great. 
I see before that you post this solution in your notes. Sorry for asking here.
And thank you for your hard work in developing WiFiManager.

tablatronix, 2022-03-19:
Yeah power failures can be a problem, maybe we can add a wificheck() and have it make sure it autorecconnects in the back ground and not in loop