tzapu/WiFiManager #1376: TODO Fix badge links in readme

tablatronix, 2022-03-26:
Badges should link to pages

tablatronix, 2022-06-16:
[image]

link to the images and not actual pages

JoergAJ, 2022-06-24:
Links to pages at espressif  like https://www.espressif.com/en/products/socs/esp32-c3 for the esp32c3 ?

tablatronix, 2022-06-24:
Yeah but mostly the CI test status pages, but it would be cool if they are linked to some reference maybe?

seanhagen, 2022-09-29:
@tablatronix like this: https://github.com/seanhagen/WiFiManager/tree/1376-fix-badge-links-in-readme ? I put all the badges on one line, added a link to the releases page for the "Release" badge, and put the chat & discord badges on the same line. All the other badges already have links that *seem* to go to the right places, but I'm not 100% sure

tablatronix, 2022-09-30:
Looks bad on mobile

aumpatel2624, 2023-07-27:
hey I want to contribute in solving this issue.

tablatronix, 2023-07-27:
Do it

tablatronix, 2023-07-27:
And feel free to add anything else to the readme or wiki it needs all the new updated stuff

belanasaikiran, 2026-02-27:
Is this fixed or needs to be fixed ? What can I help with ?

aumpatel2624, 2026-02-28:
It still needs to be fixed I guess