tzapu/WiFiManager #1380: Insecure onDemandAP

Desat, 2022-03-28:
First of all: My English is not that good but I'm trying my best. 

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master or 2.0.10-beta

Esp8266/Esp32:  ESP32 

Hardware:	ESP-Wroom-32

Core Version:  Espressif 1.0.6

### Description

Problem description

### Settings in IDE

Module:  	AZ-Delivery ESP32 Dev Board

Additional libraries:  none

### Sketch
A slightly modified advanced example version. 

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#define TRIGGER_PIN 0

bool wm_nonblocking = true; // change to true to use non blocking
WiFiManager wm; // global wm instance

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  Serial.begin(115200);
  while(!Serial); 
  Serial.setDebugOutput(true);  
  Serial.println("\n Starting");

  pinMode(TRIGGER_PIN, INPUT);
  
  // wm.resetSettings(); // wipe settings

  if(wm_nonblocking) wm.setConfigPortalBlocking(false);

   const char* menu[] = {"wifi","info","param","sep","close","exit","restart","erase"};
   wm.setMenu(menu,8);

  wm.setClass("invert");

  wm.setConfigPortalTimeout(60); // auto close configportal after n seconds
  wm.setCaptivePortalEnable(true); // disable captive portal redirection
  wm.setAPClientCheck(true); // avoid timeout if client connected to softap

   wm.setShowInfoErase(true);      // do not show erase button on info page
   wm.setScanDispPerc(true);       // show RSSI as percentage not graph icons
  
  bool res;
  res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

  if(!res) {
    Serial.println("Failed to connect or hit timeout");
  } 
  else {
    Serial.println("connected...yeey :)");
  }
}

void checkButton(){
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
    delay(500);
    if( digitalRead(TRIGGER_PIN) == LOW ){
      Serial.println("Button Pressed");
      delay(3000); // reset delay hold
      if( digitalRead(TRIGGER_PIN) == LOW ){
        Serial.println("Button Held");
        Serial.println("Erasing Config, restarting");
        wm.resetSettings();
        ESP.restart();
      }
 
      Serial.println("Starting config portal");
      wm.setConfigPortalTimeout(120);
      
      bool res;
      res = wm.startConfigPortal("OnDemandAP","password"); 
      
      if (!res) {
        Serial.println("failed to connect or hit timeout");
        delay(3000);
      } else {
        //if you get here you have connected to the WiFi
        Serial.println("connected...yeey :)");
      }
    }
  }
}

void loop() {
  if(wm_nonblocking) wm.process(); // avoid delays() in loop when non-blocking and other long running code  
  checkButton();
}
#END
```

### Debug Messages
none
```
When I start the "onDemandAP" for the first time after storing the WiFi credentials, it is started with a password query. 
If I now let the time-out elapse or log in with a password and then also log off with "Exit", an insecure access without a password query is started when "onDemandAP" is activated again. 
Restarting the dev board or leaving it without power does not change the situation, which from the second "onDemandAP" I only create an insecure access. 
Do you have an idea or are you not aware of the error yet? 
If you need more information about my system, I'll be happy to answer it. 
First of all, I just set up my system again a few days ago with the new Arduino IDE and also Visual Studio Code with PlatformIO. 
But I also have to say thank you because I finally found a WiFi manager that also saves the credentials. 
Two years ago I gave up because I hadn't found anything useful. 
Only the " captive portal redirection" on the PC (Opera) is unfortunately not the IP of the AP but the MSN start page. 
However , it works perfectly on the mobile phone .

tablatronix, 2022-03-29:
WiFi.mode(WiFi_STA); in setup

Desat, 2022-03-29:
Yes, that's correct, but that's not the problem.
The problem is that if the ESP has no network credentials stored, the ESP starts the "AutoConnectAP" with a secure connection.
Here I now enter my network credentials and it starts as expected in station mode. 
If I now request an "onDemandAP" it will also be started as a secure connection with the specified network credentials, in addition to the "station mode". 
Up to here everything is as expected. 
But if the login timeout expires or I log in "with a password" and then log out via "exit", it still behaves as expected. 
However, from now on every further requested "onDemandAP" (even after a reboot of the ESP) will be established with an insecure connection although a secure connection should actually be established. 
I, and theoretically anyone who is not authorized to do so, can connect to the ESP "without a password". 
Either the ESP is not able to create a secure access again, even after a reboot, or the library has taken the wrong path somewhere. 
I hope I was able to explain the problem clearly with my "Google" translated English.

tablatronix, 2022-03-29:
can you show screenshots or a video, I do not understand exactly

tablatronix, 2022-03-29:
hmmm fascinating, this might be a ESP bug

Desat, 2022-03-29:
Yes i am sure. :) my devices show the wifi icon without the small lock icon. also i remove the known ESP AP networks before testing.
At the first onDemandAP it is WPA2 but all following onDemandAP are without security. i can make a screenshot from the network scan on my mobile phone where you can see that the wifi icon has no lock icon. But i see you have the same problem. the first onDemandAP is secure btw ;) Do you still need the screenshots?

tablatronix, 2022-03-29:
I do not reproduce, that was a mistake, I forgot I was starting that one anonymous in my test.

When you say the following? What do you mean?

Like if I press a button and do wm.startConfigPortal("OnDemandAP","12345678")
everytime I press it its the same..

Have you done a full erase? Have you made sure you are setting wifi mode to sta only in your code?

Desat, 2022-03-29:
its the original WiFiManager-2.0.10-beta library with the original example "Advanced". but i see now the "AutoConnectAP" makes an insecure wifi network. i hope i have another unused ESP32 board at home. My test outside the WiFiManager works fine with "onDemandAP". on off on off all times no problem.

tablatronix, 2022-03-29:
hmm
advanced.ino
  res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

Desat, 2022-03-29:
I know, but I have no idea how to erase all potentially corrupted data from the ESP. You can hardly find reasonable Wifi managers or explanations in my language and the English help has rarely been successful in realizing my actual project.
Unfortunately, I can no longer use the WiFi manager for myself, which does not necessarily have to be due to the library. I would search the library myself for the error, but unfortunately my knowledge is too weak and the library too "powerful". But otherwise a very strong tool if the hardware plays along. Thumbs up for all the work that went into it. But I'll leave it at that and don't want to keep other important projects from happening. Keep it up and maybe I'll try it again with different hardware.

tablatronix, 2022-03-29:
Ok I might be seeing the same issue, let me keep testing.

tablatronix, 2022-03-29:
This seems to be something in the ESP lib, and unknown bug, maybe corrupt memory, I will keep testing for a workaround.
I tried another esp and its fine, so its a strange issue.

tablatronix, 2022-03-29:
you will want to be checking the return of WiFi.softAP()
should be boolean true or false,

Desat, 2022-03-29:
yes of cours but i check it with network scan on my mobile phone. The softAPdisconnect return value is every time false.
if have now added the Wifi.getMode() before and after the softAPdisconnect. Before the value is 3 (WIFI_MODE_APSTA) after the value is 1 (WIFI_MODE_STA) . everything as expected. 
You could actually use it as feedback for "startConfigPortal", right? As far as I've read and I remember correctly, the return value there is always false, I think.

tablatronix, 2022-03-30:
Yeah softapdisconnect fails alot not sure why

tablatronix, 2022-03-30:
still debugging, are you calling resetsettings at all?

Desat, 2022-03-30:
Did you mean "erase" in config portal?
If so, yes several times.

Which kind of code? My last code that i posted yesterday? But with the return value from "softApdisconnect" and the AP start command?

tablatronix, 2022-03-30:
I am just trying to find what can cause this, I can reproduce very rarely in my own code, and never in your example.
I am on 1.0.6 also

Ill keep testing

Desat, 2022-03-30:
I have now installed the ESP32 Core Version 2.0.2. There, the "onDemandAP" seems to always work securely now (tested 4 times first). The error message "[ERROR] disconnect configportal - softAPdisconnect FAILED" when exiting the softAP has also disappeared.
just the part
      if (!wm.startConfigPortal("OnDemandAP","password")) {
        Serial.println("failed to connect or hit timeout");
still seems to just return False. But it can also be because of "bool wm_nonblocking = true; ".

I will continue to test the secure access point.

Maybe it's because of this highlighted original ESP32 library code
-------------------------------------------------- ------------------------------------------------

Extract Core1.0.6 WiFiAP.cpp Line 196
`
bool WiFiAPClass::softAPdisconnect(bool wifioff)
{
    bool ret;
    wifi_config_tconf;

    if(WiFiGenericClass::getMode() == WIFI_MODE_NULL){
        return false;
    }

**
    *conf.ap.ssid = 0;
    *conf.ap.password = 0;
    conf.ap.authmode = WIFI_AUTH_OPEN; // auth must be open if pass=0
    ret = esp_wifi_set_config(WIFI_IF_AP, &conf) == ESP_OK;
**

    if(wifioff) {
        ret = WiFi.enableAP(false) == ESP_OK;
    }

    return ret;
}
`

Extract Core 2.0.2 WiFi.cpp Line 213
`
bool WiFiAPClass::softAPdisconnect(bool wifioff)
{
    bool ret;
    wifi_config_tconf;
    wifi_softap_config(&conf);

    if(WiFiGenericClass::getMode() == WIFI_MODE_NULL){
        return false;
    }

    **ret = esp_wifi_set_config((wifi_interface_t)WIFI_IF_AP, &conf) == ESP_OK;**

    if(ret && wifioff) {
        ret = WiFi.enableAP(false) == ESP_OK;
    }

    return ret;
} `

tablatronix, 2022-03-30:
That is where I have been debugging, I think the issue is in the sdk inside esp_wifi_set_config, because everything passes

Desat, 2022-03-30:
in the core 2.0.2 of the esp32 the problem doesn't exist anymore, so it's ok for me. 
Of course there could be another error in Core 2.0.2, namely that the station mode logs off after about 7 minutes because of "Reason: 2 - AUTH_EXPIRE" but that wasn't the real problem and I think everyone can live with that . 
The auto mode also has a small problem when connecting, but by and large at least the security of the "onDemandAP" works with the core 2.0.2. 
But since my knowledge is not yet big enough to confirm that, it's just a guess on my part ;)

tablatronix, 2022-03-30:
Yeah I will note this to espressif only because its a major issue, so ill log all my info and bin and report it.

I have not heard of the auth expire, only the reboot auth fail/expire issue

tablatronix, 2022-03-30:
https://github.com/espressif/arduino-esp32/issues/5038

closed 2022-03-30 by tablatronix

Desat, 2022-03-30:
this is also new in core 2.0.2
every second boot of the esp32 in unable to connect in station mode (see timestamp 20:16:50.605)

20:16:38.627 -> ets Jun  8 2016 00:22:57
20:16:38.627 -> 
20:16:38.627 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:16:38.673 -> configsip: 0, SPIWP:0xee
20:16:38.673 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:16:38.673 -> mode:DIO, clock div:1
20:16:38.673 -> load:0x3fff0030,len:1324
20:16:38.673 -> ho 0 tail 12 room 4
20:16:38.673 -> load:0x40078000,len:13508
20:16:38.673 -> load:0x40080400,len:3604
20:16:38.673 -> entry 0x400805f0
20:16:38.906 -> [2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
20:16:38.953 -> 
20:16:38.953 ->  Starting
20:16:39.000 -> [    57][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
20:16:39.095 -> [   149][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
20:16:39.095 -> [   149][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
20:16:41.086 -> *wm:[1] AutoConnect 
20:16:41.086 -> *wm:[2] ESP32 event handler enabled 
20:16:41.086 -> *wm:[2] Connecting as wifi client... 
20:16:41.086 -> *wm:[2] setSTAConfig static ip not set, skipping 
20:16:41.086 -> *wm:[1] Connecting to SAVED AP: Desat-Wifi
20:16:41.596 -> [  2661][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
20:16:41.596 -> *wm:[2] 20000 ms timeout, waiting for connect...
20:16:41.596 -> *wm:[2] . 
20:16:41.689 -> *wm:[2] . 
20:16:41.781 -> *wm:[2] . 
20:16:41.872 -> *wm:[2] . 
20:16:42.011 -> *wm:[2] . 
20:16:42.104 -> *wm:[2] . 
20:16:42.197 -> *wm:[2] . 
20:16:42.291 -> *wm:[2] . 
20:16:42.384 -> *wm:[2] . 
20:16:42.475 -> *wm:[2] . 
20:16:42.571 -> *wm:[2] . 
20:16:42.710 -> *wm:[2] . 
20:16:42.802 -> *wm:[2] . 
20:16:42.896 -> *wm:[2] . 
20:16:42.989 -> *wm:[2] . 
20:16:43.082 -> *wm:[2] . 
20:16:43.173 -> *wm:[2] . 
20:16:43.312 -> *wm:[2] . 
20:16:43.406 -> *wm:[2] . 
20:16:43.497 -> *wm:[2] . 
20:16:43.590 -> *wm:[2] . 
20:16:43.683 -> [  4764][V][WiFiGeneric.cpp:284] _arduino_event_cb(): STA Connected: SSID: Desat-Wifi, BSSID: 2c:3a:fd:a3:38:af, Channel: 6, Auth: UNKNOWN
20:16:43.683 -> [  4765][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
20:16:43.683 -> *wm:[  4780][V][WiFiGeneric.cpp:294] _arduino_event_cb(): STA Got New IP:192.168.178.40
20:16:43.728 -> [2] .[  4783][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
20:16:43.728 -> [  4788][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.178.40, MASK: 255.255.255.0, GW: 192.168.178.1
20:16:43.728 ->  
20:16:43.821 -> *wm:[2] Connection result: WL_CONNECTED
20:16:43.821 -> *wm:[1] AutoConnect: SUCCESS 
20:16:43.821 -> *wm:[2] Connected in 2759 ms
20:16:43.821 -> *wm:[1] STA IP Address: 192.168.178.40
20:16:43.821 -> connected...yeey :)
20:16:45.588 -> ets Jun  8 2016 00:22:57
20:16:45.634 -> 
20:16:45.634 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:16:45.634 -> configsip: 0, SPIWP:0xee
20:16:45.634 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:16:45.634 -> mode:DIO, clock div:1
20:16:45.634 -> load:0x3fff0030,len:1324
20:16:45.634 -> ho 0 tail 12 room 4
20:16:45.634 -> load:0x40078000,len:13508
20:16:45.634 -> load:0x40080400,len:3604
20:16:45.634 -> entry 0x400805f0
[... 72 lines cut ...]
20:16:52.736 -> [  6852][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 10 - AP_START
20:16:53.246 -> *wm:[1] AP IP address: 192.168.4.1
20:16:53.246 -> *wm:[1] Starting Web Portal 
20:16:53.246 -> [  7352][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
20:16:53.246 -> [  7354][V][WebServer.cpp:85] WebServer(): WebServer::Webserver(port=80)
20:16:53.246 -> *wm:[2] HTTP server started 
20:16:53.246 -> *wm:[2] Config Portal Running, non blocking/processing 
20:16:53.246 -> *wm:[2] Portal Timeout In 60 seconds
20:16:53.246 -> Failed to connect or hit timeout
20:17:15.905 -> *wm:[2] Portal Timeout In 36 seconds
20:17:45.877 -> *wm:[2] Portal Timeout In 6 seconds
20:17:52.699 -> *wm:[1] config portal has timed out 
20:17:52.699 -> *wm:[2] shutdownConfigPortal 
20:17:52.699 -> [ 66832][V][WiFiGeneric.cpp:317] _arduino_event_cb(): AP Stopped
20:17:52.747 -> [ 66832][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 11 - AP_STOP
20:17:52.747 -> *wm:[2] restoring usermode STA
20:17:52.747 -> [ 66836][V][WiFiGeneric.cpp:314] _arduino_event_cb(): AP Started
20:17:52.747 -> [ 66847][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 10 - AP_START
20:17:53.718 -> [ 67847][V][WiFiGeneric.cpp:317] _arduino_event_cb(): AP Stopped
20:17:53.718 -> [ 67847][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 11 - AP_STOP
20:17:53.765 -> [ 67848][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
20:17:53.765 -> *wm:[ 67855][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
20:17:53.765 -> [2] wifi status: Unknown
20:17:53.765 -> *wm:[2] wifi mode: STA
20:17:53.765 -> *wm:[2] configportal closed

tablatronix, 2022-03-30:
I linked an issue above for this that suggests it is the lack of authmode comparison in the equals assertion check that causes startap to return true even if auth setting failed.

tablatronix, 2022-03-30:
also this issue is known and fixes are in the works for it.
I added a fix to wm to workaround it also

#1067

Desat, 2022-03-30:
It's funny, actually I just wanted to have a simple way of entering the WiFi credentials via a Nextion display and the ESP32 then using them for the next restart. Haven't had any success so far, because the credentials are only accepted as "const" and my programming skills are really on the lower level. Let alone understand the "persistent" feature.

tablatronix, 2022-03-30:
ah, you might want to try making a const char* in your sketch, and memcpy them from the inputs, not sure how you are getting them in , as strings? etc. but you will probably have to do a copy to make sure the memory is safe when you pass it to the begin()

or if you want to just use WM then you can use

`    // to preload autoconnect for test fixtures or other uses that skip esp sta config
    bool          preloadWiFi(String ssid, String pass);
`

Desat, 2022-03-30:
Actually, I would like to use as few libraries as possible because my project is a "digital analogue clock" with addressable LEDs and the memory on the ESP is also limited. In addition to the clock function, temperature, humidity and air pressure should also be shown on the display. The WiFi access was originally only intended to be used for time synchronization, but I'm thinking about an OTA update function. Furthermore, a calendar function with public holidays, sunrise and sunset and moon phases was planned for the display. In addition, a spectrum analyzer as a "decoration effect" and a brightness sensor to adjust the luminosity of the LEDs depending on the interior brightness.
Yes, it's "just" a small project for someone whose programming skills are on the lower level / simplest basics.