tzapu/WiFiManager #1381: Missing core_version.h when building using non-finalized platforms

thorrak, 2022-03-28:
As a result of [this change](https://github.com/tzapu/WiFiManager/commit/da02b59a21cff08f6e1c9858e06228622ecf9568#diff-71e2eff4cef6c58b55b7fa975c102042205487f444ac3a080c99de73280c12c6R68-R74), building in PlatformIO using upstreams that are not yet released versions results in an error.

Example of un-released upstream in platformio.ini:
```
[env:esp32_s2_wifi]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```

Error:
`.pio/libdeps/esp32_s2_wifi/WiFiManager/WiFiManager.h:69:14: fatal error: core_version.h: No such file or directory`

Using the feature/arduino-upstream platform is the recommended method for being able to compile for the ESP32-S2, amongst other chips.

thorrak, 2022-03-28:
PR added which resolves the bug

tablatronix, 2022-03-28:
I thought I removed this already?

tablatronix, 2022-03-28:
Well shoot my bad I thought I committed a fix already, oops

PR #1382 (2022-03-29): Output "Unknown" if Arduino/ESP-IDF version is unknown
This change closes #1381 by not attempting to include core_version.h (which will not exist if the platform has been fully released).
commit ea4ee025 (2022-03-29): Output "Unknown" if Arduino/ESP-IDF version is unknown (#1382)

Closes #1381
closed 2022-03-29 by tablatronix