tzapu/WiFiManager #1383: Calling autoConnect() in loop() causes lag

OekoSolveMG, 2022-03-29:
### Basic Infos

#### Hardware
WiFiManager Branch/Release:  Master 2.0.5-beta
Esp8266/Esp32: ESP32
Hardware: M5 Core 2

### Description
When attempting to reconect in the loop, for the case that the device loses WiFi connection and automatically attempts to reconnect (with delays or not) the device becomes laggy and blocks input after a while.

The main reason from what I could debug is that the ```autoConnect()``` subscribes to a callback in ```WiFi_autoReconnect()```. In this case that would the the ```WiFi.onEvent```.

```cpp
void WiFiManager::WiFi_autoReconnect(){
  #ifdef ESP8266
    WiFi.setAutoReconnect(_wifiAutoReconnect);
  #elif defined(ESP32)
    // if(_wifiAutoReconnect){
      // @todo move to seperate method, used for event listener now
      #ifdef WM_DEBUG_LEVEL
      DEBUG_WM(DEBUG_VERBOSE,F("ESP32 event handler enabled"));
      #endif
      using namespace std::placeholders;
      wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2)); // <---------- This Line here
    // }
  #endif
}
```

The problem is this subscription get's pushed back to a ```vector``` of callbacks in the ```WiFiGeneric``` class.

```cpp
wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventFuncCb cbEvent, system_event_id_t event)
{
    if(!cbEvent) {
        return 0;
    }
    WiFiEventCbList_t newEventHandler;
    newEventHandler.cb = NULL;
    newEventHandler.fcb = cbEvent;
    newEventHandler.scb = NULL;
    newEventHandler.provcb = NULL;
    newEventHandler.event = event;
    cbEventList.push_back(newEventHandler); // <---------- This Line here
    return newEventHandler.id;
}
```

Meaning that if you call ```autoConnect()``` in a loop then over a shorter or longer period depeding on how fast you recall ```autoConnect()```, the system will be subscribed to that many callbacks that the rest of the system starts to lag.

### Possible fixes

1. Make WiFiManager::WiFiEvent actually reconnect to the WiFi when using the ESP32. This is not done anymore becuase esp32autoreconnect is disabled. Therefore using autoConnect in a loop is not needed anymore.

```cpp
// #define esp32autoreconnect    // implement esp32 autoreconnect event listener kludge, @DEPRECATED
```

2. Make ```autoConnect()``` removeEvent instead of just the deconstructor.

```cpp
WiFi.removeEvent(wm_event_id);
```

3. Add a check that only subscribed to the callback if it isn't subscribed already.

### Sketch (Quite big so only relevant wifi parts have been included)

```cpp
#BEGIN
if (!wifi_connected) {
    Logger::log(F("[W] Reconnecting to WiFi"));
    wm.autoConnect();
  }

  // Check if reconnecting failed.
  if (!wifi_connected) {
    Logger::log(F("[W] Reconnecting to WiFi failed"));
    return connected;
  }
#END
```

tablatronix, 2022-03-29:
hmm, I did not expect people to call autoconnect in a loop

tablatronix, 2022-03-29:
Try this

OekoSolveMG, 2022-03-30:
Thanks a lot, my use case stems from me wanting to uphold the connection for a long time and because it is an enduser device the wifi could be turned off / not have enough signal strength for a while and lost the connection etc.

In this case I want to reconnect the device, as soon as possible because I am uploading data over MQTT.

Is there a better way to reconnect that could be called more than once? Because I looked through the source code of the WiFiManager and sadly didn't find anything that seemed to reconnect and wasn't private besides autoConnect().

I am still wondering if there is a cleaner way to reconnect the device at runtime if it ever looses connection, additionally do I have to turn on any features with the setters to make sure it reconnects or should that be done automatically anyway? Because ```setWiFiAutoReconnect(true)``` seems to set a boolean that is never used. Only with the ESP8266, does it actually set the ```autoReconnect``` on the underlyiung ```WiFiSTA.h```.

This is not done however for the ESP32 which only subscribed to the WiFiEvent, which also does not seem to reconnect.

```cpp
void WiFiManager::WiFi_autoReconnect(){
  #ifdef ESP8266
    WiFi.setAutoReconnect(_wifiAutoReconnect);
  #elif defined(ESP32)
    // if(_wifiAutoReconnect){
      // @todo move to seperate method, used for event listener now
      #ifdef WM_DEBUG_LEVEL
      DEBUG_WM(DEBUG_VERBOSE,F("ESP32 event handler enabled"));
      #endif
      using namespace std::placeholders;
      wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2));
    // }
  #endif
}
```

Perhaps using something like ```wifiConnectDefault()``` could be used in the Event callback when the connection has been lost, when using ESP32 or if possible setAutoReconnect could be set for the ESP32 as well?

tablatronix, 2022-03-30:
Well my point is that the esp libraries are supposed to have autoreconnect enabled by default but I have found they are not always working in the background, not sure why.

tablatronix, 2022-04-07:
ill create a new issue for autoreconnect testing and solutions

closed 2022-04-07 by tablatronix

OekoSolveMG, 2022-04-08:
After looking into the WiFiClient / WiFiClientGeneric it seems that the ```WiFi.setAutoReconnect()``` does do something, if you loose connection because of authentication or a lot of other reasons etc. You seem to automatically reconnect. This is not the case tough for the case where the access point is turned off and then turned on again. Because it doesn't register a ```WiFiEven```t when an access point that was previously turned off is turned on again.

Meaning this case needs to be polled by the end user to reconnect yourself. Additionaly the ```WiFiEvent``` of the WiFiManager doesn't seem to register any WiFiEvents even if ```setWiFiAutoReconnect()``` was set.

tablatronix, 2022-04-08:
Ahh that makes sense, it should still emit a disconnected event ?

tablatronix, 2022-04-12:
I fixed this wrong, and backwards. oops

commit 483b8348 (2022-04-12): Fix #1383