tzapu/WiFiManager #1386: Issue when Scan is cached xxxx ms ago

jonathanvanpeteghem, 2022-04-06:
First of all a big thanks to Tzapu and of course Tablatronix for doing such a great job! Special thanks to Tablatronix for maintaining this library and make it into something that actually works and is stable. In the last year(s) I saw this lib evolving from something buggy to something that is now actually usable. The problem I have is a minor problem, and is only something you will get in a very specific case. 

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master : latest version to date => v2.0.10-beta 

Hardware ESP32: tested both Wrover and Wroom 

### Description

The basic problem can be replicated by using the on demand config portal example:

When initializing the wifimanager with => WiFiManager wm;    as a global object , and when the button is pushed, it is possible to have a problem. The first time the button is pushed, all is ok. The wifinetworks are found. We can select ssid and set password.
Then we save, and the ESP32 connects to the network. 

When pushing the button a second time, to go to the config portal, within 60 seconds. => The wifimanger code detects that the networks have been scanned once before (debug : wm:[2] Scan is cached xxxx ms ago)=> no need to scan again => but after that only empty SSID's are given. (in debug you see multiple : wm:[2] DUP AP:        )

Note that you can only replicate this: when you move the initialisation from local (in the loop where the button is pressed) to global.
In other words it will only happen when the wm object is not trown away after the button is pushed and the wifi is actyally connected.
And you need to reconnect withing 60 seconds (wich is obvious and can be found in the wifimanager code : 60000 ms) 

### Settings in IDE

Module:  			It will happen with whatever module.

Additional libraries: NONE

### Sketch

```/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

int timeout = 120; // seconds to run for

WiFiManager wm;    // ==> the lib is called globaly, the wm object is kept in memory

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW) {
    //WiFiManager wm;    ==> placed in comments, the lib is called now globaly

    //reset settings - for testing
    //wm.resetSettings();
  
    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);

    if (!wm.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.restart();
      delay(5000);
[... 105 lines cut ...]
19:37:43.715 > *wm:[2] Scan is cached 31143 ms ago
19:37:43.715 > *wm:[1] 9 networks found
19:37:43.723 > *wm:[2] DUP AP:
19:37:43.723 > *wm:[2] DUP AP:
19:37:43.723 > *wm:[2] DUP AP: 
19:37:43.731 > *wm:[2] DUP AP:
19:37:43.731 > *wm:[2] DUP AP:
19:37:43.731 > *wm:[2] DUP AP:
19:37:43.731 > *wm:[2] DUP AP:
19:37:43.731 > *wm:[2] DUP AP:
19:37:43.731 > *wm:[2] AP:  0
19:37:43.739 > *wm:[2] Portal Timeout In 119 seconds
19:37:43.755 > [E][WebServer.cpp:633] _handleRequest(): request handler not found
19:37:43.771 > *wm:[2] <- Request redirected to captive portal
19:37:43.779 > [E][WebServer.cpp:633] _handleRequest(): request handler not found
19:37:43.787 > *wm:[2] <- Request redirected to captive portal
19:37:44.557 > [E][WebServer.cpp:633] _handleRequest(): request handler not found
19:37:44.557 > *wm:[2] <- Request redirected to captive portal
19:37:44.765 > [E][WebServer.cpp:633] _handleRequest(): request handler not found
19:37:44.773 > *wm:[2] <- Request redirected to captive portal
19:37:44.773 > [E][WebServer.cpp:633] _handleRequest(): request handler not found
19:37:44.789 > *wm:[2] <- Request redirected to captive portal
19:38:01.131 > [E][WebServer.cpp:633] _handleRequest(): request handler not found
19:38:01.148 > *wm:[2] <- Request redirected to captive portal
```

tablatronix, 2022-04-06:
hmm, good find, ill test this

closed 2022-04-07 by tablatronix

tablatronix, 2022-04-07:
changed this 590e212ddb4ff319fb6982ba9bd9c1e528d941f2 will be a better fix