tzapu/WiFiManager #1388: BUG, non-bocking, AP is shut aften connect when set first time

0x0fe, 2022-04-08:
In non blocking mode, with a timeout (15*60)  the AP is shut down right after STA connection succeed, after the wifi password has been set. 
This bug does not occur on subsequent boot (so, when the password is already saved).

tablatronix, 2022-04-09:
What bug? What is the expected behavior?

non blocking doesn't mean always on, if you are using autoconnect that is the expected behavior.

0x0fe, 2022-04-09:
When non blocking and timeout are set the portal keeps running after connect,  except at first time when you set the password, i am not sure ifs its a bug or a feature, but i feel should be consistent, either stop portal after connect or leave it open, regardless if its the first time you set the password or not, no?
I also notice something very strange, when adding a custom label it works fine at first time we set password, but the datas are corrupted any subsequent access to the portal (withint the same sessions at least)

tablatronix, 2022-04-09:
The portal should NOT keep running after connect, the portal should never run unless no wifi or you ask it too.

What is the code? Which example are you using , i can test

tablatronix, 2022-04-09:
You are calling wm.startConfigPortal("FillLand Smart Parasol");

ok so I see, which version you running?

0x0fe, 2022-04-09:
last version from manager. I think i know the problem : char devId[16]; this buffer was initially used for the UUID only, its too small for the HTML tags, shame on me

tablatronix, 2022-04-09:
yeah that shouds like a memory issue ( also length)

are you calling wm.process(); in loop?

0x0fe, 2022-04-09:
yes, the application is a free running state machine. By the way is there a define or call to avoid closing the AP after connect?

tablatronix, 2022-04-09:
The process of connecting turns off the softap by design, as sta connections cannot be made with it on

0x0fe, 2022-04-09:
ESP32C has AP-STA mode, with hardware handling of both, how comes it is a softAP?

tablatronix, 2022-04-09:
Also there was some changed to the code recently to non blocking timeouts, it might only be in git

0x0fe, 2022-04-09:
i recall now, ok i will pull from the master branch, thank you for all the clarifications

closed 2022-04-09 by 0x0fe

tablatronix, 2022-04-09:
If you really want softap+sta you can do it, let me see , the best is probably to startap after sta is connected.

I might have some code in there to play with this

```c++
    bool          _disableSTA             = false; // disable sta when starting ap, always
    bool          _disableSTAConn         = true;  // disable sta when starting ap, if sta is not connected ( stability )
```

0x0fe, 2022-04-09:
nice, that is convenient

tablatronix, 2022-04-09:
So I am thinking it should be working like that already, let me check and see

tablatronix, 2022-04-09:
Mine seems to work, 
`    Serial.println(wm.getModeString(WiFi.getMode()));`

tablatronix, 2022-04-09:
remove this line

`  // WiFi_Mode(_usermode); // restore users wifi mode, BUG https://github.com/esp8266/Arduino/issues/4372`

tablatronix, 2022-04-09:
ok if you remove

in processConfigportal
          shutdownConfigPortal(); 

Ill see how to make this a toggle, it seems to work OK depends on your wifi environment.

tablatronix, 2022-04-09:
also if you are gonna run the cp portal always, no reason to start it in autoconnect

```
/**
 * toggle configportal if autoconnect failed
 * if enabled, then the configportal will be activated on autoconnect failure
 * @since $dev
 * @access public
 * @param boolean enabled [true]
 */
void WiFiManager::setEnableConfigPortal(boolean enable)
{
    _enableConfigPortal = enable;
}```

tablatronix, 2022-04-09:
7bffda3fe67ae7d4cb567fbc6bfa7d4322871d29

```c++
    // if true (default) then stop the config portal from autoConnect when wifi is saved
    void          setDisableConfigPortal(boolean enable);
```

I will probably change name by release