tzapu/WiFiManager #1394: Example on using wifimanager with deep sleep

kargarisaac, 2022-04-14:
Hi,

Is there any simple example on how to use wifimanager with deep sleep? I have to configure wifi credentials after each deep sleep/wake up round. I couldn't find any example over the internet.

Thank you

tablatronix, 2022-04-14:
hmm perhaps try rebooting after save, wifi creds should be persistent.

gfaman, 2022-08-14:
Hi, 
I spent countless time to try to find a reconnection solution with WM after deep sleep or light sleep. It never reconnect by itself.
Read many comments, proposals. Seems it should be reconnected by itself but never in my case.

The first solution but not recommended by Tablatronix was to add a loop with wifiManager.autoConnect("ESP_AP", "12345678") after the sleep if Wifi not connected. It works sometimes but also it create self reset. Not reliable.

Finally, it seems I found a solution for light_sleep (need test on deep_sleep):
- Add esp_wifi_stop(0 before the call light_sleep and esp_wifi_start() after the sleep instruction.
- Then test Wifi connection, if not connected, WiFi.reconnect()
- If still not connected WiFi.begin();

I didn t figure out yet which part makes the connection, the reconnect or the begin but definitively, the stop and start make the job.
Hope it is helping.

Laurent

tablatronix, 2022-08-14:
hmm i started a test somewhere and forgot about it, you definitely have to manually handle the wifi state, I know i found at least one bug in wm for this, I think i fixed it, but not in reconnect, only autoconnect

gfaman, 2022-08-15:
Thanks Shawn for your consideration. So far, that workaround is working. Only one reset during the night (after multiple tentatives for reconnection).
Honestly, I'm quite surprising that not many people discussed about this reconnection issue after sleep mode. Not sure it is only me and my particular case.
For sake of clarity, please find the workaround I'm using:

    esp_wifi_stop();
    ret = esp_light_sleep_start();  /////////////////////////////////////////////////// SLEEP //////////////////////////////////////////////
    Serial.printf("light_sleep: %d\n", ret);
    esp_wifi_start();
    TRACE_WAKE_UP;
//    delay(period * 1000);
    delay(500);
uint16_t  loop = 0;
     if(WiFi.status() != WL_CONNECTED)
     {
         TRACE_WIFI_DISCONNECTED_SLEEP;
         WiFi.reconnect();
         delay(1000);
         if(WiFi.status() != WL_CONNECTED)
         {
            WiFi.begin();
         }
     }
     while(WiFi.status() != WL_CONNECTED)
     {
        Serial.println("Wifi disconnected. Need reconnect");
            Serial.println("failed to connect and hit timeout");
            delay(300);
             if(loop++ >= (3*15*60))// 15 min
             {
                 TRACE_WIFI_SLEEP;
                 ESP.restart();
                 delay(5000);
             }
     }

gfaman, 2022-08-17:
After few days on 2 different devices, results is soso. One device reset only one time per day. It is fine. But the other one, probably link to the Wifi router configuration, continuously reset after each light_sleep. Need fallback to delay() instead of light_sleep. No solution for that device. Only, before, by recall wifiManager.autoConnect after the sleep, it works sometimes.