tzapu/WiFiManager #1395: Store the value of mqtt server address(pubsubclient) from custom parameter and reuse it dynamically

shariq-azim, 2022-04-15:
#### Hardware
WiFimanager Branch/Release:  Master
Esp32

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

I am trying to set the default value in the custom parameter  first, but if there is a change of value i am trying to copy and update another variable that can be re-used as the mqtt server address. So that if the connection is lost it will try to connect again with the last known value. The following code runs one and connects properly during setup but if the connection is lost with MQTT server it wont connect as it would now have garbage value in it.

The obvious reason is because the previous value of the pointer is lost after the setup function exists.

I have tried so many variations but it is not working ,and unfortunately, the solutions in internet is mostly theory and not the code. Thus, i am reaching for the experts to help. Unfortunately, the solutions in internet is mostly theory and not the code.

Please help retrieve the value from a different function.

Version :2.03-alpha, 2.0.11 beta

Additional libraries: 
Pubsubclient
### Sketch

```cpp
#BEGIN

#include <WiFi.h>
#include <WiFiClient.h>
//#include <WebServer.h> 
#include <ESPmDNS.h> 
#include <Update.h> 

#include <PubSubClient.h>//0.4
//#include <HTTPClient.h> //0.4
#include <ArduinoJson.h> //0.4  // https://github.com/bblanchon/ArduinoJson
#include<string.h> //0.4
#include<cstring> //0.4
#include <WiFiManager.h>
#include <Preferences.h>

bool InternetStatus = false; 

// Your WiFi credentials.
const char* dnshost = "shariqESP"; //copiedota
char ssid[25] = "xxxxxxxx";
char pass[20] = "xxxxxxxxx";
char* mqttServer = "192.168.0.6"; //check actual config of node-red
//const int mqttPort = 1881;
char mqttPort[6] = "1881";
const char* mqttUser = "";
const char* mqttPassword = "";

String mqttServerPref;
String mqttPortPref;

String topicOTA = "ESP/OTAStatus/";
String topicStatus = "ESP/Status/";
String PiStatus = "PiServer/"; //"PiServer/x/Status"
String PiOTAInpStatus = "x/ESP/OTA/";
//String PiRestart ;//"PiServer/x/Status/Restart"

long lastReconnectAttempt = 0;

WiFiClient espClient; //0.4
PubSubClient pubsubclient(espClient);//0.4
Preferences preferences; //v0.5

void cloud_write(uint8_t pin, int pinValue)
{
[... 222 lines cut ...]
    if (pubsubclient.connected()) {
      Serial.println("connected with MQTT");
      //digitalWrite(R6, HIGH);
      pubsubclient.loop();
      InternetStatus =true;

    } else {
      Serial.print("Reconnect to MQTT server failed with state ");
      Serial.println(pubsubclient.state());
      //digitalWrite(R6, LOW);
      InternetStatus = setup_mqtt();

    }

  }
}
#END
```

### Debug Messages

```
messages here
```
Additional note if i do strcpy(mqttServer , (char*)custom_mqtt_server.getValue()); instead of   mqttServer = (char*)custom_mqtt_server.getValue(); //because datatype of mqttServer is char* it will go into panic mode

tablatronix, 2022-04-15:
hmm would think you would have to use a global char[] array and either set its size ahead of time, or dynamically allocate it , or just use a Strings

shariq-azim, 2022-04-15:
> hmm would think you would have to use a global char[] array and either set its size ahead of time, or dynamically allocate it , or just use a Strings

Good day to you.

I dont think pubsubclient.setserver accepts either char[] or string as an input parameter?

shariq-azim, 2022-04-17:
> hmm would think you would have to use a global char[] array and either set its size ahead of time, or dynamically allocate it , or just use a Strings

hello @tablatronix , i was just wondering if it is possible to set the value into a global variable from custom_mqtt_server.getValue() so that it can be used at later point? the step you provided earlier somehow didnt work for me..

tablatronix, 2022-04-17:
you can do something like the example

char mqtt_server[40];
strcpy(mqtt_server, custom_mqtt_server.getValue());

shariq-azim, 2022-04-19:
Thank you .it worked

closed 2022-04-19 by shariq-azim