tzapu/WiFiManager #1396: ESP32 wont connect to WiFi

sommerper, 2022-04-16:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  v2.0.10-beta  #9182622

Esp32: 

Hardware:			ESP Wroom 32

### Description

After connecting to the AP created by the ESP32 I select my network and enter the password. WiFiManager tries to connect 4 times and fails.

### Settings in IDE

Platformio.ini
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 921600
monitor_speed = 115200
lib_deps =
  https://github.com/tzapu/WiFiManager.git#9182622
```

```cpp
#BEGIN
#include <Arduino.h>
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup()
{
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  // it is a good practice to make sure your code sets wifi mode how you want it.

  // put your setup code here, to run once:
  Serial.begin(115200);

  // WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;

  // reset settings - wipe stored credentials for testing
  // these are stored by the esp library
  wm.resetSettings();

  wm.setHostname("Sensor2");
  wm.setConnectRetries(4);

  // Automatically connect using saved credentials,
  // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
  // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
  // then goes into a blocking loop awaiting configuration and will return success result

  bool res;
  // res = wm.autoConnect(); // auto generated AP name from chipid
  // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
  res = wm.autoConnect("ap", "12345678"); // password protected ap

  if (!res)
  {
    Serial.println("Failed to connect");
    // ESP.restart();
  }
  else
  {
    // if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
  }
}

void loop()
{
  Serial.println("In loop");
  delay(5000);
  // put your main code here, to run repeatedly:
}```

### Debug Messages

```
======================== [SUCCESS] Took 21.00 seconds ========================
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�*wm:[1] resetSettings 
*wm:[1] SETTINGS ERASED 
*wm:[1] AutoConnect 
*wm:[1] No Credentials are Saved, skipping connect 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  ap
*wm:[1] AP IP address: 192.168.4.1
*wm:[2] setting softAP Hostname: Sensor2
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started 
*wm:[2] WiFi Scan completed in 5515 ms
*wm:[1] 11 networks found
*wm:[2] AP:  -50 wlan4142412
*wm:[2] AP:  -51 ********************
*wm:[2] AP:  -59 ********************
*wm:[2] AP:  -69 ********************
*wm:[2] AP:  -77 ********************
*wm:[2] AP:  -77 ********************
*wm:[2] AP:  -83 ********************
*wm:[2] AP:  -84 ********************
*wm:[2] AP:  -86 ********************
*wm:[2] AP:  -91 ********************
*wm:[2] AP:  -95 *********************
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] NUM CLIENTS:  1
*wm:[2] <- HTTP WiFi save  
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connect Wifi, ATTEMPT # 1 of 4
*wm:[1] Connecting to NEW AP: wlan4142412
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_DISCONNECTED
*wm:[1] Connect Wifi, ATTEMPT # 2 of 4
*wm:[1] Connecting to NEW AP: wlan4142412
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_DISCONNECTED
*wm:[1] Connect Wifi, ATTEMPT # 3 of 4
*wm:[1] Connecting to NEW AP: wlan4142412
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_DISCONNECTED
*wm:[1] Connect Wifi, ATTEMPT # 4 of 4
*wm:[1] Connecting to NEW AP: wlan4142412
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_DISCONNECTED
*wm:[0] [ERROR] Connect to new AP Failed 
*wm:[2] Processing - Disabling STA 
*wm:[2] NUM CLIENTS:  1
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
[E][WebServer.cpp:633] _handleRequest(): request handler not found
*wm:[2] <- Request redirected to captive portal 
*wm:[2] NUM CLIENTS:  1
```

MythicalForce, 2022-04-24:
i have the same issue.. `wm.setHostname(  "testmachine" );` but i can only ping and connect to my ESP32 via IP and not hostname

tablatronix, 2022-04-24:
how is that the same issue?

tablatronix, 2022-04-24:
@sommerper change this to true and see if it connects, also enable esp debugging this sounds like a known esp issue

https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.h#L487

MythicalForce, 2022-04-24:
> how is that the same issue?

it was the wrong issue xD

sommerper, 2022-04-24:
@tablatronix I've been pulling my hair out for several days now. Tried 4 different routers and set up an openwrt network on my pi to see if it was the network somehow.

Turns out something is interfering with the esp32 wifi when connected to an external power source.

If I'm powering the esp32 with USB only there is not problem.
If I'm powering the esp32 with USB _and_ a 5V laboratory power source it doesn't work.

I can't pinpoint what it is but at this point I'm pretty sure it has nothing to do with this library and I'm sorry for taking your time.
If I find the reason for this error I will update the thread but I think we can close the issue.

Thanks! (for an awesome library)

tablatronix, 2022-04-24:
Interesting, I know analog read breaks wifi makes me wonder if it has something to do with wifi calibration and vref

OTHMANERAH, 2022-06-06:
HELP MEE PLEASE?????

Arduino : 1.8.9 (Windows 10), Carte : "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

c:\temp\arduino_modified_sketch_122643\sketch_jun06a.ino: In function 'void setup()':

sketch_jun06a:39:10: error: 'class WiFiClass' has no member named 'softAP'

     WiFi.softAP(ssid, password);

          ^

sketch_jun06a:41:23: error: 'class WiFiClass' has no member named 'softAPIP'

   IPAddress IP = WiFi.softAPIP();

                       ^

exit status 1
'class WiFiClass' has no member named 'softAP'

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

tablatronix, 2022-06-17:
suspected #1067

pa0o, 2023-02-28:
Wifimanager can not connect to Wlan  same here.
Using ESP32 Wroom 30 pin
both of them will not connect with Wlan saved credetial as the monitor text below shows

The same sketch  did work with  ESP8266  and an othe ESP32 wroom 38pin type
I have tried al i could think of and on a friends  PC with his IDE     old  IDE  1.8x  or new  2.04 
no clue yet 
awaiting an other mini ESP32 and hope it connects

serial monitor shows: 

*wm:[2] NUM CLIENTS:  1
*wm:[2] <- Request redirected to captive portal 
*wm:[2] NUM CLIENTS:  1
Now listening at IP 0.0.0.0, UDP port 4210

*wm:[1] AutoConnect 
*wm:[2] ESP32 event handler enabled 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: Eminent
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] [EVENT] WIFI_REASON:  2
*wm:[2] [EVENT] WIFI_REASON:  2
*wm:[2] [EVENT] WIFI_REASON:  2
*wm:[2] Connection result: WL_DISCONNECTED
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal

pa0o, 2023-02-28:
connectTimeout not set   is the complain...   could that be our common issue?
I did not have to set anything extra at WifiManager until now
question how to set this  "connectTimeout ?
is it Wifimanager.set connecttimeout=5  ore anything like that?
where are those hidden pameters documented?