tzapu/WiFiManager #1397: Change favicon

pienmatla, 2022-04-19:
Hi,

Is it possible to change the favicon of the WifiManager? I'd like to put my logo init.

Thanks,
pienmatla.

makeSmartio, 2022-04-19:
I haven't done it via the samples here, but you can set it with this line in Setup: 

  client.print("HTTP/1.1 301 Moved Permanently\r\nLocation: http://{ipAddress}/favicon.ico\r\n\r\n");

tablatronix, 2022-04-19:
Yeah there is a custom head html setter

tablatronix, 2022-04-19:
```c++

  const char* icon = "<link rel='icon' type='image/png' href='data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAADAAAAAwCAYAAABXAvmHAAADQElEQVRoQ+2YjW0VQQyE7Q6gAkgFkAogFUAqgFQAVACpAKiAUAFQAaECQgWECggVGH1PPrRvn3dv9/YkFOksoUhhfzwz9ngvKrc89JbnLxuA/63gpsCmwCADWwkNEji8fVNgotDM7osI/x777x5l9F6JyB8R4eeVql4P0y8yNsjM7KGIPBORp558T04A+CwiH1UVUItiUQmZ2XMReSEiAFgjAPBeVS96D+sCYGaUx4cFbLfmhSpnqnrZuqEJgJnd8cQplVLciAgX//Cf0ToIeOB9wpmloLQAwpnVmAXgdf6pwjpJIz+XNoeZQQZlODV9vhc1Tuf6owrAk/8qIhFbJH7eI3eEzsvydQEICqBEkZwiALfF70HyHPpqScPV5HFjeFu476SkRA0AzOfy4hYwstj2ZkDgaphE7m6XqnoS7Q0BOPs/sw0kDROzjdXcCMFCNwzIy0EcRcOvBACfh4k0wgOmBX4xjfmk4DKTS31hgNWIKBCI8gdzogTgjYjQWFMw+o9LzJoZ63GUmjWm2wGDc7EvDDOj/1IVMIyD9SUAL0WEhpriRlXv5je5S+U1i2N88zdPuoVkeB+ls4SyxCoP3kVm9jsjpEsBLoOBNC5U9SwpGdakFkviuFP1keblATkTENTYcxkzgxTKOI3jyDxqLkQT87pMA++H3XvJBYtsNbBN6vuXq5S737WqHkW1VgMQNXJ0RshMqbbT33sJ5kpHWymzcJjNTeJIymJZtSQd9NHQHS1vodoFoTMkfbJzpRnLzB2vi6BZAJxWaCr+62BC+jzAxVJb3dmmiLzLwZhZNPE5e880Suo2AZgB8e8idxherqUPnT3brBDTlPxO3Z66rVwIwySXugdNd+5ejhqp/+NmgIwGX3Py3QBmlEi54KlwmjkOytQ+iJrLJj23S4GkOeecg8G091no737qvRRdzE+HLALQoMTBbJgBsCj5RSWUlUVJiZ4SOljb05eLFWgoJ5oY6yTyJp62D39jDANoKKcSocPJD5dQYzlFAFZJflUArgTPZKZwLXAnHmerfJquUkKZEgyzqOb5TuDt1P3nwxobqwPocZA11m4A1mBx5IxNgRH21ti7KbAGiyNn3HoF/gJ0w05A8xclpwAAAABJRU5ErkJggg==' />";

  // set custom html head content , inside <head>
  wm.setCustomHeadElement(icon);
```

[image]

tablatronix, 2022-04-19:
You could also add a custom webhook and return an image from whereever, spiffs and proper header etc

closed 2022-04-19 by tablatronix

pienmatla, 2022-04-20:
I think I might be too new in C/C++ for this haha. Anyone who can explain what to do exactly to change it? :)

tablatronix, 2022-04-20:
Do you know how to encode the image ? (its web design)

this just adds this html into the <head> of the output
and needs to be set before anything else like autoconnect() etc

```
const char* customhtmlhead = "BLARGHHHHH";
wm.setCustomHeadElement(customhtmlhead);
```

pienmatla, 2022-05-29:
I know how to create a base64 image, i always use base64-image.de

But i dont know where to change any code haha, as in which file or something. 
Someone told me i can do it for you but i wanna learn it myself ofcourse :)

tablatronix, 2022-05-29:
I added an example

https://github.com/tzapu/WiFiManager/blob/master/examples/Super/OnDemandConfigPortal/OnDemandConfigPortal.ino#L147

pienmatla, 2022-05-29:
Hmm i cant find that line number in my code tho. I use the wifimanager whitin another project: https://github.com/ESPWortuhr/Wortuhr

And this is the icon i wanna use:
data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAABAAAAAQCAYAAAAf8/9hAAACZElEQVQ4y5WR30tTYRjHv+fwnvO87zlnO/7YdsYpydbUubnpNHI/nNNlE3NDwkzC6y7C6LaL8CIhiO67KC+EMuo/KJCkG/FWEC+C8iIoiArKCF3mTjebbKhUz933hc/n+fECAC5PTUn4z7pYKMgHwWNZD4dyOfGv8LlEwu9qaLh78KAKsam73euZbNb9N7g7Hj/Ndf0TI3pSJ2BEjjCMtz19fZ7j4Fg83kWa9k3hfOdMe3umTqAKUVI43+O6/iGRSp2sBQvFItpCoYQqxE9ViB+RaLQvlck01wmEYWy1hUJjCucl0rQvXd3d7QBwe24OwY6OUYXzXdK0r7F4vKM1EJhUOF88JLgzP4/OSCRV6bQdicXigWDwSmWyjz29vS1c14kR7Ry6gTCMrWoOR6MxVYjvCuclhWhfM4x3yXTaCwCMiDOi3apAxhEV6ezcME1zwXEcVVXVzXA43Lu2uvq5MD0Nn9ebBaDUAbUTZIeHJcu27zOisss0V5PptAYA12dnEQgGpxTO9xiRc+QK5/N52ePzLTAix2xqejGSz6sAcGlyEqcCgWsK5/vCMN4zot9HChTOGxnRfqPH82xgcJABwIXRUcluabnFiMq6273h8/ut2hvUCVQhGj2W9WC8WJQB4OrMjOS37XuMqGyY5tpQLqcxIjpWcLa/XypOTEgAMJjNyh7LelRZ52VqYIAqv3BYwHV9XXe731Tz2Pi43Oz1PlaFKDV7vc9H8nlW04yrQmyrQiwCAAOAZCKRlWVZerW8DABwuVyOz+e7EY1Gb56w7e2nS0vlqiCZTJYkx2mFJP16vbKCP8G9l0P6iF3cAAAAAElFTkSuQmCC

pienmatla, 2022-05-29:
So if someone has time and wanna have a look (bc i use the wifimanager in another project i mentioned above) that would be great. If not, i understand :) Let me know, thanks in advance :)

tablatronix, 2022-05-31:
What version of wm is it?

pienmatla, 2022-05-31:
The only thing I can find is @Version 0.0.0 in the WifiManager.cpp and WifiManager.h files.