tzapu/WiFiManager #1399: ESP32 "Guru Meditation Error" after inserting "#include <WiFiManager.h>" into sketch

jkrasinger, 2022-04-23:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  2.0.11-beta

Esp32: 

Hardware:			ESP32 DEV KIT V1

Core Version:      	ESP32 2.0.2

### Description

After adding "#include <WiFiManager.h>" to sketch without any other modifications to the Program get 
"Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled."

This error seems to happen in the NTPClient routines.

"

### Settings in IDE

Module:  			DOIT ESP32 DEVKIT V1

Additional libraries:  TFT, TFT_eSPI, NTPClient, PubSubClient, WiFiUdp

### Sketch

### Debug Messages

```
messages here
.....Attempting MQTT connection...MQTT connected
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d7177  PS      : 0x00060430  A0      : 0x800d8575  A1      : 0x3ffb2750  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800de964  A9      : 0x3ffb2760  
A10     : 0x3ffb27cc  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffc337c  
A14     : 0x3ffc39a4  A15     : 0x3ffb2774  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008974d  LEND    : 0x4008975d  LCOUNT  : 0xffffffff  

Backtrace:0x400d7174:0x3ffb27500x400d8572:0x3ffb2790 0x400d32af:0x3ffb27c0 0x400d365a:0x3ffb2800 0x400df03d:0x3ffb2820

tablatronix, 2022-04-23:
I need a sketch

jkrasinger, 2022-04-23:
I understand, but the sketch is a big one, should i really add it to the bug report?

tablatronix, 2022-04-23:
No you should use a miminal reproduction sketch, like an example from the library

jkrasinger, 2022-04-24:
Excuse me for the late response, but i tried to reproduce the error with a minimalistic sketch, but i was not able to reproduce it. I first assumed the problem within the WiFi routines, but this was not the case, they worked as expected. The next component to check was the TFT-Touch screen with the libraries TFT, TFT_eSPI in use. I deactivated the code step by step until i reached the "check for a touch" on the screen. As soon as i disabled this routine all was working as expected. So i made a deeper dive in this routine ... and found the bug in my code. I made a loop over 15(???) touch buttons instead of three. I have no idea why this happened. After correcting this little mistake the sketch was up and running again.
This sketch was working even with this error, but by only inserting the "#include <WiFiManager.h>" without using the WiFiManager(!) my sketch died. This is very interesting.

Excuse me for the false alarm, i will close this issue shortly.
Thank you for your quick response.

jkrasinger, 2022-04-24:
I just finished the full implementation of WiFiManager in my sketch and it works perfectly. Thanks you for the library and your intervention. Regards Jonny

closed 2022-04-24 by jkrasinger

tablatronix, 2022-04-24:
These errors are usually a memory issue accessing a bad pointeror something went out of scope