tzapu/WiFiManager #1401: Config mode in WiFi off mode

kargarisaac, 2022-04-25:
Hi,

Thank you for the great library. I want to use esp32 in a very power critical application. So I need to optimize power consumption as much as possible. I'm looking for a way to do the config mode in a more efficient way and not to be in wifi active mode for 2 to 3 minutes. I was thinking if you can guide me to do sth like below.

- Open captive portal in the same way (maybe wifi activate and after portal is up, disable wifi).
- then get the wifi credentials from user (maybe also some other inputs from user).
- then when user presses a button, wifi activates again and get and save credentials and maybe test if it can connect quickly or not.

Do you have any suggestion to do something like this?

Thank you in advance for the help

tablatronix, 2022-04-25:
Ideally for this you would require the user to do a special action or press a reset and only then use the captive portal.

why 2-3 minutes?

tablatronix, 2022-04-25:
wm.setConfigportalTimeout() controls how long to stay on

kargarisaac, 2022-04-25:
So how do you suggest to set wifi setting for battery device? It is not possible to hard code it and the user should be able to set the credentials based on his/her wifi. Right now the board goes to config mode when the user puts the battery in for 120 seconds and the user should set the wifi credentials in that time. Then it goes to deep sleep and every few hours, it wakes up and send sensed data from sensors to cloud. 

Any suggestion would be appreciated.

tablatronix, 2022-04-25:
ok so not sure how you can get any better than that

kargarisaac, 2022-04-29:
I thought about this and couldn't find a better solution to config wifi credentials and will continue with this for now.

closed 2022-06-17 by tablatronix