tzapu/WiFiManager #1403: setHostname does not set the host name

DonatelloX, 2022-04-29:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master v2.0.11-beta
ESP32 dev module

### Description
**setHostname** does not set the host name.

### Settings in IDE
Arduino IDE + ESP32 Core 1.0.6

### Sketch

```
  WiFi.mode(WIFI_STA);
  WiFiManager wm;
  std::vector<const char *> menu = {"wifi", "exit"};
  wm.setMenu(menu);
  wm.setHostname("MyTestWiFi");
  wm.setWiFiAutoReconnect(true);
  wm.setConfigPortalTimeout(120);
  wm.startConfigPortal();

  if(!wm.autoConnect("MyTestWiFi")) {
    Serial.println("WIFImanager: failed to connect :(");
  } else {
    Serial.println("WIFImanager: connected to " + wm.getWiFiSSID());
  }
```

### Debug Messages

```
*wm:[2] Starting Config Portal 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  ESP32_1BA4AE30
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[2] setting softAP Hostname: MyTestWiFi
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] Portal Timeout In 120 seconds
*wm:[2] Portal Timeout In 97 seconds
*wm:[2] Portal Timeout In 67 seconds
[...]
```
The host name is ESP32_1BA4AE30.

olonsoft, 2022-06-23:
Exactly the same problem here.
The same code on ESP8266 sets the hostname correctly.
But on ESP32 the hostname is "esp32-8071E0"

DonatelloX, 2022-06-23:
Thank you @tablatronix for reply.
How can I set the SSID name?

olonsoft, 2022-06-24:
@tablatronix 
> Thats the SSID, not the hostname

There must be a misunderstanding here. As you see in @DonatelloX code, he sets hostname `wm.setHostname("MyTestWiFi");`.
When the ESP connects to wifi successfully, the router should show `MyTestWiFi` as a connected client. But is shows `esp32-8071E0`.
This only happens on ESP32. The same code on ESP8266 shows `MyTestWiFi`.

mindforger, 2022-06-24:
the code is a bit confusing, so i might have misread it, but can you try calling setupHostname(restart=False) after setHostname ? it looks like this is missing in the ap start sequence for ESP32

olonsoft, 2022-06-24:
> 

` bool setupHostname(bool restart);` is a private function. Unless you mean to modify the source code.

mindforger, 2022-06-24:
yeah, please try that (making it public temporale). I am curious and i don't have a esp32 before weekend to test myself

olonsoft, 2022-06-24:
> yeah, please try that (making it public temporale). I am curious and i don't have a esp32 before weekend to test myself

I 've tried what you suggest `setupHostname(true/false)`. Same result.

mindforger, 2022-06-24:
even the debug output the same? that sounds odd!

olonsoft, 2022-06-24:
No. The code is not missing.

Same code for ESP32 and ESP8266.

ESP32:
```
*wm:[1] AutoConnect
*wm:[2] ESP32 event handler enabled
*wm:[2] Setting Hostnames:  My-IoT-App
*wm:[2] Setting WiFi hostname
*wm:[2] Connecting as wifi client...
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to SAVED AP: MyTestAP
*wm:[1] connectTimeout not set, ESP waitForConnectResult...
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS
*wm:[2] Connected in 827 ms
*wm:[1] STA IP Address: 192.168.1.231
```

ROUTER:
```
MAC-Address	        Host
30:AE:A4:80:71:E0	esp32-8071E0 (192.168.1.231)
```

ESP8266:
```
*wm:[1] AutoConnect
*wm:[2] Setting Hostnames:  My-IoT-App
*wm:[2] Setting WiFi hostname 
*wm:[2] Connecting as wifi client...
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to SAVED AP: MyTestAP
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS
*wm:[2] Connected in 6422 ms
*wm:[1] STA IP Address: 192.168.1.241
```

Router:
```
MAC-Address	        Host
18:FE:34:D5:3F:AF	My-IoT-App (192.168.1.241)
```

olonsoft, 2022-06-24:
Looking at the logs, I suspect that the problem might be with the autoconnect feature of ESP32. I see that the connect time of ESP32 is only 827ms. I am thinking that as soon as the ESP32 restarts, it automatically connects to WiFi, BEFORE my code reaches the `sethostname(xxx)` function.
My code blinks a led for a second on boot before calling `wm.autoconnect(ap,psw)` and that time might be enough for ESP to connect before any other call to wifi functions.
I'll investigate it further.

tablatronix, 2022-06-24:
I swear I test this every other week..
Must be a esp issue, cause I just went through this in another issue, let me check again

tablatronix, 2022-06-24:
yup definitely an issue, I think there is some issue with when this is called or a race condition.
testing now

tablatronix, 2022-06-24:
yup ok so there seems to be something going no across versions.
My notes say sethostname MUST be set after sta starts, BUT another issue hinted and I just tested it, that it must be set BEFORE sta starts. SOOOOO

I need to find out if this changed or what

*wm:[1] ESP SDK version:  v4.4-beta1-189-ga79dc75f0a

tablatronix, 2022-06-25:
https://github.com/tzapu/WiFiManager/issues/1400#issuecomment-1111682560

What version sdk you all on ?

tablatronix, 2022-06-25:
I pushed a test fix, I need to test esp8266 and see if it broke, and older esp32 libs

olonsoft, 2022-06-25:
I am on v4.4.1-1-gb8050b365e
I can't test it before Monday. I'll keep you informed.

olonsoft, 2022-06-27:
It works now for both ESP8266 and ESP32.
BUT I had to remove the line ` WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP ` from the `setup()` as suggested in most of the examples.
With that line included, the hostname does not set.

tablatronix, 2022-06-27:
Yeah i am looking into that also , it seems hostname MUST be set before wifi is enabled, used to be connected. I need to find out why this changed in the IDF or if there is just some bug in esp we can fix. If not I can find another wya, I just have ot be careful to not stop and start stuff if it causes even a 500ms delay in connections.

PeterHouseJr, 2022-07-29:
I have the same issue with inability to set the Hostname.  The Hostname seems to be the AP Hostname when the configuration was saved.

Removing the WiFi.mode( WIFI_STA ); had no effect.

Otherwise, WiFiManager has been wonderful and saved me a lot of time.

tablatronix, 2022-07-29:
hmm, still not working? I pushed a fix for this, are you using latest git?

PeterHouseJr, 2022-08-01:
I did not know there was a fix and am using 2.0.9.  I will update and test and let you know . . .

Thank you.

tablatronix, 2022-08-01:
sorry ill try to cut a new beta

PeterHouseJr, 2022-08-01:
I will wait for your new beta.

PeterHouseJr, 2022-08-02:
I tried the 2.0.11beta firmware and still no success setting the Hostname.  After calling the setHostname function the getHostname function returns the correct setHostname string but the actual Hostname on the wire is the original AP Hostname.

I tried the following code at the end of Setup() and still no success.
`
  Serial.println( "Setup(): set Hostname and Restart" );
  WiFi.setHostname( Hostname );
  Serial.print( "   Set AP Hostname Set to: " );
  Serial.println( Hostname );
  Serial.print( "   Get AP Hostname: " );
  Serial.println( WiFi.getHostname() );

  Serial.println( "   AP soft Restart" );
  WiFi.reconnect();
`

I also cleared the lease from the router to make sure the router is not caching the Hostname and also confirmed using ping Hostname from the command line.

tablatronix, 2022-08-03:
Ill try to do it tonight

PeterHouseJr, 2022-08-09:
Do you have an idea what or where the problem is located?  Is it in the WiFiManager library or is the problem deeper in the WiFi library?  How can I help?

tablatronix, 2022-08-09:
Sorry I got pretty sick this week, I will make a beta tomorrow, the only change was moving the hostname call in autoconnect, you can see the commit above

PeterHouseJr, 2022-08-16:
I am still having a Hostname issue.

When I start the device from an unconfigured state (AP Mode):

Hostname = 'esp32s2-69A284'
SSID = 'esp32-34B47269A284'
IP = 192.168.4.1

After entering the local WiFi info (STA Mode):

Hostname = 'esp32s2-69A284'
Hostname should be 'ELC-69A284'
IP = 192.168.99.106

Nothing I can do will change the Hostname.

I guess the Hostname, esp32s2-69A284, is from somewhere in the firmware.  It is certainly not a string that appears anywhere in my code.

tablatronix, 2022-08-17:
Is this using latest git or beta .12?

PeterHouseJr, 2022-08-17:
The project is using the latest .12-beta you made available a couple of days ago.

The project was started from an example file from 'Dronbot Workshop'.  Everything works great except setHostname.

The development environment is using VSCode and Platformio.  Happy to post my source file at your request or if you would like to suggest I use one of the example files from the git page I would be happy to try that out.

Have previously tried many different combinations of intermixing WiFi.h and WiFiManager.h library calls with no success.  The Hostname is always the same and I check it using both my DD-WRT router which provides the DHCP and confirm using the command line 'ping Hostname' which has only ever responded to the Hostname esp32s2-69A284.

Thank you for your time and effort creating, maintaining, and troubleshooting this library.

tablatronix, 2022-08-17:
I will test again with a basic example, and with an s2.

What esp32 version you using?

PeterHouseJr, 2022-08-17:
ESP32-S2-WROOM from DFRobot DFR0743-WROOM.  

Here is some setup info:

```
platform = espressif32
board = esp32-s2-saola-1
framework = arduino
lib_deps = 
	bblanchon/ArduinoJson @ ^6.19.3
	sstaub/Ticker@^4.4.0
	ottowinter/AsyncMqttClient-esphome@^0.8.6
```

The setup is intended for a Adafruit saola board based on the esp32-s2-wrover module.  I am using an esp32-s2-wroom module on my custom pcb since the Adafruit parts were out of stock when I began this project.  As far as I can tell, the only difference is the WROVER has extra external serial RAM, PSRAM, which the WROOM does not have.

Please let me know which basic example and I will test also.

tablatronix, 2022-08-17:
Confirmed, broken here, let me check esp32 again and make sure they didn't change something

tablatronix, 2022-08-18:
#1068 

Added a fix, hostname must be set BEFORE any wifi on some versions of IDF, and it seems to differ last time i fixed this.

SO DO NOT setmode() in your code and try this last commit and see if it works

PeterHouseJr, 2022-08-18:
I have tried the WiFi.setHostname placed as soon as possible after the first call to WiFi to connect the Event messages to an event handler.  It is also as soon as possible after the first call to wm which gets the MAC address so I can create the Hostname for the device.

I will try out the new library in a couple of hours . . .

PeterHouseJr, 2022-08-18:
I made the changes to my 2012-beta files as you suggested.  

The reported messages on my debug terminal now seem to show the Hostname has been changed.

The device has changed to a different default Hostname: esp32s2-02003F as shown on the Router and verified by command line ping.

Here is a list of the debug messages.  Everywhere in debug messages where I read the Hostname, I now read it back from both the WiFi and wm.  The debug messages show this and seem to show the new Hostname has been accepted although both the Router and command line ping show otherwise.

Here are the Debug Messages:
```
*---*---*---*---*---*---*
* Startup
*---*---*---*---*---*---*
* Version0.5.r3.b84 - 2022-08-18 11:42:30
*---*---*---*---*---*---*
Setup(): Begin
Setup(): Setup Switch NOT Pressed
loadConfigFile() Reading ...
   Mounting File System...
loadConfigFile: mounted file system
   Reading Config
   Opened Config File
   JSON Pretty Result:
{
  "HostnamePrefix": "ELC",
  "MQTTServer": "192.168.99.29"
}
   JSON Parsed Correctly
Setup(): 
   MQTT Server:   192.168.99:29
   Port:1883
   WiFi.getHostname(): esp32s2-02003F
   wm.getWiFiHostname(): esp32s2-02003F
*wm:[2] Added Parameter: HostnamePrefix
*wm:[2] Added Parameter: MQTTServer
Setup(): forceConfig is False
*wm:[1] AutoConnect
WiFiEvent: WiFi Ready
   WiFi.getHostname(): esp32s2-02003F
   wm.getWiFiHostname(): esp32s2-02003F
WiFiEvent: station start
   MAC: 34:B4:72:69:A2:84
   Setting Hostname to: ELC-69A284
   WiFi.getHostname(): esp32s2-02003F
   wm.getWiFiHostname(): esp32s2-02003F
*wm:[2] Setting Hostnames:  ELC-69A284
*wm:[2] Setting WiFi hostname
*wm:[2] ESP32 event handler enabled
*wm:[2] Connecting as wifi client...
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to SAVED AP: theCatinthehatstrikesback2
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
WiFiEvent: station connected to AP
WiFiEvent: station got IP from connected AP
   Connected to: theCatinthehatstrikesback2
   RRSI: -53
   WiFi.getHostname(): ELC-69A284
   wm.getWiFiHostname(): ELC-69A284
   IP Address: 192.168.99.105
   MAC Address: 34:B4:72:69:A2:84
MQTT: Connecting ...
MQTT: Connected
MQTT: Session present: 0
MQTT: Publishing stat Boot Message
MQTT: Subscribe acknowledged.
MQTT: packetId: 1
MQTT: qos: 0
MQTT: Subscribe acknowledged.
MQTT: packetId: 2
MQTT: qos: 0
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS
*wm:[2] Connected in 6351 ms
*wm:[1] STA IP Address: 192.168.99.105
Setup(): HostnamePrefix: ELC
Setup(): MQTTServer: 192.168.99.29
Setup(): set Hostname and Reconnect
   Set AP Hostname Set to: ELC-69A284
   WiFi.getHostname(): ELC-69A284
   wm.getWiFiHostname(): ELC-69A284
   WiFi.reconnect()
WiFiEvent: station disconnected from AP
*wm:[2] [EVENT] WIFI_REASON:  8
MQTT: Disconnected
   WiFi.getHostname(): ELC-69A284
   wm.getWiFiHostname(): ELC-69A284
loop() Begin Forever Loop
WiFiEvent: station connected to AP
WiFiEvent: station got IP from connected AP
   Connected to: theCatinthehatstrikesback2
   RRSI: -52
   WiFi.getHostname(): ELC-69A284
   wm.getWiFiHostname(): ELC-69A284
   IP Address: 192.168.99.105
   MAC Address: 34:B4:72:69:A2:84
MQTT: Connecting ...
MQTT: Connected
MQTT: Session present: 0
MQTT: Publishing stat Boot Message
MQTT: Subscribe acknowledged.
MQTT: packetId: 1
MQTT: qos: 0
MQTT: Subscribe acknowledged.
MQTT: packetId: 2
MQTT: qos: 0
```

tablatronix, 2022-08-18:
Yeah unfortunately there is no way to get the actual hostname from the IDF, the esp library just returns what you last tried to set it to ( its just a local variable ) And mayh or may not actually get set in the wifi SDK. afaik it is not exposed in the IDF

PeterHouseJr, 2022-08-18:
I am surprised the reconnect() did not work since it should request the DHCP all over again afaik.

tablatronix, 2022-08-19:
Ok I am testing some fixes to make sure this always works, good news it does not add too much to connection time to reset

PeterHouseJr, 2022-08-19:
I look forward to your edits and will try them in the morning (in Sunny, Rain Soaked NE Florida).

tablatronix, 2022-08-19:
I need to test older IDF and esp32, I have a feeling they will need custom code to set hostname AFTER wifi_sta is up and not before

PeterHouseJr, 2022-08-25:
Have you had any success?  Is there anything I can do to help?

tablatronix, 2022-08-26:
Is it not working?

PeterHouseJr, 2022-08-27:
Thank you.  I missed the commit.  I will test this on Monday and let you know.

PeterHouseJr, 2022-08-29:
I had an issue while applying the changes to WiFiManager.cpp

The line 3610 showing a change of only comments was found in my WiFiManager at line#3530 and the surrounding code lines matched.

This tells me there are either some lines missing or something does not match.

I am defficient in the use of .git.  How can I make sure I have the files with your latest, correct, changes?

I tried the changes as is and as far as I can tell using my pfsense router, Advanced IP Scanner and the command line, there is now a zero length Hostname and the device is only ping-able by IP.

PeterHouseJr, 2022-08-29:
Wait - I copied the entire contents of the WiFiManager from github to my editor, line 3612 contains the correct code, and am recompiling.

Same results. 0 lenght Hostname.

I believe the failure is probably now my code which needs refactoring after all the tail chasing I have done to try and make this work.

Please let me know which example I should try and How can I download the entrire WiFiManager library after the last three commits?

tablatronix, 2022-08-30:
I have no idea , you should be testing with this library master branch

PeterHouseJr, 2022-08-30:
When I get the library master branch, it does not have the changes.  I know I must be doing something wrong.  Can you give me a .git command line to get the correct branch?

PeterHouseJr, 2022-08-30:
@tablatronix Thank you for your help.  

The latest commit to master works and it works well.

Thank you again for your time and I hope I have contributed to making the library better without wasting too much of your time.

Erriez, 2022-09-12:
Related to: https://github.com/tzapu/WiFiManager/pull/1483#issuecomment-1243526723

tablatronix, 2022-11-28:
I think this is fixed now, there may be some wrong behavior in older IDF still, shrug

closed 2022-11-28 by tablatronix

Erick20000, 2022-12-20:
sethostname: usa o miniaplicativo 'Rede' no 'Painel de controle' para definir
o nome do host.

Erick20000, 2022-12-20:
sethostname: usa o miniaplicativo 'Rede' no 'Painel de controle' para definir
o nome do host.
alguém pode me orientar a fazer isso por favor?

tinkering4fun, 2024-12-21:
I observed the same symptom on my ESP32 D1 R32 board, and came accross this issue thread while looking for a solution.
rem: I still run Arduino 1.8.19 and  esp32  2.0.17 and did not like to update immediately.

Found this workaround for the symptom:
In function setup() add a  `WiFi.disconnect();` call, just before initializing  `WiFiManager wm;` 
Of course, this approach may not be desirable in all use cases.

btw: Thank You for providing this library!