tzapu/WiFiManager #1408: strings_en.h when using WIFI_MANAGER_OVERRIDE_STRINGS how to include mystrings_en.h

ben700, 2022-05-03:
I was expecting to be able to do something like 

	-iquote wifi_strings
	-D WIFI_MANAGER_OVERRIDE_STRINGS
	-include "wifi_strings/myStrings_en.h"

I've tried #include in main.cpp 

Can you not add two lines in strings_en.h as below
**#ifdef WIFI_MANAGER_OVERRIDE_STRINGS
    #include WIFI_MANAGER_OVERRIDE_STRINGS**
#ifndef WIFI_MANAGER_OVERRIDE_STRINGS

tablatronix, 2022-05-03:
-DWIFI_MANAGER_OVERRIDE_STRINGS 
?

ben700, 2022-05-04:
So adding 
-DWIFI_MANAGER_OVERRIDE_STRINGS
Stops the strings being included from strings_en.h but then I need to include the new versions of the strings say that file is called myStrings_en.h and stored in my_strings_dir

That's why I expected this to work:-
 
-iquote my_strings_dir
-D WIFI_MANAGER_OVERRIDE_STRINGS
-include myStrings_en.h

tablatronix, 2022-05-04:
you cannot include from a directory afaik, so just name it different

ben700, 2022-05-05:
arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from .pio/libdeps/droneInstall/WiFiManager/WiFiManager.h:81,
                 from .pio/libdeps/droneInstall/WiFiManager/WiFiManager.cpp:13:
.pio/libdeps/droneInstall/WiFiManager/WiFiManager.h:525:51: error: 'S_brand' was not declared in this scope; did you mean 'srand'?
  525 |     String        _title                  = FPSTR(S_brand); // app title -  default WiFiManager

ben700, 2022-05-05:
Adding these lines to strings_en.h works but not when I add to main.cpp

#ifdef WIFI_MANAGER_OVERRIDE_STRINGS
 #include "strings_es.h"
#endif

tablatronix, 2022-05-05:
Ill add a filename directive

ben700, 2022-05-05:
Yes, 

https://github.com/ben700/WiFiManager_1408 I'll tag files so you know we have same understanding

ben700, 2022-05-05:
https://github.com/ben700/WiFiManager_1408/tree/add-include

In strings_en if you change the line 16 i think 
 
#ifndef WIFI_MANAGER_OVERRIDE_STRINGS

  to 
 
#ifdef WIFI_MANAGER_OVERRIDE_STRINGS
    #include WIFI_MANAGER_OVERRIDE_STRINGS
#else

That would reduce people like me asking

georgevbsantiago, 2022-05-13:
Hi, I'm also researching the best strategy to translate the content of "strings_en.h" to Portuguese/Brazil with a "strings_pt-br.h".

A strategy that:
- Allow updating the html text, without changing the code of "strings_en.h";
_The solution proposed by @ben700  solves this challenge._

- Do not cause a bug, if "strings_en.h" adds a new html text, which is not foreseen in "strings_pt-br.h".
_I don't know if there is a strategy to avoid this bug. One option would be WiFiManager to preserve the original text of "strings_en.h", in case a substitute text is not defined in "strings_pt-br.h". If there's a way to do that, it wouldn't cause bugs in a future update. The developer would only have to update the text.
But I don't know if it's a good strategy, since an error caused by the compiler can be something desirable for the developer to update his code in "strings_pt-br.h"._

ben700, 2022-05-14:
Are you thinking something like 

**strings_base.h**
class StringBase
{
public:
	const char *WM_VERSION_STR PROGMEM = "v1.0.11-beta";
        const char HTTP_HEAD_START[] PROGMEM = "<!DOCTYPE html>"
                                       "<html lang='en'><head>"
                                       "<meta name='format-detection' content='telephone=no'>"
                                       "<meta charset='UTF-8'>"
                                       "<meta  name='viewport' content='width=device-width,initial-scale=1,user-scalable=no'/>"
                                       "<title>{v}</title>";
}

strings_en.h
class Strings : public StringBase
{
public:

	const char *WM_VERSION_STR PROGMEM = "v1.0.11-strings_en";
};

strings.h
#ifdef WIFI_MANAGER_OVERRIDE_STRINGS
#include WIFI_MANAGER_OVERRIDE_STRINGS
#else
#include "strings_en.h"
#endif

Strings *strings = new Strings();
const char *WM_VERSION_STR PROGMEM =strings->WM_VERSION_STR;

camilo-sx, 2022-08-05:
Hello @ben700 ! Did you manage to include  "strings_en.h" properly? 
I've seen your code but I still don't understand how you include that.
You have the same "strings_en.h" in _src_, _wifi_strings_ and _lib>WiFi_Manager_ folders so I'm a little confused.

When you use in platformio.ini :
build_flags =
    -iquote wifi_strings
	'-DWIFI_MANAGER_OVERRIDE_STRINGS="strings_es.h"'

Which file are you pointing to?

I would really appreciate your help if you could explain it to me simply c:
Thanks in advance.

ben700, 2022-08-05:
Yes

ben700, 2022-08-05:
I couldn't get lipped to work so created my own folder called Wi-Fi strings.

Look at the tag add-include

https://github.com/ben700/WiFiManager_1408/tree/add-include

Ben

ben700, 2022-08-05:
In the repository there's two tags.

Master shoes as is.
https://github.com/ben700/WiFiManager_1408/tree/master

Add-include shows the change.
https://github.com/ben700/WiFiManager_1408/tree/add-include

Ben

tablatronix, 2022-08-09:
I have a test branch i am working on a better solution, I will try to get it posted soon

tablatronix, 2022-08-09:
https://github.com/tzapu/WiFiManager/pull/1474

raomin, 2023-03-15:
Ouch... sorry but this clearly needs some documentation.

After quite some struggle, here is what I came with that seems to work.
In markdown so maybe it can be added to the Readme
 
##Customizing interface

All html is contained in a single file `wm_strings_en.h`. To customize the html code, copy this file into your project folder (or the `/include` folder), give it a custom name, eg `wm_custom.h` and comment out these two lines:
```c
//#ifndef WIFI_MANAGER_OVERRIDE_STRINGS 
```
and
```c
// END WIFI_MANAGER_OVERRIDE_STRINGS
//#endif //<--- comment this one 
```

Make you changes in this file.

Then add these build flags to your `platformio.ini` file:
```
build_flags = -DWIFI_MANAGER_OVERRIDE_STRINGS
			  '-D WM_STRINGS_FILE="wm_custom.h"'
```

tablatronix, 2023-03-16:
you should not need to use WIFI_MANAGER_OVERRIDE_STRINGS at all now

tablatronix, 2023-03-16:
4c0d7fa77780558ee058ce5335a6bb585db96fd9

closed 2023-03-16 by tablatronix