tzapu/WiFiManager #1410: Can I use it with ESPAsyncWebServer?

baloyou, 2022-05-04:
i like this library, simple and stable.

but i need to used it with ESPAsyncWebServer.

how to do it?

MaralS, 2022-05-07:
Hello,

I had to give up because of the conflicts between with the two library.
I'm still working on it to find a solution to use both but it's frustrating.

tablatronix, 2022-05-07:
There is an async branch that is being worked on atm, eventually I hope to combine it into main lib but the functionality is totally different

Darkest, 2022-05-07:
What do you mean by saying "totally different functionality"?

tablatronix, 2022-05-08:
its async, so all callback methods need to be also with access to the server res obj

MaralS, 2022-05-13:
> its async, so all callback methods need access to the server res obj. also there seems to be some ota differences that need to be handled differently

It's good to know !

tablatronix, 2022-05-13:
I will work on this in a week or so, async branch might be behind master, i am gonna redo it from scratch in a bit as I am restructuring stuff to see if we can just change all wm methods to use a arg templates for hooking into different servers

baloyou, 2022-07-24:
> have you found any solution.
> i am also tryig to do the same but no luck

i don't have a solution either

Pratched, 2023-03-15:
Hello, I tried this branch mentioned here with my ESP8266 as I am in need of an AsyncWebServer...

Originally I started of with the master branch of WifiManager and adding the credentials worked quite well.
However later on (after I saved and didnt delete the credentials) I switched to the Async capable branch, which was working alright so long
That is to say, until I did reset the WiFi credentials and suddenly could no longer enter the credentials when using the async-branch. I switched back to using the master branch and everything works very well again.  See below for more detail...is there anything I could do, to use WiFi-Manager WITH Async as of now with its essential capabilities?

When using the Async-Branch without saved credentials, you would connect to the AP and hit "Update Wifi". But If you do so the following serial output is printed:

```
...
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started 

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fffeae0 end: 3fffffb0 offset: 01a0
3fffec80:  00000000 3ffef5c0 3ffef258 3ffeffe0  
...
```
 which is ended by restarting the ESP (and print out in serial monitor like this:
```
...
3fffffa0:  00000000 3f007472 00000000 3ffef73c  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0005b520
~ld

*wm:[1] AutoConnect 
*wm:[1] No Credentials are Saved, skipping connect 
*wm:[2] Starting Config Portal 
...
```

tablatronix, 2023-03-15:
I will take a look, I need to update that branch anyway

Pratched, 2023-03-15:
FYI: on other occasions I had similar issues with code snippets that would block the ESP like `delay(5000)` even...

Giermann, 2024-12-10:
Is there any progress?
It seems most of the repositories are not being updated, at least they have not been updated during the last 2 years.
I am simply looking for a small implementation of non-blocking WiFiManager, OTA and a common Web interface, no matter if connected to an existing WiFi or acting as an AP - while the ESP itself is doing other, more important tasks without the need of a network connection.
And I was told, that ESPAsyncWebServer should be preferred over the blocking WebServer.

Do I have to implement everything on my own or is there a library available that I did not find yet?

StefanoTesla, 2025-06-25:
> Is there any progress? It seems most of the repositories are not being updated, at least they have not been updated during the last 2 years. I am simply looking for a small implementation of [...]

Did you found a solution??

Giermann, 2025-06-25:
No, not yet.

alexhopeoconnor, 2026-09-09:
For anyone still looking for a maintained answer to this, I have opened #1873 to discuss a new-major-version path for async portal support. The maintained fork linked there uses ESPAsyncWebServer with ESP8266 and ESP32 transport dependencies, but it intentionally has breaking portal/API changes; the issue is to establish whether a documented migration path and restored translations could make that direction viable upstream.