tzapu/WiFiManager #1415: cannot detect wifi using board but see it with my phone

kargarisaac, 2022-05-11:
Hi,

I'm trying to use wifimanager and connect my board to wifi. I see the wifi in my phone but cannot find it in wifimanager list of Wifis. I changed `setMinimumSignalQuality` but it didn't help. The WiFi is weak at that spot but my phone can find it and ESP cannot. What do you think is the difference and what can I do to improve ESP performance in terms of finding weak WiFi access points and send data through them?

Thank you

tablatronix, 2022-05-11:
Does it find it if you go closer to it?

kargarisaac, 2022-05-12:
> Does it find it if you go closer to it?

You mean phone closer to esp or esp closer to wifi access point? first case no, second one yes. In some points of the apartment, wifi strength is not good. My phone can detect wifi there but esp cannot. My question is: can I solve this with software or is it a hardware problem, antenna or sth else?

tablatronix, 2022-05-17:
umm not sure, sounds like an esp question. see about making sure your power supply is strong, big caps etc. use a board with an external antenna connector or chip antenna maybe, much better than a pcb antenna.