tzapu/WiFiManager #1417: Disable Save button until user fills ssid, pass, and also custom input fields

kargarisaac, 2022-05-13:
Hi,

Is it possible to disable the save button until user fills the input fields? I have added also some custom input fields using:

```C++
WiFiManagerParameter email_text_box("email", "Enter your email here", "", 50);
wm.addParameter(&email_text_box);
```

kargarisaac, 2022-05-14:
I tried to modify the `strings_en.h` file like this:

```C++
const char HTTP_SCRIPT[]           PROGMEM = "<script>function c(l){"
"document.getElementById('s').value=l.getAttribute('data-ssid')||l.innerText||l.textContent;"
"p = l.nextElementSibling.classList.contains('l');"
"document.getElementById('p').disabled = !p;"
"if(p)document.getElementById('p').focus();};let inputs = document.querySelectorAll('input');"
"let buttonSend = document.getElementById('save-btn');"
"let inputValidator = {'s': false,'p': false, 'email': flase};"
"inputs.forEach((input) => {input.addEventListener('input', () => {let name = event.target.getAttribute('name');"
"if (event.target.value.length > 0) {inputValidator[name] = true;} else {inputValidator[name] = false;};"
"let allTrue = Object.keys(inputValidator).every((item) => {return inputValidator[item] === true});"
"if (allTrue) {buttonSend.disabled = false;} else {buttonSend.disabled = true;}})})</script>"; // @todo add button states, disable on click , show ack , spinner etc

.
.
.

const char HTTP_FORM_END[]         PROGMEM = "<br/><br/><button type='submit' id='save-btn' disabled>Save</button></form>";
```

But I cannot see any changes there. Also, I tried to just disable the save button using html but I don't see any change there too. around lines 70-80 in `strings_en.h` file:
```C++
const char HTTP_FORM_END[]         PROGMEM = "<br/><br/><button disabled>Save</button></form>";
```

Am I missing sth?

tablatronix, 2022-05-14:
you might have to add css for disabled state

kargarisaac, 2022-05-15:
When I remove the style part, I see that the button is disabled. But I cannot enable it using javascript. Do you think I put the code in a right place? I remove the custom input box and just use ssid and password inputs but no result yet.

tablatronix, 2022-05-15:
disabled style was only for inputs, I updated it, pull master
https://github.com/tzapu/WiFiManager/commit/c56eec9018f93e491011bfa374fb944ee26a7ef7

tablatronix, 2022-05-15:
I will have the strings reworked later toda with the easier includes also

kargarisaac, 2022-05-15:
I used the code from [here](https://javascript.tutorialink.com/how-to-disable-submit-button-until-all-mandatory-fields-are-filled-using-html-and-vanilla-js/).

My code:
added `id` for save button in cpp code.
```C++
const char HTTP_FORM_END[]         PROGMEM = "<br/><br/><button type='submit' id='save-btn' disabled>Save</button></form>";
```
and appended the following js code to the `HTTP_SCRIPT[]` code inside the function:
```js
let inputs = document.querySelectorAll('input');
let buttonSend = document.getElementById('save-btn');

let inputValidator = {
  "s": false,
  "p": false
}

inputs.forEach((input) => {
  input.addEventListener('input', () => {
    let name = event.target.getAttribute('name');
    if (event.target.value.length > 0) {
      inputValidator[name] = true;
    } else {
      inputValidator[name] = false;
    };

    let allTrue = Object.keys(inputValidator).every((item) => {
      return inputValidator[item] === true
    });

    if (allTrue) {
      buttonSend.disabled = false;
    } else {
      buttonSend.disabled = true;
    }
  })
})
```

kargarisaac, 2022-05-15:
Seems that the problem is putting script in head. When I put it at the end of body, it works in the html template. In cpp file, it is a bit complex. I don't know yet the logic there and have to go deeper in the code to change it.

tablatronix, 2022-05-15:
maybe add an onload event wrapper ?

kargarisaac, 2022-05-16:
Thanks for the tip. Now it works.  Here is the code in html template:

```html
<!-- HTTP_HEAD -->
<!DOCTYPE html>
<html lang="en">
	<head>
		<meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/>
		<title>{v}</title>
<!-- /HTTP_HEAD -->
<!-- HTTP_STYLE -->
<style>

:root{
	 /* CSS VARIABLE THEME COLOR */
	/*--primarycolor:#9933CC;*/
	--primarycolor:#1fa3ec;
}

body.invert,
body.invert a,
body.invert h1 {
	background-color:#060606;
	color:white;
}
body.invert .msg{
	background-color: #282828;
	border-top: 1px solid #555;
	border-right: 1px solid #555;
	border-bottom: 1px solid #555;
	color:#fff;
}
body.invert .q[role=img] {
    -webkit-filter: invert(1);
    filter: invert(1);
}
.c,
body {
    text-align: center;
    font-family: verdana
}
.wrap {
	text-align:left;
	display:inline-block;
	min-width:260px;
	max-width:500px;
}
div,
input,select {
    padding: 5px;
    font-size: 1em;
    /*width: 100%;*/
    /*/*margin: 3px 0 10px 0; // BEFORE spacing shows label closer to input, but need to do same for AFTER*/
    margin: 5px 0;
	box-sizing: border-box;
}
div{
	margin: 5px 0;
}
input,button,select,.msg{
	border-radius:.3rem;
    width: 100%;
}
input[type=radio],input[type=checkbox]{
    width: auto;
}
button,input[type="button"],input[type="submit"] {
    border: 0;
    background-color: var(--primarycolor);
    color: #fff;
[... 343 lines cut ...]
  			<input style='display: inline-block;' type='radio' id='choice2' name='program_selection' value='2'>
  			<label for='choice2'>Choice2</label><br>

  			<!-- INPUT SELECT -->
			<br/>
			<label for='input_select'>Label for Input Select</label>
			<select name="input_select" id="input_select" class="button">
			<option value="0">Option 1</option>
			<option value="1" selected>Option 2</option>
			<option value="2">Option 3</option>
			<option value="3">Option 4</option>
			</select>

			</form>

			<!-- /FORM_EXTRAS -->

<!-- HTTP_END -->
    </div>

	
	</body>
</html>
<!-- /HTTP_END -->
```