tzapu/WiFiManager #1421: ESP32-S3 support

acirulis, 2022-05-20:
Has anyone successfully used WifiManager library on ESP32-S3?

tablatronix, 2022-06-17:
I do not yet have an S3 dev kit

touchgadget, 2022-06-26:
WM fails to compile on ESP32S3 because the temperature sensor API functions are not implemented.

```
Linking everything together...
tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32s3-elf-g++ -Wl,--Map=/tmp/arduino_build_345963/usbjoywa.ino.map -L/tmp/autobuild_usbjoywa_21717/arduino-1.8.19/portable/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s3/lib -L/tmp/autobuild_usbjoywa_21717/arduino-1.8.19/portable/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s3/ld -L/tmp/autobuild_usbjoywa_21717/arduino-1.8.19/portable/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32s3/qspi_qspi -T memory.ld -T sections.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib-time.ld -T esp32s3.peripherals.ld -mlongcalls -Wl,--cref -Wl,--gc-sections -fno-rtti -fno-lto -u _Z5setupv -u _Z4loopv -u esp_app_desc -u pthread_include_pthread_impl -u pthread_include_pthread_cond_impl -u pthread_include_pthread_local_storage_impl -u pthread_include_pthread_rwlock_impl -u include_esp_phy_override -u ld_include_highint_hdl -u start_app -u start_app_other_cores -u __ubsan_include -Wl,--wrap=longjmp -u __assert_func -u vfs_include_syscalls_impl -Wl,--undefined=uxTopUsedPriority -u app_main -u newlib_include_heap_impl -u newlib_include_syscalls_impl -u newlib_include_pthread_impl -u newlib_include_assert_impl -u __cxa_guard_dummy -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 -Wl,--start-group /tmp/arduino_build_345963/sketch/usbjoywa.ino.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFi.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiAP.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiClient.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiGeneric.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiMulti.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiSTA.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiScan.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiServer.cpp.o /tmp/arduino_build_345963/libraries/WiFi/WiFiUdp.cpp.o /tmp/arduino_build_345963/libraries/WebServer/Parsing.cpp.o /tmp/arduino_build_345963/libraries/WebServer/WebServer.cpp.o /tmp/arduino_build_345963/libraries/WebServer/detail/mimetable.cpp.o /tmp/arduino_build_345963/libraries/ESPmDNS/ESPmDNS.cpp.o /tmp/arduino_build_345963/libraries/WiFiManager/WiFiManager.cpp.o /tmp/arduino_build_345963/libraries/Update/HttpsOTAUpdate.cpp.o /tmp/arduino_build_345963/libraries/Update/Updater.cpp.o /tmp/arduino_build_345963/libraries/DNSServer/DNSServer.cpp.o /tmp/arduino_build_345963/libraries/FS/FS.cpp.o /tmp/arduino_build_345963/libraries/FS/vfs_api.cpp.o /tmp/arduino_cache_459655/core/core_53bd8a414b9632a526d3ae113b33cd8a.a -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lasio -lbt -lcbor -lunity -lcmock -lcoap -lnghttp -lesp-tls -lesp_adc_cal -lesp_hid -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lesp_https_server -lesp_lcd -lprotobuf-c -lprotocomm -lmdns -lesp_local_ctrl -lsdmmc -lesp_serial_slave_link -lesp_websocket_client -lexpat -lwear_levelling -lfatfs -lfreemodbus -ljsmn -ljson -llibsodium -lmqtt -lopenssl -lperfmon -lspiffs -lusb -lulp -lwifi_provisioning -lbutton -lrmaker_common -ljson_parser -ljson_generator -lesp_schedule -lesp_rainmaker -lqrcode -lws2812_led -lesp-dsp -lesp-sr -lesp32-camera -lesp_littlefs -lfb_gfx -lasio -lcbor -lcmock -lunity -lcoap -lesp_lcd -lesp_websocket_client -lexpat -lfreemodbus -ljsmn -llibsodium -lperfmon -lusb -lesp_adc_cal -lesp_hid -lfatfs -lwear_levelling -lopenssl -lesp_rainmaker -lesp_local_ctrl -lesp_https_server -lwifi_provisioning -lprotocomm -lbt -lbtdm_app -lprotobuf-c -lmdns -lrmaker_common -lmqtt -ljson_parser -ljson_generator -lesp_schedule -lqrcode -larduino_tinyusb -lcat_face_detect -lhuman_face_detect -lcolor_detect -lmfn -ldl -lwakenet -lhufzip -lesp_audio_front_end -lesp_audio_processor -lmultinet -lesp-sr -lwakenet -lhufzip -lesp_audio_front_end -lesp_audio_processor -lmultinet -ljson -lspiffs -ldl_lib -lc_speech_features -lesp_tts_chinese -lvoice_set_xiaole -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip -llog -lheap -lsoc -lesp_hw_support -lxtensa -lesp_common -lesp_timer -lfreertos -lnewlib -lcxx -lapp_trace -lnghttp -lesp-tls -ltcp_transport -lesp_http_client -lesp_http_server -lesp_https_ota -lsdmmc -lesp_serial_slave_link -lulp -lmbedtls -lmbedcrypto -lmbedx509 -lcoexist -lcore -lespnow -lmesh -lnet80211 -lpp -lsmartconfig -lwapi -lesp_ringbuf -lefuse -lesp_ipc -ldriver -lesp_pm -lmbedtls -lapp_update -lbootloader_support -lspi_flash -lnvs_flash -lpthread -lesp_gdbstub -lespcoredump -lesp_phy -lesp_system -lesp_rom -lhal -lvfs -lesp_eth -ltcpip_adapter -lesp_netif -lesp_event -lwpa_supplicant -lesp_wifi -lconsole -llwip - [... cut ...]
diff --git a/WiFiManager.cpp b/WiFiManager.cpp
index 6ca3018..e6f0317 100644
--- a/WiFiManager.cpp
+++ b/WiFiManager.cpp
@@ -2172,7 +2172,7 @@ String WiFiManager::getInfoData(String id){
     p.replace(FPSTR(T_1),WiFi.getAutoConnect() ? FPSTR(S_enable) : FPSTR(S_disable));
   }
   #endif
-  #ifdef ESP32
+  #if defined(ESP32) && !defined(ARDUINO_ESP32S3_DEV)
   else if(id==F("temp")){
     // temperature is not calibrated, varying large offsets are present, use for relative temp changes only
     p = FPSTR(HTTP_INFO_temp);
```

tablatronix, 2022-06-26:
untested

closed 2022-06-26 by tablatronix

touchgadget, 2022-06-27:
https://github.com/tzapu/WiFiManager/blob/06c63720f60ec6e169c9c4cb10f52896f6b97d74/WiFiManager.cpp#L2175

The above produces a warning. `warning: extra tokens at end of #ifdef directive`

I am pretty sure #ifdef only allows the name of a symbol, not an expression.

The following works for me.

```
#if defined(ESP32) && !defined(WM_NOTEMP)
```

The same `extra tokens` warning appears when building AutoConnectWithFSParameters.ino. There are 3 occurences. And 3 more in AutoConnectWithFSParametersAndCustomIP.ino.

https://github.com/tzapu/WiFiManager/blob/06c63720f60ec6e169c9c4cb10f52896f6b97d74/examples/Parameters/SPIFFS/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino#L49

Maybe the #ifdef should look like this.

```
    #if defined(ARDUINOJSON_VERSION_MAJOR) && ARDUINOJSON_VERSION_MAJOR >= 6
```

tablatronix, 2022-06-27:
I was wondering about those, kept forgetting to check that as my compiler is not always -Wall

touchgadget, 2022-06-27:
Thanks, ESP32S3 builds are working fine.

shashi5841, 2022-09-10:
> Thanks, ESP32S3 builds are working fine.

I am facing the same issue can you please guide me on how to solve it .

shashi5841, 2022-09-10:
Tested with esp32-s3 with the same issue.
ESP32-S3 with  WiFiManager lib.

Can't find SSID AutoConnectAP, when the module is in AP mode.

Terminal output:

SPIFFS mounted
*wm:[1] AutoConnect
*wm:[2] ESP32 event handler enabled
*wm:[2] Connecting as wifi client...
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to SAVED AP: espressif
*wm:[1] connectTimeout not set, ESP waitForConnectResult...
*wm:[2] [EVENT] WIFI_REASON: 201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED
*wm:[2] Starting Config Portal
*wm:[2] Disabling STA
*wm:[2] Enabling AP
*wm:[1] StartAP with SSID: AutoConnectAP
*wm:[2] AP has anonymous access!
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal
*wm:[2] HTTP server started
*wm:[2] Config Portal Running, blocking, waiting for clients...
*wm:[2] NUM CLIENTS: 0
*wm:[2] NUM CLIENTS: 0
*wm:[2] NUM CLIENTS: 0
*wm:[2] NUM CLIENTS: 0

I used 4 different mobile phones and different laptop PC to find esp32s3 in the WiFi list but the module was not broadcasting.

Used the latest library of WiFi manager v2.0.12-beta

Please advise what to do.

tablatronix, 2022-09-10:
@shashi5841 please see #1482

roleroz, 2023-06-01:
Do we still need to disable the temp sensor for ESP32S3?, it seems to be working now

At least, I forked this, removed the WM_NOTEMP macro, and linked it into my ESP32-S3 devkit board, it compiled, linked and flashed, and it seems to be working fine

roleroz, 2023-06-01:
For the record, this seems to have been added with IDF's 4.4 version (https://github.com/espressif/arduino-esp32/pull/6994)

tablatronix, 2023-06-01:
I can add a ver check maybe

roleroz, 2023-06-18:
Should we reopen this issue?, or start a new one?

tablatronix, 2023-06-18:
New one

roleroz, 2023-06-19:
Filed https://github.com/tzapu/WiFiManager/issues/1629. Please let me know if I'm missing anything, as this is the first time filing an issue against WiFiManager :wink: