tzapu/WiFiManager #1422: WiFi doesn't work with LOLIN/WEMOS C3 Mini

jeffryr, 2022-05-21:
Arduino IDE 1.8.19

WiFimanager Branch/Release:  2.0.11-beta

Hardware:			ESP32 C3

ESP32 Core Version:      	2.0.4

### Description

WiFimanager (really WiFi in general) doesn't work with LOLIN/WEMOS C3 Mini due to buggy power circuitry on the board.

### Settings in IDE

Module:  			LOLIN/WEMOS C3 Mini

### Debug Messages

Fri May 20 21:59:28 GMT 2022   Debug 2 This is setup_wifi()
*wm:[1] AutoConnect 
*wm:[2] ESP32 event handler enabled 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: rtn66u24
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] [EVENT] WIFI_REASON:  2
*wm:[2] [EVENT] WIFI_REASON:  2
...
*wm:[2] [EVENT] WIFI_REASON:  2
*wm:[2] [EVENT] WIFI_REASON:  2
*wm:[2] [EVENT] WIFI_REASON:  201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND 
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] [EVENT] WIFI_REASON:  202
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  c3minia
*wm:[1] AP IP address: 192.168.4.1
*wm:[2] [CB] _apcallback calling 
Fri May 20 22:00:17 GMT 2022   Debug 2 Entered config mode, AP name: 192.168.4.1
Fri May 20 22:00:17 GMT 2022   Debug 2 AP IP address: c3minia
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] Portal Timeout In 180 seconds
*wm:[2] Portal Timeout In 178 seconds
*wm:[2] Portal Timeout In 148 seconds
*wm:[2] Portal Timeout In 118 seconds
*wm:[2] Portal Timeout In 88 seconds
*wm:[2] Portal Timeout In 58 seconds
*wm:[2] Portal Timeout In 28 seconds
*wm:[1] config portal has timed out 
*wm:[2] shutdownConfigPortal 
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_DISCONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed 
*wm:[1] config portal exiting 
failed to connect and hit timeout
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381918
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: a9753a4fc647c6545c1b919ef08db429130a48592727edca270f1e5a3da0d0a9
Expected: 3bf6ef2cf3b9eefcd4b3c70cc5d1ce5138292d101a5cb1d5db6fbebf081b0a19
Attempting to boot anyway...
entry 0x403ce000

### Cause
According to https://www.wemos.cc/en/latest/c3/c3_mini.html

**You need set WIFI Tx Power to 8.5dBm to use WIFI.**

### Workaround

The following patch works arround the issue without affecting other boards:

```
*** WiFiManager.cpp.ORIGINAL	2022-04-14 14:02:05.000000000 -0400
--- WiFiManager.cpp	2022-05-20 18:19:25.599206924 -0400
***************
*** 483,488 ****
--- 483,492 ----
        ret = WiFi.softAP(_apName.c_str(),"",1,_apHidden);
      }  
    }
+   #ifdef ARDUINO_LOLIN_C3_MINI
+     DEBUG_WM(F("Reducing WiFi Tx Power for Lolin C3 mini"));
+     WiFi.setTxPower(WIFI_POWER_8_5dBm);
+   #endif
  
    if(_debugLevel >= DEBUG_DEV) debugSoftAPConfig();
  
***************
*** 1053,1058 ****
--- 1057,1066 ----
    WiFi_enableSTA(true,storeSTAmode); // storeSTAmode will also toggle STA on in default opmode (persistent) if true (default)
    WiFi.persistent(true);
    ret = WiFi.begin(ssid.c_str(), pass.c_str(), 0, NULL, connect);
+   #ifdef ARDUINO_LOLIN_C3_MINI
+     DEBUG_WM(F("Reducing WiFi Tx Power for Lolin C3 mini"));
+     WiFi.setTxPower(WIFI_POWER_8_5dBm);
+   #endif
    WiFi.persistent(false);
    #ifdef WM_DEBUG_LEVEL
    if(!ret) DEBUG_WM(DEBUG_ERROR,F("[ERROR] wifi begin failed"));
***************
*** 1082,1087 ****
--- 1090,1099 ----
    #endif
  
    ret = WiFi.begin();
+   #ifdef ARDUINO_LOLIN_C3_MINI
+     DEBUG_WM(F("Reducing WiFi Tx Power for Lolin C3 mini"));
+     WiFi.setTxPower(WIFI_POWER_8_5dBm);
+   #endif
  
    #ifdef WM_DEBUG_LEVEL
    if(!ret) DEBUG_WM(DEBUG_ERROR,F("[ERROR] wifi begin failed"));
```

tablatronix, 2022-05-21:
Can you provide a link to where it mentions this? I didn't see anything.

I will see if there is a way to set this via build flags also that would be ideal for boards files.

tablatronix, 2022-05-21:
Oh nm, I see thats all it says, not why. Do you know the actual cause, hardware bug that a extra capacitor doesnt fix?

tablatronix, 2022-05-21:
Since this seems to have to be set after wifi setups, why not just keep this in usercode?

if(autoconect() == true or isConnected etc..){ setpower }

wm.setAPCallback(configModeCallback);
configModeCallback(){ setpower }

jeffryr, 2022-05-22:
I tried putting the setpower in configModeCallback(), no go.  See https://github.com/arendst/Tasmota/discussions/15443 for some more discussion.  I purchased a few of the Wemos C3 Mini's along with the https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html to get some experience with the RiscV flavor.  The Espressif board works fine, the Wemos board not so much, not one of their better efforts...

tablatronix, 2022-05-22:
Hmm interesting. Ill check the esp docs.

ericvrp, 2022-06-05:
I probably have something similar while trying to use the WiFi of the Wemos C3 Mini.
I do not set the 8.5dBm tx power but always get this whenever I try to scan or connect with wifi:

```
I (2492) wifi_init: tcp tx win: 5744
I (2492) wifi_init: tcp rx win: 5744
I (2502) wifi_init: tcp mss: 1440
I (2502) wifi_init: WiFi IRAM OP enabled
I (2512) wifi_init: WiFi RX IRAM OP enabled
wifi_test() scan...
I (2522) phy_init: phy_version 909,aa05aec,Apr 16 2022,13:42:08
Error: 
  × Broken pipe
```

jeffryr, 2022-06-07:
> I probably have something similar while trying to use the WiFi of the Wemos C3 Mini. I do not set the 8.5dBm tx power (because I don't know how from Rust) but always get this whenever I try to scan or connect with wifi:
> 
> ```
> I (2492) wifi_init: tcp tx win: 5744
> I (2492) wifi_init: tcp rx win: 5744
> I (2502) wifi_init: tcp mss: 1440
> I (2502) wifi_init: WiFi IRAM OP enabled
> I (2512) wifi_init: WiFi RX IRAM OP enabled
> wifi_test() scan...
> I (2522) phy_init: phy_version 909,aa05aec,Apr 16 2022,13:42:08
> Error: 
>   × Broken pipe
> ```

Not a Rust user (introduced to C/unix in 1977 and it is still my native language) but if you are using https://github.com/esp-rs/esp-wifi for WiFi it supports the esp_wifi_set_max_tx_power() function from the SDK.  Search for set_max_tx_power in https://github.com/esp-rs/esp-wifi/blob/main/src/binary/include_esp32c3.rs.  Calling esp_wifi_set_max_tx_power(34) in the right part of your code should correspond to 8.5dBm.

tablatronix, 2022-06-07:
The docs say you have to `esp_wifi_set_max_tx_power` so no reason for this to be in wm code at all or even need a callback to do it.

jeffryr, 2022-06-12:
> The docs say you have to `esp_wifi_set_max_tx_power` after wifi start. so no reason for this to be in wm code at all or even need a callback to do it.

Verified the following works both for initial configuration (no previous AP setup) as well as reboot when AP has been previously saved using stock WiFi Manager version v2.0.11-beta .  My setup_wifi() function now uses the following startup sequence:
```
void setup_wifi() {
  D(debug(2, F("This is %s()"), __FUNCTION__));
#if defined(ESP32)
  int TxPower;
  WiFi.setHostname(defaults.hostname); // Has to happen before WiFi.begin()
#endif
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
#if defined(ARDUINO_LOLIN_C3_MINI)
  D(debug(2, F("Reducing WiFi Tx Power for Lolin C3 mini")));
  WiFi.setTxPower(WIFI_POWER_8_5dBm);
#endif
  // Reset saved settings for testing
  // wifiManager.resetSettings();
  // Set callback so we know that we entered the Access Portal
  wifiManager.setAPCallback(configModeCallback);
  // Set timeout until configuration portal gets turned off
  wifiManager.setTimeout(180);
  // Configure WiFi if necessary
  if (!wifiManager.autoConnect(defaults.hostname, http_password)) {
    Serial.println(F("failed to connect and hit timeout"));
    delay(3000);
    // reset and try again, or maybe put it to deep sleep
#if defined(ESP8266)
    ESP.reset();
#else
    ESP.restart();
#endif
    delay(5000);
  }
  WiFi.printDiag(Serial);
#if defined(ESP8266)
  wifi_station_set_hostname(defaults.hostname);
  D(debug(3, F("hostname: %s"), wifi_station_get_hostname()));
#elif defined(ESP32)
  D(debug(3, F("hostname: %s"), WiFi.getHostname()));
  WiFi.setSleep(false); // See https://github.com/espressif/arduino-esp32/issues/4855
#else
# error "Tested on ESP8266 and ESP32 only"
#endif
  ip = WiFi.localIP().toString();
  D(debug(3, F("IP address: %s"), ip.c_str()));
  mac = WiFi.macAddress();
  D(debug(3, F("MAC Addr: %s"), mac.c_str()));
  ssid = WiFi.SSID();
  D(debug(3, F("SSID: %s"), ssid.c_str()));
  gatewayip = WiFi.gatewayIP().toString();
  D(debug(3, F("Gateway IP: %s"), gatewayip.c_str()));
  rssi = WiFi.RSSI();  // Point in time
  D(debug(3, F("Current RSSI: %d"), rssi));
  display_wifiip(WiFi.localIP().toString());
#if defined(ESP32)
  TxPower = WiFi.getTxPower();
  D(debug(3, F("TxPower (raw): %d"), TxPower));
#endif // ESP32
  while (!MDNS.begin(defaults.hostname)) {
    D(debug(1, F("Error setting up mDNS responder!")));
    for (int i = 0; i < 20; i++) {
      Serial.print(".");
      delay(100);
    }
    Serial.println(".");
  }
  D(debug(2, F("mDNS responder started")));
}
```
Recommend this issue can be closed with a documentation note that WiFi TX power needs to be reduced for the C3 Mini prior to calling WiFiManager.

tablatronix, 2022-06-12:
hmm ok so for ap mode this is also needed to be set?

popy2k14, 2025-05-09:
!!!!!! WARNING FROM LOLIN C3 MINI !!!!!!
!!!!!! WARNING FROM LOLIN C3 MINI !!!!!!
!!!!!! WARNING FROM LOLIN C3 MINI !!!!!!
!!!!!! WARNING FROM LOLIN C3 MINI !!!!!!

Those boards have serious wifi issues!!!!!!
Replaced my 8266 wemos d1 mini with them and none of those connect to wifi!
Tried 7 of them, all have the same issue.
There are a lot of issues, see those links. There are other users which warns from thos boards:

https://github.com/tzapu/WiFiManager/issues/1422
https://github.com/esphome/issues/issues/4893
https://github.com/arendst/Tasmota/discussions/15443

I have bought v2.1.0 and also v2.1.0 those have the issues!
Will try to get a refund!

DB9JG, 2025-05-09:
These boards often suffer from a very bad designed and mismatched WiFi antenna circuit which even triggers  the internal protection circuit. 
An antenna mod helps:
https://peterneufeld.wordpress.com/2025/03/04/esp32-c3-supermini-antenna-modification/

popy2k14, 2025-05-09:
thx for the info and hint about the mod.
Sadly i dont have space for the antenna.
Wanted to upgrade my ESP8266+SEN0395 which are mounted inside an 3D Printed case.
Those combo running for stable for years.