tzapu/WiFiManager #1424: PlatforrmIO vs Arduino IDE version mismatch

minhaj6, 2022-06-02:
Hi, 
[PlatformIO registry](https://registry.platformio.org/libraries/tzapu/WiFiManager) version is very old compared to current development and the version available in arduino IDE. 

I am using ESP32 and trying out PlatformIO. Seems like author `tzapu` submitted it to platformIO. Can you please submit an updated version there? If I could do that by myself, I would. 

Thanks!!

albznw, 2022-06-03:
A solution to this problem is to add the git url to the `lib_deps` option in the `platformio.ini` file. This way you will get the latest version.

Like so:
```
lib_deps =
     . . . (your other dependencies here)
    https://github.com/tzapu/WiFiManager.git
```

tablatronix, 2022-06-03:
Platformio does not support prerelease versions afaik, so it does not pick them up.
Ill try to find a solution or release

linxcow, 2022-08-31:
I have the same problem , latest version on platformIO is 0.16 ( 2 years old) no esp32 support

tablatronix, 2022-08-31:
https://community.platformio.org/t/library-manager-pre-release-support/23674

linxcow, 2022-08-31:
thx , workaround by albznw worked fine for me.
`lib_deps =
     . . . (your other dependencies here)
    https://github.com/tzapu/WiFiManager.git`

tablatronix, 2022-08-31:
I think I will release 2.1 after I have this s2 c3 issue nailed down

masterrex, 2022-10-08:
Best way is to pin the dependency against a tag for now: `lib_deps = https://github.com/tzapu/WiFiManager.git#v2.0.13-beta`

Jadeko, 2023-07-14:
Hello, as of today the latest release (stable), version 2.0.16-rc.2 is still not available through platformio libraries.
Would be great if this issue could be sorted.
Thx

tablatronix, 2023-07-14:
Will be sorted today

Jadeko, 2023-07-20:
Hello,
Still not updated unfortunately :(
https://registry.platformio.org/libraries/tzapu/WiFiManager