tzapu/WiFiManager #1427: Custom parameters box  getting added each time after running code

rudro101, 2022-06-07:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
#define ESP_DRD_USE_SPIFFS true

#include <WiFi.h>
#include <FS.h>
#include <SPIFFS.h>
#include <WiFiManager.h>
#include <ArduinoJson.h>
#define trigPin 5

#define JSON_CONFIG_FILE "/test_config.json"
 
bool shouldSaveConfig = false;
 
int testNumber1;
int testNumber2;
int testNumber3;
 
WiFiManager wm;
 
void saveConfigFile()
{
  Serial.println(F("Saving configuration..."));
  
  StaticJsonDocument<512> json;
  json["testNumber1"] = testNumber1;
  json["testNumber2"] = testNumber2;
  json["testNumber3"] = testNumber3;
 
  File configFile = SPIFFS.open(JSON_CONFIG_FILE, "w");
  if (!configFile)
    {
      Serial.println("failed to open config file for writing");
    }
 
  serializeJsonPretty(json, Serial);
  
  if (serializeJson(json, configFile) == 0)
    {
      Serial.println(F("Failed to write to file"));
    }
  configFile.close();
}
 
bool loadConfigFile()
{
 
  Serial.println("Mounting File System...");
 
  if (SPIFFS.begin(false) || SPIFFS.begin(true))
  {
    Serial.println("mounted file system");
    if (SPIFFS.exists(JSON_CONFIG_FILE))
    {
      Serial.println("reading config file");
      File configFile = SPIFFS.open(JSON_CONFIG_FILE, "r");
      if (configFile)
      {
        Serial.println("Opened configuration file");
        StaticJsonDocument<512> json;
        DeserializationError error = deserializeJson(json, configFile);
        serializeJsonPretty(json, Serial);
        if (!error)
        {
          Serial.println("Parsing JSON");
 
          testNumber1 = json["testNumber1"].as<int>();
          testNumber2 = json["testNumber2"].as<int>();
          testNumber3 = json["testNumber3"].as<int>();
          return true;
        }
        else
        {
         
          Serial.println("Failed to load json config");
        }
      }
    }
  }
  else
  {
    Serial.println("Failed to mount FS");
  }
 
  return false;
}
 
 
void saveConfigCallback()
{
  Serial.println("Should save config");
  shouldSaveConfig = true;
}
 
void configModeCallback(WiFiManager *myWiFiManager)
{
  Serial.println("Entered Configuration Mode");
  Serial.print("Config SSID: ");
  Serial.println(myWiFiManager->getConfigPortalSSID());
  Serial.print("Config IP Address: ");
  Serial.println(WiFi.softAPIP());
}
 
void setup()

{
  WiFi.mode(WIFI_STA);
  Serial.begin(115200);
  Serial.println("Start");
  pinMode(trigPin, INPUT_PULLUP); 
}
 
 
 
 
void loop() {
 if (digitalRead(trigPin)==LOW){
 
 bool forceConfig = false;
 
  bool spiffsSetup = loadConfigFile();
  if (!spiffsSetup)
  {
    Serial.println(F("Forcing config mode as there is no saved config"));
    forceConfig = true;
  }
 
 
  wm.resetSettings();
 
  wm.setSaveConfigCallback(saveConfigCallback);
 
  wm.setAPCallback(configModeCallback);
  
  char convertedValue_a[6];
  char convertedValue_b[6];
  char convertedValue_c[6];

  
  sprintf(convertedValue_a, "%d", testNumber1);
  sprintf(convertedValue_b, "%d", testNumber2);
  sprintf(convertedValue_c, "%d", testNumber3); 
  
  WiFiManagerParameter custom_text_box_num1("key_num1", "LED1", convertedValue_a, 7);
  WiFiManagerParameter custom_text_box_num2("key_num2", "LED2", convertedValue_b, 7);
  WiFiManagerParameter custom_text_box_num3("key_num3", "LED3", convertedValue_c, 7); 
 
  wm.addParameter(&custom_text_box_num1);
  wm.addParameter(&custom_text_box_num2);
  wm.addParameter(&custom_text_box_num3);
 
  if (forceConfig)
  {
    if (!wm.startConfigPortal("NEWTEST_AP", "password"))
    {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      ESP.restart();
      delay(5000);
    }
  }
  else
  {
    if (!wm.autoConnect("NEWTEST_AP", "password"))
    {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      ESP.restart();
      delay(5000);
    }
  }
 
 
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
 
 
  testNumber1 = atoi(custom_text_box_num1.getValue());
  Serial.print("testNumber1: ");
  Serial.println(testNumber1);
  
  testNumber2 = atoi(custom_text_box_num2.getValue());
  Serial.print("testNumber2: ");
  Serial.println(testNumber2);
  
  testNumber3 = atoi(custom_text_box_num3.getValue());
  Serial.print("testNumber3: ");
  Serial.println(testNumber3);

  

  
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());
  Serial.print("psk: ");
  Serial.println(WiFi.psk());
 
 
  if (shouldSaveConfig)
  {
    saveConfigFile();
    
  }
}

}
![WhatsApp Image 2022-06-07 at 10 48 36 PM](https://user-images.githubusercontent.com/58129107/172444284-3162eddf-998e-40ce-9904-1ead40b735ef.jpeg)
![WhatsApp Image 2022-06-07 at 10 48 40 PM](https://user-images.githubusercontent.com/58129107/172444291-d6a491ed-0bf9-4de1-8c29-c0649d64c3b5.jpeg)

### Debug Messages

```
messages here
```
![WhatsApp Image 2022-06-07 at 10 48 40 PM](https://user-images.githubusercontent.com/58129107/172444589-57b054f6-b7e6-4c47-8aca-2842101be82a.jpeg)
![WhatsApp Image 2022-06-07 at 10 48 36 PM](https://user-images.githubusercontent.com/58129107/172444619-3f78c8b9-8864-42b4-ad24-a979ffee6eb8.jpeg)

rudro101, 2022-06-07:
1st time:
[image]
2nd time:
[image]

closed 2022-06-13 by rudro101

tablatronix, 2022-06-13:
Lol did you figure it out?

rudro101, 2022-06-14:
Yes