tzapu/WiFiManager #1428: Problem handling reset with webserver

Sentinel8000, 2022-06-09:
Currently the code itself of the wifimanager working so, if go into AP mode, and user set a wifi, try after connect withot restart, this working well on expressif dev modul, but my problem is same: webserver settings running once and fail - without restart, i need restart the esp32 after wifimanager go from AP-> to Wifi mode.

This will be no problem, if after go into AP and set wifi name, pass will be automaticaly a restart, but currently i cant catch the event or restart the esp. Esp will restart only if AP starting fail.

Im happy for any help or guide - how to detect exit AP/wifi settings done to init a restart to run webserver with the new received IP.

tablatronix, 2022-06-10:
Without any logs? no clue

Sentinel8000, 2022-06-10:
Im not sure, log will help, because i think this is a normal working. Because wifimanager using a some kind a webserver for AP and i try use also too a webserver not a good idea.

Curently working scenario looks so:
1., reset password, ssid - for simulate starting condition
2., ESP restart
3., wifimanager start working starting AP mode
4.,Entering AP webserver site on phone to configure ssid and pass of the used wifi network
5.,Wifimanager save the password and making wifi reconnect without full esp board restart - this is what generate problem on my side
6.,ip received, network working well  but esp webserver not working, esp need restart again 
7,esp webserver working well after restart and connect direct to the saved wifi natwork using saved ssid/pass.

For this i have modded this section of the library code for restart after saving wifi passw::

`// attempt sta connection to submitted _ssid, _pass
        uint8_t res = connectWifi(_ssid, _pass, _connectonsave) == WL_CONNECTED;
        if (res || (!_connectonsave)) {
          #ifdef WM_DEBUG_LEVEL
          if(!_connectonsave){
            DEBUG_WM(F("SAVED with no connect to new AP"));
            ESP.restart();  <---
          } else {
	
            DEBUG_WM(F("Connect to new AP [SUCCESS]"));
            DEBUG_WM(F("Got IP Address:"));
            DEBUG_WM(WiFi.localIP());
            ESP.restart();  <---	
          }
          #endif `

Also i have noticed i think a real issue - if im choose a wifi / ssid and enter bad password, however i have set

 wm.setConnectTimeout(5);
 wm.setConnectRetries(3);

wifimanager try three times connect to a new AP, but lookslike not using setConnectTimeout parameter and we need wait more time:

20:05:20.552 -> *wm:[1] Connect Wifi, ATTEMPT # 1 of 3
20:05:20.552 -> *wm:[1] Connecting to NEW AP: DIGI-J7ch
20:05:20.694 -> *wm:[1] connectTimeout not set, ESP waitForConnectResult... 

the setConnectTimeout looks working well if im entering the good pass for ssid after exit AP:

20:10:51.589 -> *wm:[1] Connect Wifi, ATTEMPT # 1 of 3
20:10:51.589 -> *wm:[1] Connecting to SAVED AP: DIGI-J7ch
20:10:52.108 -> *wm:[2] 5000 ms timeout, waiting for connect...

Maybe this is because i have touched the library code, im not sure.

tablatronix, 2022-06-11:
Do you have the connect failure reasons logged?

Sentinel8000, 2022-06-11:
> Do you have the connect failure reasons logged?

I have set esp32 logging to Debug level, restored original wifimanager and compiled again the code, here is the log:

16:21:19.605 -> *wm:[2] processing save 
16:21:21.587 -> *wm:[2] Connecting as wifi client... 
16:21:21.587 -> *wm:[2] setSTAConfig static ip not set, skipping 
16:21:22.582 -> *wm:[1] Connect Wifi, ATTEMPT # 1 of 3
16:21:22.582 -> *wm:[1] Connecting to NEW AP: DIGI-J7ch
16:21:22.724 -> *wm:[1] connectTimeout not set, ESP waitForConnectResult... 
16:21:29.654 -> [ 73258][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
16:21:31.959 -> [ 75519][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
16:21:31.959 -> [ 75519][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.100.16, MASK: 255.255.255.0, GW: 192.168.100.1
16:21:31.959 -> *wm:[2] Connection result: WL_CONNECTED
16:21:31.959 -> *wm:[1] Connect to new AP [SUCCESS] 
16:21:31.959 -> *wm:[1] Got IP Address: 
16:21:31.959 -> *wm:[1] 192.168.100.16 
16:21:31.959 -> *wm:[2] shutdownConfigPortal 
16:21:31.959 -> [ 75549][E][WebServer.cpp:635] _handleRequest(): request handler not found
16:21:31.959 -> *wm:[2] <- Request redirected to captive portal 
16:21:32.006 -> *wm:[2] restoring usermode STA
16:21:32.006 -> [ 75566][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 13 - AP_STADISCONNECTED
16:21:32.006 -> [ 75566][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 11 - AP_STOP
16:21:32.006 -> [ 75571][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 10 - AP_START
16:21:32.998 -> [ 76566][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 11 - AP_STOP
16:21:32.998 -> *wm:[2] wifi status: WL_CONNECTED
16:21:32.998 -> *wm:[2] wifi mode: STA
16:21:32.998 -> *wm:[2] configportal closed 
16:21:32.998 -> *wm:[1] config portal exiting 
16:21:32.998 -> connected...local ip:
16:21:32.998 -> 192.168.100.16
16:21:32.998 -> Signal level: -56dBm
16:21:33.423 -> [ 77000][D][VS1053.cpp:156] begin(): 
16:21:33.423 -> 
16:21:33.423 -> [ 77000][D][VS1053.cpp:157] begin(): Reset VS1053...
16:21:33.423 -> 
16:21:33.941 -> [ 77500][D][VS1053.cpp:161] begin(): End reset VS1053...
16:21:33.941 -> 
16:21:34.460 -> [ 78020][D][VS1053.cpp:132] testComm(): Slow SPI,Testing VS1053 read/write registers...
16:21:34.460 -> 
16:21:34.553 -> [ 78152][D][VS1053.cpp:132] testComm(): Fast SPI, Testing VS1053 read/write registers again...
16:21:34.553 -> 
16:21:36.198 -> [ 79775][D][VS1053.cpp:183] begin(): endFillByte is 0
16:21:36.198 -> 
16:21:36.481 -> [ 80062][D][VS1053.cpp:325] switchToMp3Mode(): Switched to mp3 mode
16:21:36.481 -> 
16:21:36.481 -> [ 80062][D][VS1053.cpp:268] softReset(): Performing soft-reset
16:21:36.481 -> 
16:21:36.481 -> [ 80075][E][AsyncTCP.cpp:1287] begin(): bind error: -8
16:21:36.481 -> [ 80076][W][esp32-hal-timer.c:225] timerAttachInterrupt(): EDGE timer interrupt is not supported! Setting to LEVEL...
16:21:36.529 -> [ 80080][D][ESP32_VS1053_Stream.cpp:97] connecttohost(): connecting to http://retro.dancewave.online/retrodance.mp3
16:21:36.529 -> [ 80090][D][HTTPClient.cpp:298] beginInternal(): protocol: http, host: retro.dancewave.online port: 80 url: /retrodance.mp3
16:21:36.529 -> [ 80101][D][HTTPClient.cpp:595] sendRequest(): request type: 'GET' redirCount: 0
16:21:36.529 -> 
16:21:36.671 -> [ 80268][D][HTTPClient.cpp:1141] connect():  connected to retro.dancewave.online:80
16:21:36.813 -> [ 80373][D][HTTPClient.cpp:1275] handleHeaderResponse(): code: 302
16:21:36.813 -> [ 80373][D][HTTPClient.cpp:1278] handleHeaderResponse(): size: 138
16:21:36.813 -> [ 80374][D][HTTPClient.cpp:619] sendRequest(): sendRequest code=302
16:21:36.813 -> 
16:21:36.813 -> [ 80380][D][HTTPClient.cpp:662] sendRequest(): following redirect (dropped to GET/HEAD): 'http://stream.dancewave.online:8080/retrodance.mp3' redirCount: 1
16:21:36.813 -> 
16:21:36.813 -> [ 80394][D][HTTPClient.cpp:383] disconnect(): still data in buffer (138), clean up.
16:21:36.813 -> 
16:21:36.813 -> [ 80404][D][HTTPClient.cpp:390] disconnect(): tcp keep open for reuse
16:21:36.861 -> [ 80408][D][HTTPClient.cpp:292] beginInternal(): switching host from 'retro.dancewave.online' to 'stream.dancewave.online'. disconnecting first
16:21:36.861 -> [ 80420][D][HTTPClient.cpp:392] disconnect(): tcp stop
16:21:36.861 -> [ 80426][D][HTTPClient.cpp:298] beginInternal(): protocol: http, host: stream.dancewave.online port: 8080 url: /retrodance.mp3
16:21:36.861 -> [ 80436][D][HTTPClient.cpp:595] sendRequest(): request type: 'GET' redirCount: 1
16:21:36.861 -> 
16:21:36.908 -> [ 80505][D][HTTPClient.cpp:1141] connect():  connected to stream.dancewave.online:8080
16:21:37.001 -> [ 80593][D][HTTPClient.cpp:1275] handleHeaderResponse(): code: 200
16:21:37.001 -> [ 80593][D][HTTPClient.cpp:619] sendRequest(): sendRequest code=200
16:21:37.001 -> 
16:21:37.001 -> [ 80594][D][ESP32_VS1053_Stream.cpp:131] connecttohost(): connection made in 493 ms
16:21:37.048 -> [ 80601][D][ESP32_VS1053_Stream.cpp:137] connecttohost(): connected to http://retro.dancewave.online/retrodance.mp3
16:21:37.048 -> [ 80612][D][ESP32_VS1053_Stream.cpp:188] connecttohost(): codec MP3
16:21:37.048 -> 
16:21:37.048 -> Showstation saved:
16:21:37.048 -> Dance Wave Retro
16:21:37.048 -> 
16:21:37.048 -> Showstation decoded:
16:21:37.048 -> Dance Wave Retro!
16:21:37.048 -> [ 80629][D][ESP32_VS1053_Stream.cpp:197] connecttohost(): metadata interval is 16384
[... cut ...]
16:21:37.096 -> The function decode(&results)) is deprecated and may not work as expected! Just use decode() without a parameter and IrReceiver.decodedIRData.<fieldname> .
16:21:37.143 -> [ 80716][D][ESP32_VS1053_Stream.cpp:271] _handleStream():  1696 bytes to decoder
16:21:37.143 -> [ 80738][D][ESP32_VS1053_Stream.cpp:365] loop(): startmute ms: 60
16:21:37.143 -> [ 80740][D][ESP32_VS1053_Stream.cpp:271] _handleStream():   416 bytes to decoder
16:21:37.190 -> [ 80779][D][ESP32_VS1053_Stream.cpp:271] _handleStream():   416 bytes to decoder

tablatronix, 2022-06-11:
hmm, yeah this all looks fine. I have never gotten the webserver to reset properly, it seems to leave something around.
Not sure if its an arduino bug or esp bug.

Sentinel8000, 2022-06-11:
As workaround - and also this works -  i have added a ESP.restart() after saving password - because so looks my project little bit weird - touching anytime the esp32 reset to get webserver.

If this issue /bug  cant be handle from code side, also will be a good idea if library code giving a option/parameter to set after saving or succesfully connecting to choose - user want faster method for connecting reconnect or full esp32 restart - so can be same situations with webserver handled. restartAfterSuccesFullyConnect(true) :) 

But this reset will do only, if gived ssid/passw succesfully connect - to make sure after reset will start.

dmlambo, 2023-05-20:
I think I found the problem, and a workaround. See my comment here https://github.com/tzapu/WiFiManager/issues/1142#issuecomment-1555950494