tzapu/WiFiManager #1429: Switching to DHCP mode

HeatseekerXXX, 2022-06-12:
Hi.
Here is some algorithm (approach) of setting an IP I'm using. 
1. Setting up default static IP/MASK/GW/DNS that are shown  when config portal starts by calling 
`wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn, _gw);`
2.  When config portal starts user surely has an opportunity to set his own required  IP/MASK/GW/DNS **or**  to choose DHCP method of  setting an IP. To manage this I've added  custom parameter: 0 - use static IP, 1 - use DHCP.
`WiFiManagerParameter force_DHCP("F_DHCP", "1-Use DHCP 0-StaticIP", "0", 1);`
3. After Save button was pressed, I'm checking force_DHCP parameter  in callback  
`wifiManager.setSaveParamsCallback(AfterSaveParamCallback);`
4. If force_DHCP is enabled I'm setting  IP/MASK/GW = 0.0.0.0 
```c++
_IPAddress n_ip, n_mask, n_gw, n_dns;
  n_ip.fromString("0.0.0.0");
  n_gw.fromString("0.0.0.0");
  n_mask.fromString("0.0.0.0");
  wifiManager.setSTAStaticIPConfig(n_ip, n_mask, n_gw);_
```
If IP equals "0.0.0.0" wifimanager would connect to AP using DHCP.
5.  After exiting from AfterSaveParamCallback callback wifimanager returns to handlewifisave() function
```c++
`void WiFiManager::handleWifiSave() 
{
  #ifdef WM_DEBUG_LEVEL
  DEBUG_WM(DEBUG_VERBOSE,F("<- HTTP WiFi save "));
  DEBUG_WM(DEBUG_DEV,F("Method:"),server->method() == HTTP_GET  ? (String)FPSTR(S_GET) : (String)FPSTR(S_POST));
  #endif
  handleRequest();'

 // @todo use new callback for before paramsaves
  if (!_paramsInWifi && _presavecallback != NULL) {
    _presavecallback(); // CALLBACK is CALLED
  }

  //SAVE/connect here
  _ssid = server->arg(F("s")).c_str();
  _pass = server->arg(F("p")).c_str();

if(_paramsInWifi) doParamSave();

  if (server->arg(FPSTR(S_ip)) != "") {
    //_sta_static_ip.fromString(server->arg(FPSTR(S_ip));
    String ip = server->arg(FPSTR(S_ip));
    optionalIPFromString(&_sta_static_ip, ip.c_str());
    #ifdef WM_DEBUG_LEVEL
    DEBUG_WM(DEBUG_DEV,F("static ip:"),ip);
    #endif
```
6. Unfortunately after setting _sta_static_ip ="0.0.0.0" wifimanager will read IP settings from server rewriting IP stored in _sta_static_ip.

Is there is another way  to connect using DHCP? Can you add another callback that will be called after String ip = server->arg(FPSTR(S_ip));?

tablatronix, 2022-06-13:
Not sure I understand, dhcp is the default.

have you tried just setting ips to '0'

HeatseekerXXX, 2022-06-13:
Yes, user can enter static IPs. `wifiManager.setShowStaticFields(true);`
But also I want to give user a choice to set IPs directly or let DHCP server to assign IPs.
I consider  two ways to set  IP using DHCP.
First.  Setting `wifiManager.setShowStaticFields(false);`, but user isn't allowed to set IP manually.
Second. User is allowed to set IPs manually, but if he wants to use DHCP he have to delete all IPs or  set to 0.  At least it is not convenient. Using simple radio buttons or checkbox is the easiest way.  That’s why I want to check  settings in Params Callback function. But _presavecallback() is called just before String ip = server->arg(FPSTR(S_ip)), so I'm unable to make successful IPs manipulations in callback.
I can make changes in wifimanager.cpp by myself and create new callback function.  But I want to be sure that there is no other way to let user to switch between dhcp an static IP config.

tablatronix, 2022-06-13:
Ah ok could do it in javascript. Ill take a look at moving that callback or a way to invalidate ips

tablatronix, 2022-06-17:
I moved the presaveconfigcallback to after the ips are set, so maybe now they can be cleared before the wifi connect

tablatronix, 2022-06-17:
I also added a presaveparams callback to deal with the new params seperation easier