tzapu/WiFiManager #1434: problem using WiFiManager together with WebSerial

MAWMN, 2022-06-18:
### Basic Infos
Hello,

I have a problem in using WiFiManager at the same time as WebSerial in a Program, because the two libraries below are NOT compatible with each other..
  used for WiFiManager : #include <ESP8266WebServer.h>
  used for WebSerial   : #include <ESPAsyncWebServer.h>
is there any way to be able to use both, or the functionality of both ?

#### Hardware
Esp8266

Hardware:			Wemos-D1-mini

Core Version:      	Core 6.0.2·Home 3.4.2

### Description

Problem description
see above 

### Settings in IDE
PlatformIO
Module:  			Wemos D1

Additional libraries: 

### Sketch
n.a.

### Debug Messages
incompattible libraries

MAWMN, 2022-06-22:
all problems solved..

tablatronix, 2022-06-22:
I have an async branch I have started working on /testing Hope that helps in the future