tzapu/WiFiManager #1435: Firmware update progress

Sladerix, 2022-06-19:
Hi everyone, I'm using this library for the "Update Firmware" function. Very useful.
However I was wondering if there is a way to get the update progress with some callback or something..
I see in the serial monitor that while updating, the library, prints a lot of dots "." so I imagine that could be possible to have a callback for the progress of it.

is it something already implemented or should I edit the library myself?

Thanks in advance

tablatronix, 2022-06-19:
You can add your own arduinoota callbacks in your code also I think there is one wm callback at least

Sladerix, 2022-06-19:
> You can add your own arduinoota callbacks in your code also I think there is one wm callback at least

Ouh ok. Where can I find the full documentation about this?

tablatronix, 2022-06-19:
The Arduinoota example sketch or esp examples

Sladerix, 2022-06-20:
> 

Ok thanks, I found some examples in **_File > Examples > ArduinoOTA_** and **_File > Examples > Update_**

But I wish to keep using the WiFiManager Library at the moment. I will take a look in the source files

Sladerix, 2022-06-20:
In the source file I didn't found pretty much, apart for the **setPreOtaCallback** accessibile method.
And the **handleUpdating** function that is private... but the only things it does is to write the bytes with the _Update_ object I think (and other checks).

also there is this comment:
`...
  // add progress handler for debugging
  ...`

tablatronix, 2022-06-20:
you should be able to use them both

````  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
```

Sladerix, 2022-06-20:
> you should be able to use them both

Uhm I didn't understand how to use both given the fact that the "autoConnect" is blocking.
However I tried like so:
```c++
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

void setup() {
  Serial.begin(115200);
  
  WiFiManager wifiManager;
 
  wifiManager.setConfigPortalTimeout(180);
  //wifiManager.setConfigPortalBlocking(false);

  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  //ArduinoOTA.begin();

  wifiManager.resetSettings();
  if(!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("failed to connect and hit timeout");
  } 

  Serial.println("connected...yeey :)");
}

void loop() {
  // put your main code here, to run repeatedly:
  ArduinoOTA.handle();
}
```

if I try to uncomment `ArduinoOTA.begin(); `it goes in crash loop

tablatronix, 2022-06-21:
Sorry I wasnt thinking, let me see what I can add

tablatronix, 2022-06-21:
```C++

void handlePreOtaUpdateCallback(){
  Update.onProgress([](unsigned int progress, unsigned int total) {
        Serial.printf("CUSTOM Progress: %u%%\r", (progress / (total / 100)));
  });
}

  wm.setPreOtaUpdateCallback(handlePreOtaUpdateCallback);
```

Sladerix, 2022-06-21:
OK.
Now it works as expected!

Thanks very much!!!

Full code example for future people
```c++
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void handlePreOtaUpdateCallback(){
  Update.onProgress([](unsigned int progress, unsigned int total) {
        Serial.printf("CUSTOM Progress: %u%%\r", (progress / (total / 100)));
  });
}

void setup() {
  Serial.begin(115200);
  
  WiFiManager wifiManager;
 
  wifiManager.setConfigPortalTimeout(180);
  //wifiManager.setConfigPortalBlocking(false);

  wifiManager.setPreOtaUpdateCallback(handlePreOtaUpdateCallback);
  wifiManager.resetSettings();
  if(!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("failed to connect and hit timeout");
  } 

  Serial.println("connected...yeey :)");
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

closed 2022-06-21 by Sladerix

Sladerix, 2022-06-21:
Do you think it's possibile to send BLE data in that callback?
Because I did something like:
```c++
void handlePreOtaUpdateCallback(){
  Update.onProgress([](unsigned int progress, unsigned int total) {
        Serial.printf("CUSTOM Progress: %u%%\r\n", (progress / (total / 100)));

        char valueByte[16];
        ltoa(int(progress / (total / 100)), valueByte, 10);
        pC_UPDATEPROGRESS->setValue(valueByte);
        pC_UPDATEPROGRESS->notify();
        free(valueByte);
  });
}
```

But only the Serial.printf is executed..
it's a WiFi + BLE limitation?

Sladerix, 2022-06-21:
> I have never tried it or seen an example, I usually avoid even updating displays or neopixels.

Actually I ended by showing the progress on an LCD, instead of sending it over BLE, and it worked. So for what I’ve to do I’m happy like that.

_(small note: both in the serial monitor and on LCD the percentage do not go to 100% but stops about 77% and then the esp32 restarts.

The flashing goes ok because it works then.  

I think this is related to the “asynchronous-ness” of the update process (?) that can’t fire the callback all the way to the end because the update actually finishes before it can call the “onProgress” the last few times? 
Just curious, not an important thing)_

ha11otronix, 2024-03-29:
> OK. Now it works as expected!
>
> Thanks very much!!!
>
> Full code example for future people
>
> ```c++
> #include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
>
> void handlePreOtaUpdateCallback(){ [...]

Is it possible to show the same Progress on browser itself instead of Serial printing? I may not have access to Serial while doing OTA then showing it on browser helps.

tablatronix, 2024-03-30:
I tried that and ota is receiving data via webserver+wifi and its not really possibly to reliablly send data during this, I find that if ota misses even 200ms it can fail