tzapu/WiFiManager #1437: Usage of WiFiManager with flashing text file to LittleFS

bbrr132, 2022-06-21:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP8266

Core Version:      	2.4.0, staging

### Description
Hello, I am trying to use the WiFi manager with my ESP01 while using littlefs. My previous sketches uses littlefs to load a text file I flashed to the esp01 using ESP8266 sketch data upload. However, when uploading the wifimanager program, my littlefs data is lost and flashing the data after upload overwrites the wifimanager.

### Settings in IDE

Module:  			ESP01

Additional libraries: 
LittleFS, FS
### Sketch

```cpp
#BEGIN
#include <Arduino.h>
#include <LittleFS.h>
#include <FS.h>

String readFile(fs::FS &fs, const char * path){
  Serial.printf("Reading file: %s\r\n", path);
  File file = fs.open(path, "r");
  if(!file || file.isDirectory()){
    Serial.println("- empty file or failed to open file");
    return String();
  }
  Serial.println("- read from file:");
  String fileContent;
  while(file.available()){
    fileContent+=String((char)file.read());
  }
  file.close();
  Serial.println(fileContent);
  return fileContent;
}
void writeFile(fs::FS &fs, const char * path, const char * message){
  Serial.printf("Writing file: %s\r\n", path);
  File file = fs.open(path, "w");
  if(!file){
    Serial.println("- failed to open file for writing");
    return;
  }
  if(file.print(message)){
    Serial.println("- file written");
  } else {
    Serial.println("- write failed");
  }
  file.close();
}

int data = 4; 

#include <WiFiManager.h>
#define TRIGGER_PIN 2
int timeout = 120; // seconds to run for

void setup() {
data = readFile(LittleFS, "/data.txt").toInt();
WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
  WiFiManager wm;
  //wm.resetSettings();
  bool res;
  res = wm.autoConnect("Setup");
  if(!res) {
     Serial.println("Failed to connect");
     // ESP.restart();
  } 

}

void loop() {
if ( digitalRead(TRIGGER_PIN) == LOW) {
    WiFiManager wm;    
    //wm.resetSettings();
    wm.setConfigPortalTimeout(timeout);
    if (!wm.startConfigPortal("Sharmander")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      ESP.restart();
      delay(5000);
    }
    Serial.println("connected...yeey :)");
}
}
#END
```

### Debug Messages

```
messages here
```

bbrr132, 2022-06-21:
Missed out 
if (!LittleFS.begin()) {
    Serial.println("LittleFS mount failed");
    return;
  }
 resulting in FS not starting

closed 2022-06-21 by bbrr132

tablatronix, 2022-06-21:
Can you post or submit a PR for a littlefs examples?

bbrr132, 2022-06-22:
### Changed made to fix the issue: added LittleFS.begin() in setup to start LittleFS

```cpp
#BEGIN
#include <Arduino.h>
#include <LittleFS.h>
#include <FS.h>

String readFile(fs::FS &fs, const char * path){
  Serial.printf("Reading file: %s\r\n", path);
  File file = fs.open(path, "r");
  if(!file || file.isDirectory()){
    Serial.println("- empty file or failed to open file");
    return String();
  }
  Serial.println("- read from file:");
  String fileContent;
  while(file.available()){
    fileContent+=String((char)file.read());
  }
  file.close();
  Serial.println(fileContent);
  return fileContent;
}
void writeFile(fs::FS &fs, const char * path, const char * message){
  Serial.printf("Writing file: %s\r\n", path);
  File file = fs.open(path, "w");
  if(!file){
    Serial.println("- failed to open file for writing");
    return;
  }
  if(file.print(message)){
    Serial.println("- file written");
  } else {
    Serial.println("- write failed");
  }
  file.close();
}

int data = 4; 

#include <WiFiManager.h>
#define TRIGGER_PIN 2
int timeout = 120; // seconds to run for

void setup() {
if (!LittleFS.begin()) { //to start littlefs
Serial.println("LittleFS mount failed");
return;
}
data = readFile(LittleFS, "/data.txt").toInt();
WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
  WiFiManager wm;
  //wm.resetSettings();
  bool res;
  res = wm.autoConnect("Setup");
  if(!res) {
     Serial.println("Failed to connect");
     // ESP.restart();
  } 

}

void loop() {
if ( digitalRead(TRIGGER_PIN) == LOW) {
    WiFiManager wm;    
    //wm.resetSettings();
    wm.setConfigPortalTimeout(timeout);
    if (!wm.startConfigPortal("Sharmander")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      ESP.restart();
      delay(5000);
    }
    Serial.println("connected...yeey :)");
}
}
#END
```

tablatronix, 2022-08-05:
https://github.com/tzapu/WiFiManager/commit/9e408e41a8000a7b87d8651a3e7e148c30421d5f