tzapu/WiFiManager #1439: getWiFiIsSaved returning wrong value

OekoSolveMG, 2022-06-22:
### Basic Infos

### Hardware
WiFimanager Branch / Release: v2.0.11-beta
Esp8266 / Esp32: ESP32
Hardware: M5 Stack Core 1
Core Version: 2.4.0, staging

### Description
When the WiFi has not been completly initalized needs some time `WiFi_SSID` has some trash data in ```conf.sta.ssid```, which then gets converted into a String meaning the String isn't empty. Causing `getWiFiIsSaved` to return true because the ssid isn't empty even tough there isn't any data saved.

### Simple fix

Instead of always returning the data converted into a string, which might cause issues see current code below:

**WiFiManager.cpp / 3602**
```cpp
if (persistent){
    wifi_config_t conf;
    esp_wifi_get_config(WIFI_IF_STA, &conf);
    return String(reinterpret_cast<const char*>(conf.sta.ssid));
}
```

The library should instead check if getting the configuration was successfull in the first place and if not return an empty string. Similar to how it returns the value if the boolean flag `persistent` isn't true.

**WiFiManager.cpp / 3602**
```cpp
if (persistent){
    wifi_config_t conf;
    if (!esp_wifi_get_config(WIFI_IF_STA, &conf)) {
        Serial.println(String(reinterpret_cast<char*>(conf.sta.ssid)));
        return String(reinterpret_cast<const char*>(conf.sta.ssid));
    }
    return String();
}
```

murmeltier08, 2023-01-31:
Same problem here. Please fix it :)

tablatronix, 2023-01-31:
This was removed already from the code

murmeltier08, 2023-02-01:
but it not working  before autoconnect. i need to check this flag in the beginning of the code.

tablatronix, 2023-02-01:
ahh yeah its not working at the moment I need to add wifiinit code, I was testing how to do it without causing delays in boot connect..

The easiest way is to just init wifi yourself in your code add a WiFi.mode(WIFI_STA); at the beginnig

Vitallyz, 2023-02-25:
Hi! Is there any alternative way I can check if the drive is going to try to connect using the existing wifi config or if it will enter AP mode? 

Or is there a way to know that we are going to enter (or entered) AP mode? Trying to show some info on the display of my device. 

Thanks!

tablatronix, 2023-03-23:
Still thinking of how to fix this, now that the fatal wifi connect bugs in ESP seem to be fixed I can test wifiinit and try to optimize any connection delays from any change I make