tzapu/WiFiManager #1440: QUESTION: Optional params in wifiConnectNew to allow passing BSSID and Channel to Wifi.Begin to quicken connection?

3dprintscanner, 2022-06-23:
As part of an IOT project I'm working on, getting a quick Wifi connection is essential to save battery and a great optimisation for this is to cache the BSSID and channel (`Wifi.BSSID()` and `Wifi.Channel()`) of the router which makes connection times twice as fast as even using a static IP.

Currently I'm storing these values in memory but would like to have an easy way of passing them back into WifiManager to make use of the `autoconnect` functionality. 

It looks like the `wifiConnectNew` method could be optionally overloaded to allow passing of BSSID and Channel values.

Something like:
`bool WiFiManager::wifiConnectNew(String ssid, String pass,bool connect, int channel = 0, *int BSSID = NULL)`

tablatronix, 2022-06-25:
fast connect branch is being tested

tablatronix, 2022-06-25:
#1342

closed 2022-06-25 by tablatronix