tzapu/WiFiManager #1442: Restoring Persistent Wi-Fi Data on the ESP8266

bergutja, 2022-06-28:
https://github.com/tzapu/WiFiManager/blob/65fec943173e254c36b152867a53c23a4135c713/WiFiManager.cpp#L3638-L3639

I think these 2 lines of code should be like this:
if(persistent) wifi_station_get_config(&conf);
    else wifi_station_get_config_default(&conf);

This resolves restoring persistent wifi data on the ESP8266.

tablatronix, 2022-06-28:
Resolves what? Whats the issue?

tablatronix, 2022-06-28:
[image]

closed 2022-06-28 by tablatronix

bergutja, 2022-06-29:
Solution to: "This resolves restoring persistent wifi data on the ESP8266."

Solve the problem in the ESP8266 of loading the wifi data configured and saved via the web.
Well, every time it restarted it went into configuration mode and did not connect the saved wifi.

I understand that persistent wifi is to save the wifi and keep it on every reboot. I'm wrong?

Thank you

tablatronix, 2022-06-29:
I mean it has been working, noone has opened any bugs saying anything was broken. Persistent should already be working. If you are having issues with data saving, it could be corrupt flash, try a full erase, or a bug in an older version of esp library you are using

bergutja, 2022-06-30:
If you look at the code, when the "persistent" variable is true it retrieves the default data, but if the variable is false it retrieves the saved data, shouldn't it be the other way around?
I have changed those 2 lines and now it works perfectly for me when recovering the data saved on the web, before they were continuously deleted.
Thank you and greetings.