tzapu/WiFiManager #1444: setDebugOutput(false) causes the automatic Wi-Fi connection to fail

Maurizio1, 2022-06-30:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: Esp32

Hardware:			LILYGO TTGO T-Camera WROVER PIR Mic (v1.6.2)

Core Version:      	2.4.0, staging

### Description

When setting setDebugOutput(false) Wi-Fi connection will systematically fail.

### Settings in IDE

Module:  			Esp32 Dev Module

Additional libraries:  

### Sketch

```cpp
#BEGIN
#include WiFi.h
#include FS.h
#include esp_camera.h

void setup() {

    WiFiManager wifiManager;
    bool connected = false;

    wifiManager.setDebugOutput(false); // ATTENTION - when debug is set to false, connection to the configured Wi-Fi SSID fails systematically and the AP comes available...
    wifiManager.setConfigPortalTimeout(30);

    Serial.println("\nCheck Wi-Fi connection...");

    connected = wifiManager.autoConnect("iomitacam", "Password1"); 
}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2022-06-30:
yikes, I bet theres a bad debug block check

tablatronix, 2022-06-30:
confirmed, but i have no idea wtf is going on

tablatronix, 2022-06-30:
Its a race condition, ill fix it

tablatronix, 2022-06-30:
b8205e71f35c942c3714537563d0553b6f5828a6

closed 2022-06-30 by tablatronix

TI-SH-K205, 2022-07-01:
> yikes, I bet theres a bad debug block check

Sir . Can i see you same things.. about wifi auto cannect in phone . With esp8266

As any programs .. to connect wifi without password . Only using esp8266 

My WhatsApp number .+8801799259190 

Place sir contract me on

tablatronix, 2022-07-01:
The workaround would be set mode to sta and add a delay immediately after setup.

tablatronix, 2022-07-01:
esp8266 works fine

TI-SH-K205, 2022-07-02:
Sir can we meet in WhatsApp place 🙏

+8801799259190

tablatronix, 2022-07-02:
we have a discord, or create a new issue for your specific issue, do not highjack closed issues

TI-SH-K205, 2022-07-04:
Sir someone give me this link.. is this link is safe for me

tablatronix, 2022-07-04:
Its on the readme

Maurizio1, 2022-07-04:
> esp8266 works fine, this is unrelated to this issue, and your wasting my time, thanks.
> 
> @Maurizio1 did you get a chance to test latest git?

I'm sorry. I didn't mean to waste your time.
Of course I did test your patch and it worked.
That's why I wrote "Thank you @tablatronix" before my last post and put a "thumb up" emoji on your answer. Sorry if I was not clear enough.

As for the "debugging" that you mentioned I just added the "delay(100)" instruction, that's all. My intent was to help to pinpoint the cause.
Also, on my device (ESP32), I get more than 25 failed attempts before Wi-Fi starts.

> E (3382) wifi:Association refused temporarily, comeback time 200 mSec
> E (3588) wifi:Association refused temporarily, comeback time 200 mSec
> E (3793) wifi:Association refused temporarily, comeback time 200 mSec
> E (3998) wifi:Association refused temporarily, comeback time 200 mSec
> E (4203) wifi:Association refused temporarily, comeback time 200 mSec
> E (4408) wifi:Association refused temporarily, comeback time 200 mSec
> E (4613) wifi:Association refused temporarily, comeback time 200 mSec
> E (4818) wifi:Association refused temporarily, comeback time 200 mSec
> E (5024) wifi:Association refused temporarily, comeback time 200 mSec
> E (5229) wifi:Association refused temporarily, comeback time 200 mSec
> E (5434) wifi:Association refused temporarily, comeback time 200 mSec
> E (5639) wifi:Association refused temporarily, comeback time 200 mSec
> E (5849) wifi:Association refused temporarily, comeback time 200 mSec
> E (6054) wifi:Association refused temporarily, comeback time 200 mSec
> E (6260) wifi:Association refused temporarily, comeback time 200 mSec
> E (6465) wifi:Association refused temporarily, comeback time 200 mSec
> E (6670) wifi:Association refused temporarily, comeback time 200 mSec
> E (6875) wifi:Association refused temporarily, comeback time 200 mSec
> E (7080) wifi:Association refused temporarily, comeback time 200 mSec
> E (7285) wifi:Association refused temporarily, comeback time 200 mSec
> E (7490) wifi:Association refused temporarily, comeback time 200 mSec
> E (7695) wifi:Association refused temporarily, comeback time 200 mSec
> E (7900) wifi:Association refused temporarily, comeback time 200 mSec
> E (8106) wifi:Association refused temporarily, comeback time 200 mSec
> E (8311) wifi:Association refused temporarily, comeback time 200 mSec
> E (8516) wifi:Association refused temporarily, comeback time 200 mSec

Again, I'm not trying to devour your time, I'm just giving some more info in the hope it can help, so please forgive me.
Anyway, your patch worked for me, so I am OK with that. Thank you again.

tablatronix, 2022-07-04:
Sorry @Maurizio1 that was not to you, I deleted some other posts

Maurizio1, 2022-07-04:
> 

No problem @tablatronix. :-)

Sure! I will open another issue for that then.

Thank you