tzapu/WiFiManager #1447: wifi:Association refused temporarily on ESP32 - TTGO T-Camera WROVER PIR Mic (v1.6.2)

Maurizio1, 2022-07-04:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp32: LILYGO TTGO T-Camera WROVER PIR Mic (v1.6.2)

Hardware: ESP32 WROVER

Core Version:      	2.4.0, staging

### Description

Problem description

During WiFi connection many timeouts are experienced.

This happens **on first reboot** after flashing and **on hardware resets**.
During "warm" reboots, software driven, everything is OK.

### Settings in IDE

Module: ESP32 Dev Module

Additional libraries: AsyncTelegram2

### Sketch

```cpp
#BEGIN
#include <FS.h>
#include <WiFi.h>
#include <AsyncTelegram2.h>
#include "esp_camera.h"
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

void setup() {

    WiFiManager wifiManager;
    bool connected = false;

    wifiManager.setConnectTimeout(30);
    wifiManager.setTimeout(5);
    wifiManager.setDebugOutput(true);
    wifiManager.setConfigPortalTimeout(30);

    Serial.println("\nCheck Wi-Fi connection...");

    connected = wifiManager.autoConnect("iomitacam", "Password1");

    Serial.print("\nTest Telegram connection... ");
    myBot.begin() ? Serial.println("Telegram connection OK!") : Serial.println("Telegram connection failed!");

}

void loop() {

}
#END
```

### Debug Messages

```
Check Wi-Fi connection...
*wm:[1] AutoConnect 
*wm:[2] ESP32 event handler enabled 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: digitech
*wm:[2] 30000 ms timeout, waiting for connect...
*wm:[2] . 
E (3406) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (3612) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (3817) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (4026) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (4233) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (4441) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (4647) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (4852) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (5058) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (5264) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (5469) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (5675) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (5880) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (6086) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (6291) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
E (6498) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (6703) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (6908) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (7113) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (7319) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (7524) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (7729) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (7934) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (8139) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (8344) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
E (8549) wifi:Association refused temporarily, comeback time 200 mSec
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] . 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS 
*wm:[2] Connected in 7126 ms
*wm:[1] STA IP Address: 10.0.0.58
Test Telegram connection...
Connection to Telegram server established
Telegram connection OK!
```

John-Reid, 2023-03-14:
I have had some success with this code in the setup phase:

.
.
.
  // delete old config
  Serial.println("WiFi disconnect ");
  WiFi.disconnect(true);

  int limitcounter = 0;
  Serial.println("WiFi connect ");
    while (((WiFi.status() != WL_CONNECTED)) && (limitcounter < 30)) {
    delay(500);
    Serial.print(".");
    limitcounter++;
    //Serial.print (limitcounter);
  }
  WiFi.disconnect(true); 

.
.
.

While not an answer, it might give a clue to a fix, or just getting the connection establishment reasonably reliable. 
'limitcounter' at 20 is not as good.

Amulqueen, 2025-03-16:
i dont want it to connecct to a network just have an access point