tzapu/WiFiManager #1449: DHCP renew

Laxilef, 2022-07-07:
#### Hardware
WiFimanager: 2.0.11 beta
Esp8266: 
Hardware:			wemos d1 mini pro
Core Version:      	3.0.2

### Description
After changing the IP address for esp8266 on the router, esp does not do dhcp renew. No reboots help.
It seems to me that esp writes down the received ip and doesn't do renew anymore.
Only one thing helps: changing the ssid on the access point and reconnecting the esp.
What am I doing wrong?

### Sketch

```cpp
  void setup() {
    // code
    // ....
    // ....

    wm.addParameter(wmMqttPrefix);

    //wm.setCleanConnect(true);
    wm.setRestorePersistent(false);

    wm.setHostname(settings.hostname);
    wm.setWiFiAutoReconnect(true);
    wm.setConfigPortalBlocking(false);
    wm.setSaveParamsCallback(saveParamsCallback);
    wm.setConfigPortalTimeout(300);
    wm.setDisableConfigPortal(false);

    if (wm.autoConnect(AP_SSID)) {
      INFO_F("Wifi connected. IP: %s, RSSI: %d\n", WiFi.localIP().toString().c_str(), WiFi.RSSI());
      wm.startWebPortal();

    } else {
      INFO(F("Failed to connect to WIFI, start the configuration portal..."));
    }
  }

  void loop() {
    wm.process();
  }
```

tablatronix, 2022-07-08:
hmm, thats indeed odd, esp only saved the address if you give it a static via sta config.. So I am not sure.
Ideally I would bring this down to a basic esp example and try to reproduce, removing wm from the picture..

Laxilef, 2022-07-10:
Tested. I tried changing the IP settings for esp and forcibly disconnecting the client (esp). In this case, everything is fine, renew happens. Code:
```
#include <ESP8266WiFi.h>
const char* ssid = "AP";
const char* password = "password";

void setup()  {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Boot");
  WiFi.hostname("ESP-device");
  WiFi.begin(ssid, password);
}
 
void loop() {
  static unsigned long last = 0;
  static bool connected = false;

  if (connected && WiFi.status() != WL_CONNECTED) {
    connected = false;
    Serial.println("Disconnected");

  } else if (!connected && WiFi.status() != WL_CONNECTED && millis() - last >= 500) {
    Serial.print(".");
    last = millis();

  } else if (!connected && WiFi.status() == WL_CONNECTED) {
    connected = true;

    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
}
```
Log:
```
Boot
..........
Connected to AP
IP address: 10.224.0.186
Disconnected
.....
Connected to AP
IP address: 10.224.0.185
Disconnected
.....
Connected to AP
IP address: 10.224.0.184
```

closed 2022-07-11 by Laxilef